realsenseai/realsense-ros #2766: Driver keep switching between uvcvideo and usbfs

lucienchia, 2023-06-05:
Hi, 

When I have roslaunch the camera, I realise sometimes the camera took a long time to give out data (around 1min - 3min++).
Upon some troubleshooting, when roslaunch, I "usb-device" on the terminal and realise that the drivers keep switching between uvcvideo and usbfs (sometimes it will go from uvcvideo to none to uvcvideo, other time it will go from uvcvideo to none to usbfs). May I ask if there is a bug in switching the drivers of the camera?

MartyG-RealSense, 2023-06-05:
Hi @lucienchia  There are not many references about unpredictable switching between uvcvideo and usbfs.  It is not usually a problem that is experienced with RealSense and so is unlikely to be a bug in the RealSense SDK or its ROS wrapper, and it occurs in non-RealSense cases too.  In non-RealSense cases where it has occurred, it was suggested that it may happen because of USB disconnecting and it was therefore recommended to consider whether there was a problem with the USB cable.

Which Linux kernel version are you using, please?

lucienchia, 2023-06-06:
I am using a ubuntu 16.04 server image.

MartyG-RealSense, 2023-06-06:
Officially supported kernel versions are listed in the image below.

[image]

Is your Ubuntu version using one of these kernels, please?

lucienchia, 2023-06-07:
we are using 4.4.0-185 generic

lucienchia, 2023-06-07:
@MartyG-RealSense Here are the logs, I forgot to mention that I am running 2 to 4 cameras (usually 2 D435i cameras).

... logging to /home/sesto/.sesto/logs/roslog/67dd6998-0564-11ee-a0f1-c400ad671df5/roslaunch-sestomagnus-8369.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.88.10:44647/

SUMMARY
========

PARAMETERS
 * /camera_back/realsense2_camera/accel_fps: 250
 * /camera_back/realsense2_camera/accel_frame_id: camera_back_accel...
 * /camera_back/realsense2_camera/accel_optical_frame_id: camera_back_accel...
 * /camera_back/realsense2_camera/align_depth: False
 * /camera_back/realsense2_camera/aligned_depth_to_color_frame_id: camera_back_align...
 * /camera_back/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_back_align...
 * /camera_back/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_back_align...
 * /camera_back/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_back_align...
 * /camera_back/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_back_align...
 * /camera_back/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_back_align...
 * /camera_back/realsense2_camera/allow_no_texture_points: True
 * /camera_back/realsense2_camera/base_frame_id: camera_back_link
 * /camera_back/realsense2_camera/calib_odom_file: 
 * /camera_back/realsense2_camera/clip_distance: -1.0
 * /camera_back/realsense2_camera/color_fps: 6
 * /camera_back/realsense2_camera/color_frame_id: camera_back_color...
 * /camera_back/realsense2_camera/color_height: 480
 * /camera_back/realsense2_camera/color_optical_frame_id: camera_back_color...
 * /camera_back/realsense2_camera/color_width: 848
 * /camera_back/realsense2_camera/decimation/filter_magnitude: 6
 * /camera_back/realsense2_camera/depth_fps: 6
 * /camera_back/realsense2_camera/depth_frame_id: camera_back_depth...
 * /camera_back/realsense2_camera/depth_height: 480
 * /camera_back/realsense2_camera/depth_optical_frame_id: camera_back_depth...
 * /camera_back/realsense2_camera/depth_width: 848
 * /camera_back/realsense2_camera/device_type: 
 * /camera_back/realsense2_camera/enable_accel: False
 * /camera_back/realsense2_camera/enable_color: True
 * /camera_back/realsense2_camera/enable_depth: True
 * /camera_back/realsense2_camera/enable_fisheye1: False
 * /camera_back/realsense2_camera/enable_fisheye2: False
 * /camera_back/realsense2_camera/enable_fisheye: False
 * /camera_back/realsense2_camera/enable_gyro: False
 * /camera_back/realsense2_camera/enable_infra1: False
 * /camera_back/realsense2_camera/enable_infra2: False
 * /camera_back/realsense2_camera/enable_infra: False
 * /camera_back/realsense2_camera/enable_pointcloud: False
 * /camera_back/realsense2_camera/enable_pose: False
 * /camera_back/realsense2_camera/enable_sync: False
 * /camera_back/realsense2_camera/filters: decimation,pointc...
 * /camera_back/realsense2_camera/fisheye1_frame_id: camera_back_fishe...
 * /camera_back/realsense2_camera/fisheye1_optical_frame_id: camera_back_fishe...
 * /camera_back/realsense2_camera/fisheye2_frame_id: camera_back_fishe...
 * /camera_back/realsense2_camera/fisheye2_optical_frame_id: camera_back_fishe...
 * /camera_back/realsense2_camera/fisheye_fps: 6
 * /camera_back/realsense2_camera/fisheye_frame_id: camera_back_fishe...
 * /camera_back/realsense2_camera/fisheye_height: 480
 * /camera_back/realsense2_camera/fisheye_optical_frame_id: camera_back_fishe...
 * /camera_back/realsense2_camera/fisheye_width: 848
 * /camera_back/realsense2_camera/gyro_fps: 400
 * /camera_back/realsense2_camera/gyro_frame_id: camera_back_gyro_...
 * /camera_back/realsense2_camera/gyro_optical_frame_id: camera_back_gyro_...
 * /camera_back/realsense2_camera/imu_optical_frame_id: camera_back_imu_o...
 * /camera_back/realsense2_camera/infra1_frame_id: camera_back_infra...
 * /camera_back/realsense2_camera/infra1_optical_frame_id: camera_back_infra...
 * /camera_back/realsense2_camera/infra2_frame_id: camera_back_infra...
 * /camera_back/realsense2_camera/infra2_optical_frame_id: camera_back_infra...
[... 288 lines cut ...]
 08/06 02:52:23,778 WARNING [140046862640896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/06 02:52:23,829 WARNING [140046862640896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/06 02:52:23,879 WARNING [140046862640896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/06 02:52:23,929 WARNING [140046862640896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/06 02:52:23,979 WARNING [140046862640896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/06 02:52:24,030 WARNING [140046862640896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/06 02:52:24,080 WARNING [140046862640896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/06 02:52:24,333 WARNING [140046862640896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 08/06 02:52:25,445 WARNING [140046862640896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/06 02:52:25,495 WARNING [140046862640896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/06 02:52:25,545 WARNING [140046862640896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/06 02:52:25,596 WARNING [140046862640896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

lucienchia, 2023-06-07:
we also notice that once the data starts to flood in, this error: (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61 stops

lucienchia, 2023-06-12:
@MartyG-RealSense May I know what is the minimum requirement of hardwares that is required to run 2-4 d435i devices?

lucienchia, 2023-06-12:
@MartyG-RealSense in our trouble shoot, we decide to rebuild the realsense lib from source. 
The cameras did manage to give out data (took around 11-20 sec).
However the dmesg was flooded with uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
I realise using "usb-devices", the drivers remained as uvcvideo. 

Hence to prevent dmesg from flooding,  
I rebuild again, this time with  RSUSB_BACKEND as True. 
I realise Im back to square one with the driver switching from uvcvideo to usbfs and  (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61 keep poping up. (This time the cameras sometimes took 1-3mins to provide data).

MartyG-RealSense, 2023-06-12:
For 2 cameras I would recommend a computer with a processor chip equivalent to Intel Core i3.  For 3 cameras, Core i5.  For 4 cameras, Core i7.

A uvcvideo error indicates a conflict with the Linux kernel.  If the dmesg messages disappear when using -DFORCE_RSUSB_BACKEND=true, this would seem to affirm that it is a kernel issue because an SDK build that uses RSUSB = true bypasses the kernel.

Does the flood of messages when not using RSUSB negatively affect the streams or do they perform okay and the messages are just an annoyance?

MartyG-RealSense, 2023-06-18:
Hi @lucienchia  Do you require further assistance with this case, please?  Thanks!

lucienchia, 2023-06-19:
Case can be close for now. 
The messages are annoying and does not affect the stream. 
For now we tried to run the camera nodes in sequence (after one camera successfully bootsup, we start the other camera), it manage to run properly.
Thanks @MartyG-RealSense .

closed 2023-06-19 by lucienchia