realsenseai/realsense-ros #2769: Limiting CPU for wrapper

tugbakara, 2023-06-07:
Hi, I use utility that is taskset to limit CPU usage and I launched relasense2_camera and I know there will be some performance limit after cpu limitation but is there anything could be bad?

MartyG-RealSense, 2023-06-07:
Hi @tugbakara  The librealsense SDK that the ROS wrapper runs as a layer upon will use a single core for its processing by default anyway unless it is built from source code with CMake with the build flag **BUILD_WITH_OPENMP=TRUE** included so that YUY to RGB conversion and Depth-Color spatial alignment can take advantage of multiple cores.

tugbakara, 2024-02-15:
Hi @MartyG-RealSense ! I am wondering that if I use librealsense source installation with GPU true flag which parts of the ros codes will use gpu?
And I install ros librealsense distro from apt not installing it from source RSUSB etc do I still have the gpu option as true?

MartyG-RealSense, 2024-02-15:
Hi @tugbakara  You would need to build librealsense from source code to enable OpenMP as it is set to False by default in the SDK, so would be disabled in an apt package installation of librealsense.

When you refer to 'GPU true', do you mean the flag **-DBUILD_WITH_CUDA** please?  If so then you would require an Nvidia GPU for CUDA to be enabled and work in the ROS wrapper.  An Nvidia Jetson computing board, which has a built-in Nvidia GPU, is usually used with -DBUILD_WITH_CUDA.  When using an Nvidia Jetson and librealsense's CUDA support is enabled, the ROS wrapper functions that benefit automatically from acceleration are depth-color alignment and pointcloud generation.

tugbakara, 2024-02-15:
> **-DBUILD_WITH_CUDA**

yes I meant this flag, I have Nvidia Jetson board, in hat case I have actualy only one way to install librealsense  is from source with the flag then, I just want to distinguish their difference wrt CUDA is included or not, wrt your answer if I want to use CUDA enabled then  I need to install librealsense with CUDA true flag from source, not from apt, right?

MartyG-RealSense, 2024-02-15:
Though you would need to build from source to enable OpenMP, if you install from the Jetson apt packages at the link below then CUDA support is automatically enabled.  When building for Jetson from source instead of apt, the **-DBUILD_WITH_CUDA=ON** flag is required.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

tugbakara, 2024-02-15:
okay, thanks @MartyG-RealSense , there was a bit confusion in the question but now its okay! Have a nice day! :)