realsenseai/realsense-ros #2770: How to use T265 with ros2 humble?

mzahana, 2023-06-07:
Hi!

I have a T265 camera and I would like to use with ROS 2 `humble` .

I tried to use the `rs_launch.py` but I get the following output on the terminal. It seems that the camera is not recognized.
```bash
[realsense2_camera_node-1] [INFO] [1686121684.150617212] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1686121684.150939659] [camera.camera]: Built with LibRealSense v2.53.1
[realsense2_camera_node-1] [INFO] [1686121684.151042448] [camera.camera]: Running with LibRealSense v2.53.1
[realsense2_camera_node-1] [WARN] [1686121686.306030276] [camera.camera]: Device 1/1 failed with exception: Unable to open device interface
[realsense2_camera_node-1] [ERROR] [1686121686.306250542] [camera.camera]: The requested device with  is NOT found. Will Try again.
[realsense2_camera_node-1] [WARN] [1686121694.449949520] [camera.camera]: Device 1/1 failed with exception: Unable to open device interface
```

Am I missing anything ?
I could not find anything in the latest documentation related to that.

Thank you.

SampathGanesh01, 2023-06-26:
Hi mzahana did you find any resources ..?

mzahana, 2023-06-26:
@SampathGanesh01 unfortunately no.

jiazhuguanhua, 2025-05-07:
metoo！

MarioCavero, 2025-06-27:
In my case, topics are up, but fail to publish anything. 

```
ros2 launch realsense2_camera rs_launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-06-27-11-34-59-282891-USR-21212
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [11738]
[realsense2_camera_node-1] [1751024099.418625913 t265_node INFO] RealSense ROS v4.51.1
[realsense2_camera_node-1] [1751024099.418791018 t265_node INFO] Built with LibRealSense v2.53.1
[realsense2_camera_node-1] [1751024099.418813740 t265_node INFO] Running with LibRealSense v2.53.1
[realsense2_camera_node-1]  27/06 11:34:59,420 WARNING [133434634663488] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.17.auto
```

A lot of:
`[realsense2_camera_node-1]  27/06 11:34:59,420 WARNING [133434634663488] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.4.auto
`
And then

```
[realsense2_camera_node-1]  27/06 11:34:59,713 ERROR [133434634663488] (librealsense-exception.h:52) /dev/video1 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  27/06 11:34:59,713 ERROR [133434634663488] (sensor.cpp:664) acquire_power failed: /dev/video1 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  27/06 11:34:59,793 WARNING [133434634663488] (rs.cpp:310) null pointer passed for argument "device"
[realsense2_camera_node-1] [1751024099.793172954 t265_node WARN] Device 1/2 failed with exception: /dev/video1 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1] [1751024099.802958091 t265_node INFO] Device with serial number 15322110409 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [1751024099.803057452 t265_node INFO] Device with physical ID 2-2.3-63 was found.
[realsense2_camera_node-1] [1751024099.803072670 t265_node INFO] Device with name Intel RealSense T265 was found.
[realsense2_camera_node-1] [1751024099.803445915 t265_node INFO] Device with port number 2-2.3 was found.
[realsense2_camera_node-1] [1751024099.803463129 t265_node INFO] Device USB type: 3.1
[realsense2_camera_node-1] [1751024099.807286957 t265_node INFO] No calib_odom_file. No input odometry accepted.
[realsense2_camera_node-1] [1751024099.807334509 t265_node INFO] getParameters...
[realsense2_camera_node-1] [1751024099.807585345 t265_node INFO] JSON file is not provided
[realsense2_camera_node-1] [1751024099.807600684 t265_node INFO] Device Name: Intel RealSense T265
[realsense2_camera_node-1] [1751024099.807612502 t265_node INFO] Device Serial No: 15322110409
[realsense2_camera_node-1] [1751024099.807623185 t265_node INFO] Device physical port: 2-2.3-63
[realsense2_camera_node-1] [1751024099.807632425 t265_node INFO] Device FW version: 0.2.0.951
[realsense2_camera_node-1] [1751024099.807641119 t265_node INFO] Device Product ID: 0x0B37
[realsense2_camera_node-1] [1751024099.807649767 t265_node INFO] Sync Mode: Off
[realsense2_camera_node-1] [1751024099.807917908 t265_node WARN] Could not set param: tracking_module.frames_queue_size with 256 Range: [0, 32]: parameter 'tracking_module.frames_queue_size' could not be set: Parameter {tracking_module.frames_queue_size} doesn't comply with integer range.
[realsense2_camera_node-1] [1751024099.808712728 t265_node INFO] Set ROS param tracking_module.profile to default: 848x800x30
[realsense2_camera_node-1] [1751024099.808784737 t265_node INFO] Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [1751024099.808820824 t265_node INFO] Set ROS param accel_fps to default: 62
[realsense2_camera_node-1] [1751024099.814458809 t265_node WARN] Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {pointcloud.stream_index_filter} doesn't comply with integer range.
[realsense2_camera_node-1] [1751024099.815327943 t265_node INFO] Stopping Sensor: Tracking Module
[realsense2_camera_node-1] [1751024099.818787787 t265_node INFO] Starting Sensor: Tracking Module
[realsense2_camera_node-1] [1751024099.819544253 t265_node INFO] Open profile: stream_type: Fisheye(1), Format: Y8, Width: 848, Height: 800, FPS: 30
[realsense2_camera_node-1] [1751024099.819562820 t265_node INFO] Open profile: stream_type: Fisheye(2), Format: Y8, Width: 848, Height: 800, FPS: 30
[realsense2_camera_node-1] [1751024099.819571810 t265_node INFO] Open profile: stream_type: Pose(0)Format: 6DOF, FPS: 200
[realsense2_camera_node-1] [1751024099.844412033 t265_node INFO] RealSense Node Is Up!
[realsense2_camera_node-1] [1751024099.907222356 t265_node WARN] 
```

Topics seem to be up, but nothing comes up from them, nor in rviz. 

```
ros2 topic list
/extrinsics/pose_to_fisheye1
/extrinsics/pose_to_fisheye2
/fisheye1/camera_info
/fisheye1/image_raw
/fisheye1/metadata
/fisheye2/camera_info
/fisheye2/image_raw
/fisheye2/metadata
/imu
/parameter_events
/pose/metadata
/pose/sample
/rosout
/tf
[... 41 lines cut ...]
 27/06 11:45:44,370 INFO [136215629543296] (synthetic-stream-gl.cpp:89)  0 GPU objects initialized
 27/06 11:45:44,370 INFO [136215629543296] (synthetic-stream-gl.cpp:132) Initializing processing, GLSL=1
 27/06 11:45:44,404 INFO [136215629543296] (synthetic-stream-gl.cpp:146)  0 GPU objects initialized
[... cut ...]
 27/06 11:45:45,061 ERROR [136214728398400] (librealsense-exception.h:52) /dev/video1 is no video capture device Last Error: Invalid argument
 27/06 11:45:45,061 ERROR [136214728398400] (sensor.cpp:664) acquire_power failed: /dev/video1 is no video capture device Last Error: Invalid argument
 27/06 11:45:45,063 WARNING [136214728398400] (rs.cpp:310) null pointer passed for argument "device"
 27/06 11:45:45,063 WARNING [136214728398400] (rs.cpp:2704) Couldn't refresh devices - /dev/video1 is no video capture device Last Error: Invalid argument
```

closed 2026-10-23 by MartyG-RealSense