realsenseai/realsense-ros #2772: Error synthetic-stream.cpp:48 - Exception was thrown during callback: Out of frame resources!

adambiomech, 2023-06-07:
I am running the L515 with the Intel Realsense Viewer v2.50.00. 

I keep getting the following errors following a couple of seconds of recording:
17:36:47 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\proc\synthetic-stream.cpp:48 - Exception was thrown during callback: Out of frame resources!
17:46:52 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\proc\synthetic-stream.cpp:48 - Exception was thrown during callback: Error occured during execution of the processing block! See the log for more info

Both RGB sensor and Depth Sensor are operating at 640x480 at 30 FPS.
[image]

MartyG-RealSense, 2023-06-07:
Hi @adambiomech  Multiple cameras attached to the same computer will consume a greater amount of that computer's resources, making multicam unsuitable for computers / computing devices with a low-end hardware configuration.  Which computer are you using, please (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc).

Do the errors still occur if you use the Max Range preset for both cameras?

Do the errors occur if you try unplugging the micro-sized end of the USB cable from the base of the L515 camera, reversing the connector's orientation and plugging it back into the camera (USB-C cables are two-way insertion at the micro-sized end).

adambiomech, 2023-06-07:
Hi @MartyG-RealSense, 

I am using ROG Zephyrus, AMD Ryzen 9 6900HS with Radeon Graphics 3.30 GHz with 32 GB RAM. 

Both cameras are on Max Range I have just saved the settings for one camera to specify the pixel ratio for the RGB sensor. I haven't tried reversing the USB C end as I figured any orientation would result in the same outcome.

MartyG-RealSense, 2023-06-07:
A characteristic of the L515 model is that it works correctly when the micro-sized connector is in one particular orientation of the two directions available.

adambiomech, 2023-06-08:
This was correct it fine as a stream but I still got the errors after a second of recording. 
[image]

MartyG-RealSense, 2023-06-08:
Have you played back the recorded bag files to see if they have been affected by the messages?  You can test them by opening the Viewer without the camera attached and then drag and dropping the file into the center panel of the Viewer to automatically initiate playback.

adambiomech, 2023-06-09:
During playback I get the same errors. 
[image]

MartyG-RealSense, 2023-06-09:
Next, please try disabling bag recording compression in the Viewer's **Settings** interface.  On some computers the processor chip may have an incompatibility with the compression algorithm.

You can access the settings by left-clicking on the gear-wheel icon in the top corner of the Viewer window and selecting 'Settings' from it.

[image]

On the **Playback and Record** section of the Settings window, select the option **Never compress (larger bag size)** and click Apply and then OK.

[image]

adambiomech, 2023-06-12:
I have tried changing the compression settings to never. This reduces the number of errors but doesn't get rid of them entirely.

MartyG-RealSense, 2023-06-12:
Do the errors reduce further if you disable the two **GLSL** options under the 'Performance' section of the settings using the instructions at https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

adambiomech, 2023-06-12:
No, that seemed to cause more errors

MartyG-RealSense, 2023-06-12:
I have not been able to replicate the synthetic stream messages during recording.  Instead, when using two L515 on my laptops (including an AMD Ryzen 3), they only appeared once and never appeared again in subsequent Viewer sessions.

With the L515 camera model, the best outcome that you are likely to achieve may be a reduction of the errors rather than eliminating them completely, unfortunately.  In the image in https://github.com/IntelRealSense/realsense-ros/issues/2772#issuecomment-1584148071 it looks as though the bag file is playing back despite the log messages.

[image]

adambiomech, 2023-06-13:
Thanks for the advice, Marty. I think this is the best that can be achieved.

closed 2023-06-13 by adambiomech