realsenseai/realsense-ros #2774: Local ROS Topics Possibility?

husseinalijaafar, 2023-06-12:
Hello team,

I had a conceptual question. 

My experiments require me to use a high number of realsense cameras (20 cameras), each recording depth and rgb. 
Currently, I am using launch files to launch modified versions of rs_launch with my required parameters. 

The main aim of this is to simply record the topics as rosbags on a local Nvidia jetson that each camera is connected to. 
So for example, 

robot_1_d435i/camera/color/image_raw
robot_1_d435i/camera/depth/image_rect_raw
and the necessary topics

**My question is:** Can I publish these topics only locally? i.e, is there a way to publish them such that I can record a rosbag, but not publish them onto my local network. As all the bandwidth is causing the network to get overloaded.

MartyG-RealSense, 2023-06-12:
Hi @husseinalijaafar  Would it be possible to add a local external storage drive such as an SSD to your Jetson and then install ROS and the RealSense ROS wrapper to that drive and launch from that location so that the rosbags are stored locally on that drive?

An example of an SSD external drive installation guide for Jetson models such as Xavier is here:

https://jetsonhacks.com/2020/05/29/jetson-xavier-nx-run-from-ssd/

husseinalijaafar, 2023-06-12:
Hi Marty

Thanks for your feedback.

So the problem isn't with the storage bandwidth, rather network bandwidth. I am recording the rosbags locally on a U90 micro sd card, it can handle the bandwidth.

I want to essentially isolate the real sense topics from the network if possible.

MartyG-RealSense, 2023-06-12:
The only way that I can think of to competely isolate RealSense ROS topics from a network would be to install Ubuntu, ROS and the RealSense ROS wrapper onto the micro SD card and boot the Jetson from that card.

husseinalijaafar, 2023-06-12:
Hi Marty. 
I am currently doing that. I am launching the topics from the jetson directly, and I want to record the topics on the jetson BUT also not broadcast certain topics on the network. But I want to keep publishing other topics. 

I guess it is not possible, they **will** be broadcast if the jetson is connected to the same network.

MartyG-RealSense, 2023-06-12:
Individual topics can be removed by unsubscribing from them if you do not need them.  Unsubscribing from topics can reduce CPU overhead.

closed 2023-06-13 by husseinalijaafar

MartyG-RealSense, 2023-06-13:
You are very welcome.  Thanks very much for the update!

husseinalijaafar, 2023-06-13:
In case it is helpful, after further research, you can change `ROS_LOCALHOST_ONLY` from 0 to 1 
(`export ROS_LOCALHOST_ONLY=1`) to isolate the ROS instance onto the local host only, and prevent all ros topics from publishing and discovering the network topics to minimise network usage.

MartyG-RealSense, 2023-06-14:
Thanks @husseinalijaafar   :)