realsenseai/realsense-ros #2777: stuck on RealSense Node Is Up!

BaranYucecengiz, 2023-06-13:
When I run roslaunch realsense2_camera rs_camera.launch I get this output. Also, when I list with rostopic list, color information does not appear.

[image]

:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/byuce/.ros/log/b95dfb2a-09f6-11ee-923e-eb18f5f13f03/roslaunch-cudatech-7983.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 848
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/ordered_pc: False
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/reconnect_timeout: 6.0
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/stereo_module/exposure/1: 7500
 * /camera/realsense2_camera/stereo_module/exposure/2: 1
 * /camera/realsense2_camera/stereo_module/gain/1: 16
 * /camera/realsense2_camera/stereo_module/gain/2: 16
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /camera/realsense2_camera/wait_for_device_timeout: -1.0
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [7991]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b95dfb2a-09f6-11ee-923e-eb18f5f13f03
process[rosout-1]: started with pid [8001]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [8008]
process[camera/realsense2_camera-3]: started with pid [8009]
[ INFO] [1686666578.117422538]: Initializing nodelet with 4 worker threads.
[ INFO] [1686666578.351808734]: RealSense ROS v2.3.2
[ INFO] [1686666578.351934899]: Built with LibRealSense v2.54.1
[ INFO] [1686666578.352045079]: Running with LibRealSense v2.54.1
[ INFO] [1686666578.383170093]:  
[ INFO] [1686666578.403793953]: Device with serial number 215322073311 was found.

[ INFO] [1686666578.404072615]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video0 was found.
[ INFO] [1686666578.404248877]: Device with name Intel RealSense D435I was found.
[ INFO] [1686666578.406136051]: Device with port number 2-2 was found.
[ INFO] [1686666578.406263517]: Device USB type: 3.2
[ INFO] [1686666578.417119660]: getParameters...
[ INFO] [1686666578.490564012]: setupDevice...
[ INFO] [1686666578.490620145]: JSON file is not provided
[ INFO] [1686666578.490665405]: ROS Node Namespace: camera
[ INFO] [1686666578.490702647]: Device Name: Intel RealSense D435I
[ INFO] [1686666578.490751257]: Device Serial No: 215322073311
[ INFO] [1686666578.490796565]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video0
[ INFO] [1686666578.490890574]: Device FW version: 5.15.0.2
[ INFO] [1686666578.490975817]: Device Product ID: 0x0B3A
[ INFO] [1686666578.491015319]: Enable PointCloud: Off
[ INFO] [1686666578.491057612]: Align Depth: Off
[ INFO] [1686666578.491094279]: Sync Mode: Off
[ INFO] [1686666578.491181069]: Device Sensors: 
[ INFO] [1686666578.494171821]: Stereo Module was found.
[ INFO] [1686666578.502219212]: RGB Camera was found.
[ INFO] [1686666578.502519053]: Motion Module was found.
[ INFO] [1686666578.502618871]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1686666578.502692579]: num_filters: 0
[ INFO] [1686666578.502740261]: Setting Dynamic reconfig parameters.
[ INFO] [1686666578.550271303]: Done Setting Dynamic reconfig parameters.
[ INFO] [1686666578.551483216]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1686666578.553900210]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1686666578.553969293]: setupPublishers...
[ INFO] [1686666578.557289030]: Expected frequency for depth = 30.00000
[ INFO] [1686666578.602502447]: Expected frequency for color = 30.00000
[ INFO] [1686666578.634266257]: setupStreams...
[ INFO] [1686666578.667022586]: SELECTED BASE:Depth, 0
[ INFO] [1686666578.679859782]: RealSense Node Is Up!
[ WARN] [1686666578.799649928]:

MartyG-RealSense, 2023-06-13:
Hi @BaranYucecengiz  I cannot see any problems in your log and the log confirms that color is enabled and published.

```
[ INFO] [1686666578.502219212]: RGB Camera was found.

[ INFO] [1686666578.553900210]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8

[ INFO] [1686666578.602502447]: Expected frequency for color = 30.00000
```

When using the **rostopic list** command, if color is enabled then it would be expected that the following two color topics would be in the list.

```
/camera/color/camera_info
/camera/color/image_raw
```

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#published-topics

BaranYucecengiz, 2023-06-13:
As you can see in the screenshot I took, the program stops. Rviz won't open. When I open Rviz on another screen, the camera does not appear

MartyG-RealSense, 2023-06-13:
I note that you are using ROS1 wrapper version 2.3.2 and librealsense version 2.54.1.  It is important to match a particular ROS wrapper version as closely as possible with a specific librealsense version in that ROS wrapper's release notes.

Development of the ROS1 wrapper has ceased and only the ROS2 wrapper continues to be developed, so the ROS1 wrapper has not been updated for newer librealsense versions such as 2.54.1.  Ideally, the newest librealsense version that should be used with 2.3.2 is **2.51.1**.  Because the ROS1 wrapper is no longer updated, it will become more and more incompatible with librealsense as new SDK versions are released.

BaranYucecengiz, 2023-06-13:
How can I download the version you recommend? Do I have to build from source?

MartyG-RealSense, 2023-06-13:
The easiest method of installation is to install correctly matched versions of both librealsense and the ROS wrapper at the same time by installing from packages instead of source code with the command below:

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

The main disadvantage of this package install method compared to a source-code build is that the RealSense Viewer tool will not be installed.

If you have already built the ROS wrapper from source code, I recommend completely deleting the entire **/catkin_ws/src** catkin workspace directory first before using the package install command.

closed 2023-06-25 by MartyG-RealSense