realsenseai/realsense-ros #2779: SLAM with d435i

apotdar2023, 2023-06-14:
Hi, I have followed the wiki to perform SLAM using the realsense d435i camera. However, I am unable to export a point cloud of the entire map. Publish the/rtabmap/cloud_map yields several point clouds of the map. Is there a way to extract the whole thing as one point cloud. Also the points seem to be quite sparse. Is there a way to generate a dense point cloud similar to the one seen on th rviz tool.

apotdar2023, 2023-06-15:
Thank you for your response. I'll try out your suggestion and update on the same.
Also, I would like to understand how the d435i camera conducts SLAM. Apart from the wiki, is there a script or some base code available? Would like to see how the IMU data is unpacked and used for SLAM.

MartyG-RealSense, 2023-06-15:
The RealSense SDK (librealsense) does not have built-in SLAM support for the 400 Series stereo depth camera and the D435i SLAM guide is the only official guide available, so there is not a script or source code to examine, unfortunately.  

Using SLAM with RealSense typically involves interfacing the camera with a third-party SLAM solution such as ORB-SLAM, RTABMAP, OpenVSLAM or Kimera. 

At the link below, a RealSense user created an adaptation of Intel's D435i SLAM guide that added localization with RTABMAP and obtaining the relative position from a recorded pointcloud map.

https://shinkansan.github.io/2019-UGRP-DPoom/SLAM

MartyG-RealSense, 2023-06-25:
Hi @apotdar2023  Do you require further assistance with this case, please?  Thanks!

apotdar2023, 2023-06-26:
The above worked fine. Thank you for the support

closed 2023-06-26 by apotdar2023