realsenseai/realsense-ros #2780: Depth Frame cap at 15fps when running with ros 2 (auto exposure error)

chijie1998, 2023-06-15:
Hi, i am running D455 camera with Ros 2 Foxy in ubuntu 20.04 and I am having a strange frame drop for depth topic despite setting the depth module profile at 30 fps, i only can get 15 fps. There are a few error when launching the camera with rs_launch.py. I could get a 30 fps depth image stream when run with realsense-viewer without any problem or error.

[image]

Things that i have tried :
1. Gold factory reset
2. Downgrade firmware to the latest production firmware
3. Launch with 'depth_module.enable_auto_exposure':'true'
4. Comment out the {'name': 'depth_module.exposure.1',  and {'name': 'depth_module.exposure.2', line in rs_launch.py

Appreciate if you could help in resolving this issue. Thank you.

MartyG-RealSense, 2023-06-15:
Hi @chijie1998  The red errors **auto_exposure_limit** and **auto_gain_limit** are occurring because firmware version 5.14.0.0 is being used with librealsense SDK version 2.54.1.  Firmware 5.14.0.0 is incompatible with 2.54.1 due to changes made to its internal structure which mean that it must be used with the later SDK version 2.53.1.

The red errors should disappear if you downgrade your firmware version to the previous one, **5.13.0.50**.  The procedure for doing so in the RealSense Viewer tool is:

1.  Download the Viewer bin file for 5.13.0.50 to your computer from the link below.

https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads

2.  Go to the **More** option near the top of the Viewer's options side-panel and select **Update Firmware** from its drop-down menu.

3.  A file navigation box will pop up.  Navigate to the location on your computer where you downloaded the .bin file to and select the file.  The firmware downgrade should now automatically begin.

If you are unable to use the RealSense Viewer tool then the 5.13.0.50 bin file can also be used to perform the downgrade with the RealSense SDK tool **rs-fw-update**

chijie1998, 2023-06-16:
@MartyG-RealSense Thank you for your quick response! I tried to update the firmware following your instruction but realsense-veiwer just crached halfway.
[image]
[image]

Is this the GUI issue? I tried multiple times and the program crashed when the progress bar is around 30% as shown on the figure on top.

MartyG-RealSense, 2023-06-16:
At the 30% point in the process the camera performs a reset and disconnects in preparation for the second half of the firmware update process.  Sometimes though the camera is not re-detected after the reset and the update fails.

Are you performing the update using the official 1 meter RealSense USB cable or a longer USB cable of your own choice, please?

chijie1998, 2023-06-17:
I see. Actually this is a D457 camera and it doesn't comes with any official realsense USB cable. I am using Newnex 1meter cable. Is this the problem? I still have the old official cable that comes with D435 that I bought few years back, will there be any issue if i use this cable instead?

MartyG-RealSense, 2023-06-17:
Newnex produce high quality USB equipment and so a 1m Newnex USB cable should be fine for performing firmware update with.

Testing with an official cable from a D435 would also be fine as the cables can be swapped between any 400 Series camera model.

chijie1998, 2023-06-19:
Hi, i have tried to update the firmware using the official able but realsense viewer still crashed at 30%. Do you have any idea how to solve this? or is there any workaround to update the firmware?

chijie1998, 2023-06-20:
Thank you for your useful infromation. The issue is resolved. It seems like the problem is that I was using the realsense ros2 package from apt install. After removing it and manually build from the ros-development branch, I am able to get 3- fps for my depth frame. Thank you for you help.

closed 2023-06-20 by chijie1998

MartyG-RealSense, 2023-06-20:
You are very welcome.  I'm pleased to hear that you achieved a solution.  Thanks very much for the update!