realsenseai/realsense-ros #2783: compile failure in ROS noetic

nicehiro, 2023-06-17:
Hello, everyone. I'm encountering an compiling issue those days and couldn't solve it by myself.

System information:
```
realsense-ros: 2.3.2
OS: ubuntu 20.04
ros: noetic
librealsense2: 2.54.1
```

Installing by using Method 2.

Part of compiling log:
```
Starting  >>> realsense2_camera                                                               
Starting  >>> realsense2_description                                                          
Finished  <<< realsense2_description                [ 0.1 seconds ]                           
______________________________________________________________________________________________
Errors     << realsense2_camera:make /home/pal/catkin_ws/logs/realsense2_camera/build.make.001.log
In file included from /opt/ros/noetic/include/ros/common.h:38,
                 from /opt/ros/noetic/include/ros/ros.h:43,
                 from /opt/ros/noetic/include/image_transport/publisher.h:38,
                 from /opt/ros/noetic/include/image_transport/image_transport.h:38,
                 from /home/pal/catkin_ws/src/realsense-ros/realsense2_camera/include/../include/realsense_node_factory.h:8,
                 from /home/pal/catkin_ws/src/realsense-ros/realsense2_camera/include/base_realsense_node.h:6,
                 from /home/pal/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/t265_realsense_node.h:3,
                 from /home/pal/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:1:
/opt/ros/noetic/include/ros/serialized_message.h: In constructor ‘ros::SerializedMessage::SerializedMessage(boost::shared_array<unsigned char>, size_t)’:
/opt/ros/noetic/include/ros/serialized_message.h:57:3: error: declaration of ‘num_bytes’ shadows a member of ‘ros::SerializedMessage’ [-Werror=shadow]
   57 |   : buf(buf)
      |   ^
/opt/ros/noetic/include/ros/serialized_message.h:43:10: note: shadowed declaration is here
   43 |   size_t num_bytes;
      |          ^~~~~~~~~
/opt/ros/noetic/include/ros/serialized_message.h:57:3: error: declaration of ‘buf’ shadows a member of ‘ros::SerializedMessage’ [-Werror=shadow]
   57 |   : buf(buf)
      |   ^
/opt/ros/noetic/include/ros/serialized_message.h:42:32: note: shadowed declaration is here
   42 |   boost::shared_array<uint8_t> buf;
      |                                ^~~
/opt/ros/noetic/include/ros/serialized_message.h: In constructor ‘ros::SerializedMessage::SerializedMessage(boost::shared_array<unsigned char>, size_t)’:
/opt/ros/noetic/include/ros/serialized_message.h:61:5: error: declaration of ‘num_bytes’ shadows a member of ‘ros::SerializedMessage’ [-Werror=shadow]
   61 |   { }
      |     ^
/opt/ros/noetic/include/ros/serialized_message.h:43:10: note: shadowed declaration is here
   43 |   size_t num_bytes;
      |          ^~~~~~~~~
/opt/ros/noetic/include/ros/serialized_message.h:61:5: error: declaration of ‘buf’ shadows a member of ‘ros::SerializedMessage’ [-Werror=shadow]
   61 |   { }
      |     ^
/opt/ros/noetic/include/ros/serialized_message.h:42:32: note: shadowed declaration is here
   42 |   boost::shared_array<uint8_t> buf;
      |                                ^~~
/opt/ros/noetic/include/ros/serialized_message.h: In constructor ‘ros::SerializedMessage::SerializedMessage(boost::shared_array<unsigned char>, size_t)’:
/opt/ros/noetic/include/ros/serialized_message.h:61:5: error: declaration of ‘num_bytes’ shadows a member of ‘ros::SerializedMessage’ [-Werror=shadow]
   61 |   { }
      |     ^
/opt/ros/noetic/include/ros/serialized_message.h:43:10: note: shadowed declaration is here
   43 |   size_t num_bytes;
      |          ^~~~~~~~~
/opt/ros/noetic/include/ros/serialized_message.h:61:5: error: declaration of ‘buf’ shadows a member of ‘ros::SerializedMessage’ [-Werror=shadow]
   61 |   { }
      |     ^
/opt/ros/noetic/include/ros/serialized_message.h:42:32: note: shadowed declaration is here
   42 |   boost::shared_array<uint8_t> buf;
      |                                ^~~
...
```

I can't install `ros-noetic-realsense2-camera` by using `apt` since I need to use local-built `ddynamic-reconfigure` package. And installing from `apt` will re-install this package in `/opt/ros/noetic/`.

So how to solve the error and compile it from source? Please let me know if there's anything I can do. Thanks.

MartyG-RealSense, 2023-06-17:
Hi @nicehiro  The message **error: declaration of ‘num_bytes’ shadows a member of ‘ros::SerializedMessage’** has not previously been reported on this RealSense ROS support forum.

I note that you are using librealsense **2.54.1** with ROS1 wrapper **2.3.2**.  It is important to match a particular wrapper version as closely as possible with a specific librealsense version in that wrapper version's release notes.  Development of the RealSense ROS1 wrapper has ceased, with the focus now on the ROS2 wrapper, so the ROS1 wrapper has not been updated for newer SDKs such as 2.53.1 and 2.54.1.  

The most recent librealsense version that should be used with 2.3.2 is **2.51.1**.  The camera firmware driver version should also be **5.13.0.50**.

*************

An approach that you could consider is to install librealsense and the wrapper together from packages with apt-get, and then remove the ddynamic-reconfigure package and compile your local ddynamic-reconfigure from source code to replace it.  A removal command might look something like this:

`apt-get remove ros-noetic-ddynamic-reconfigure`

MartyG-RealSense, 2023-06-23:
hi @nicehiro  Do you require further assistance with this case, please?  Thanks!

nicehiro, 2023-06-24:
@MartyG-RealSense Sorry for late response.

> The most recent librealsense version that should be used with 2.3.2 is **2.51.1**. The camera firmware driver version should also be **5.13.0.50**.

I try to use `librealsense==2.51.1` but still cannot compile.

> apt-get remove ros-noetic-ddynamic-reconfigure

This will also remove all packages depend on this, that means `realsense-ros` will be removed too.

Then I use `dpkg -r --force-depends ros-noetic-ddynamic-reconfigure`, but that will break apt dependence tree and trigger an error everytime I use apt. And the `realsense-ros` still detect two version of `ddynamic-reconfigure` package.

Thanks for your help.

MartyG-RealSense, 2023-06-24:
As mentione dabove, the message **error: declaration of ‘num_bytes’ shadows a member of ‘ros::SerializedMessage’** has not previously been reported on this RealSense ROS support forum, so there are no references available about how to approach a solution, unfortunately.  When an installation has got in a tangle due to multiple installs / uninstalls then I usually recommend wiping the computer (including Ubuntu) and installing everything fresh if possible in order to resolve hidden problems that cannot otherwise be solved.

nicehiro, 2023-06-24:
@MartyG-RealSense Clean wipe did clear all the problem.

But my problem is how to install `realsense-ros` package when there's already a `ddynamic-reconfigure` compiled in the system. Since I use a prebuild ubuntu system (contains packages for a particular robot).

My current solution is using one computer to install and run `realsense-ros` and another computer to run robot particular packages.

I will close this issue now since this is an outdated version. Many thanks for your following up and help.

closed 2023-06-24 by nicehiro