realsenseai/realsense-ros #2784: roslaunch rs_camera error in noetic

seonjuno0709, 2023-06-17:
OS: ubuntu 20.04
HW: Jetson xavier
ROS: noetic
librealsense2: 2.42.0

Hi, I'm asking because I couldn't resolve the error while running the package.
I tried installing about 4 realsense-sdk installations from the latest version, but the same error occurred.
I would appreciate your help :) Thank You.

------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

nvidia@ubuntu:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/nvidia/.ros/log/2afec842-0d15-11ee-ae1d-204ef60a07fb/roslaunch-ubuntu-33950.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:44171/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
[... 78 lines cut ...]
[ INFO] [1687009510.356791448]: Enable PointCloud: Off
[ INFO] [1687009510.356867257]: Align Depth: Off
[ INFO] [1687009510.356940410]: Sync Mode: Off
[ INFO] [1687009510.357130428]: Device Sensors: 
[ INFO] [1687009510.505187064]: Stereo Module was found.
[ INFO] [1687009510.596141263]: RGB Camera was found.
[ INFO] [1687009510.598094406]: Motion Module was found.
[ INFO] [1687009510.598414858]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1687009510.598561260]: num_filters: 0
[ INFO] [1687009510.598666285]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1687009514.961103526]: Done Setting Dynamic reconfig parameters.
/opt/ros/noetic/lib/nodelet/nodelet: symbol lookup error: /home/nvidia/catkin_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN2cv3MatC1Ev
[camera/realsense2_camera_manager-2] process has died [pid 33975, exit code 127, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/nvidia/.ros/log/2afec842-0d15-11ee-ae1d-204ef60a07fb/camera-realsense2_camera_manager-2.log].
log file: /home/nvidia/.ros/log/2afec842-0d15-11ee-ae1d-204ef60a07fb/camera-realsense2_camera_manager-2*.log
^C[camera/realsense2_camera-3] killing on exit
[ WARN] [1687009517.080739480]: Couldn't find service realsense2_camera_manager/unload_nodelet, perhaps the manager is already shut down
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2023-06-17:
Hi @seonjuno0709  Noting that you are using a Jetson Xavier, a RealSense ROS user at https://github.com/IntelRealSense/realsense-ros/issues/2467#issuecomment-1268417031 shares the method for how they resolved their **undefined symbol: _ZN2cv3MatC1Ev** error.  They were using a Jetson AGX Orin in their particular case.

seonjuno0709, 2023-06-19:
Thank you Bro. I solved it

closed 2023-06-19 by seonjuno0709

MartyG-RealSense, 2023-06-19:
You are very welcome.  It's great to hear that you were successful.  Thanks very much for the update!