realsenseai/realsense-ros #2785: align_depth.enable makes node to throw exception: out of frame resources

alonsnir, 2023-06-19:
Device: 
`Intel RealSense D435`

Hardware:
```
UP Squared V2
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         39 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  2
  On-line CPU(s) list:   0,1
Vendor ID:               GenuineIntel
  Model name:            Intel(R) Celeron(R) N6210 @ 1.20GHz
```

OS:
```
Ubuntu 22.04
kernel: 5.15.0-75-generic 
```

Ros2: Humle

Installation method used:
`sudo apt install ros-humble-realsense2-*`

Working mode:
```
louis@louis:~$ ros2 run realsense2_camera realsense2_camera_node
[INFO] [1687178401.177457250] [camera]: RealSense ROS v4.51.1
[INFO] [1687178401.177683760] [camera]: Built with LibRealSense v2.51.1
[INFO] [1687178401.177736947] [camera]: Running with LibRealSense v2.51.1
[INFO] [1687178401.186069750] [camera]: Device with serial number 829212072204 was found.

[INFO] [1687178401.186150046] [camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video0 was found.
[INFO] [1687178401.186178003] [camera]: Device with name Intel RealSense D435 was found.
[INFO] [1687178401.186474347] [camera]: Device with port number 2-2 was found.
[INFO] [1687178401.186497129] [camera]: Device USB type: 3.2
[INFO] [1687178401.189459752] [camera]: getParameters...
[INFO] [1687178401.189813559] [camera]: JSON file is not provided
[INFO] [1687178401.189837869] [camera]: Device Name: Intel RealSense D435
[INFO] [1687178401.189877379] [camera]: Device Serial No: 829212072204
[INFO] [1687178401.189894900] [camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video0
[INFO] [1687178401.189915740] [camera]: Device FW version: 05.13.00.50
[INFO] [1687178401.189935381] [camera]: Device Product ID: 0x0B07
[INFO] [1687178401.189956807] [camera]: Sync Mode: Off
[WARN] [1687178401.205771968] [camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
 19/06 12:40:01,206 WARNING [140468458305088] (backend-v4l2.cpp:1444) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[INFO] [1687178401.221463878] [camera]: Stopping Sensor: Stereo Module
[INFO] [1687178401.279847482] [camera]: Starting Sensor: Stereo Module
[INFO] [1687178401.315289024] [camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1687178401.315461430] [camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1687178401.315511162] [camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[INFO] [1687178401.315862415] [camera]: Stopping Sensor: RGB Camera
[INFO] [1687178401.369793849] [camera]: Starting Sensor: RGB Camera
[INFO] [1687178401.372408027] [camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[INFO] [1687178401.373261646] [camera]: RealSense Node Is Up!
[WARN] [1687178401.433307878] [camera]: 
```

Error mode, when using alignment:
```
ros2 run realsense2_camera realsense2_camera_node --ros-args -p align_depth.enable:=true -p initial_reset:=true
[INFO] [1687178670.249431200] [camera]: RealSense ROS v4.51.1
[INFO] [1687178670.249595445] [camera]: Built with LibRealSense v2.51.1
[INFO] [1687178670.249617800] [camera]: Running with LibRealSense v2.51.1
[INFO] [1687178670.257888429] [camera]: Device with serial number 829212072204 was found.

[INFO] [1687178670.257972731] [camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video0 was found.
[INFO] [1687178670.257991832] [camera]: Device with name Intel RealSense D435 was found.
[INFO] [1687178670.258296214] [camera]: Device with port number 2-2 was found.
[INFO] [1687178670.258317004] [camera]: Device USB type: 3.2
[INFO] [1687178670.258335242] [camera]: Resetting device...
[INFO] [1687178676.604257289] [camera]: Device with serial number 829212072204 was found.

[INFO] [1687178676.604352785] [camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video0 was found.
[INFO] [1687178676.604371173] [camera]: Device with name Intel RealSense D435 was found.
[INFO] [1687178676.604650817] [camera]: Device with port number 2-2 was found.
[INFO] [1687178676.604672218] [camera]: Device USB type: 3.2
[INFO] [1687178676.607769000] [camera]: getParameters...
[INFO] [1687178676.608508735] [camera]: JSON file is not provided
[INFO] [1687178676.608533028] [camera]: Device Name: Intel RealSense D435
[INFO] [1687178676.608548710] [camera]: Device Serial No: 829212072204
[INFO] [1687178676.608563795] [camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video0
[INFO] [1687178676.608579301] [camera]: Device FW version: 05.13.00.50
[INFO] [1687178676.608593613] [camera]: Device Product ID: 0x0B07
[INFO] [1687178676.608607169] [camera]: Sync Mode: Off
[WARN] [1687178676.630291605] [camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
 19/06 12:44:36,630 WARNING [140064973641280] (backend-v4l2.cpp:1444) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[INFO] [1687178676.647050654] [camera]: Stopping Sensor: Stereo Module
[INFO] [1687178676.704880486] [camera]: Starting Sensor: Stereo Module
[INFO] [1687178676.760962781] [camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1687178676.761062888] [camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1687178676.761083848] [camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[INFO] [1687178676.761404020] [camera]: Stopping Sensor: RGB Camera
[INFO] [1687178676.764019344] [camera]: Starting Sensor: RGB Camera
[INFO] [1687178676.766572551] [camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[INFO] [1687178676.767461006] [camera]: RealSense Node Is Up!
[WARN] [1687178676.827634224] [camera]: 
 19/06 12:44:37,690 ERROR [140064822638144] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 19/06 12:44:37,757 ERROR [140064822638144] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 19/06 12:44:37,790 ERROR [140064822638144] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 19/06 12:44:37,823 ERROR [140064822638144] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
```

MartyG-RealSense, 2023-06-19:
Hi @alonsnir  Depth-color alignment is calculated on the computer's CPU instead of the camera hardware and can be processing-intensive, which could be leading to the **Out of frame resources** messages on your device due to its low-end CPU.

Does the error also occur if you use ros2 launch instead of ros2 run, please?

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=848x480x30 rgb_camera.profile:=640x480x30 align_depth.enable:=true initial_reset:=true`

alonsnir, 2023-06-20:
Hi @MartyG-RealSense 
Closing this issue, as the problem was ros communication not being limited to localhost (and a client listening from other host). Once changed the error disappeared.

closed 2023-06-20 by alonsnir

MartyG-RealSense, 2023-06-20:
Thanks very much for the update, @alonsnir - it's great to hear that you found a solution!