realsenseai/realsense-ros #2788: About Depth Registration

kenloumixx, 2023-06-22:
Hello, 
I am trying to register depth image to RGB image or vice versa. In the process I'd like to ask several questions. 
1. Is there any process provided by realsense that register the images? I used [depth_image_proc/register](http://wiki.ros.org/depth_image_proc) from ROS and it is very slow. I also know that you provide [rs_aligned_depth launch file](https://github.com/IntelRealSense/realsense-ros/blob/ros1-legacy/realsense2_camera/launch/rs_aligned_depth.launch) but when I run this command, it outputs the error ` 22/06 10:02:31,586 ERROR [546685555072] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!`. I used the default parameters in the launch file which is 
```
 SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters:
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
[... 1035 lines cut ...]
   0   1   0
   0   0   1

 Translation Vector: 0.00552000012248755  -0.00510000018402934  -0.011739999987185  

Extrinsic from "Accel"  To  "Gyro" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Accel"  To  "Accel" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

```
where I think the depth and RGB both has 640x480, so for me it seems that `out of frame resources` is not correct. 

2. When I launch the [rs_camera.launch](https://github.com/IntelRealSense/realsense-ros/blob/ros1-legacy/realsense2_camera/launch/rs_camera.launch), the resolution of depth image and RGB image is respectively (848x480) and (1280x720). In the output of the above `rs-enumerate-devices -c` command, I see that depth image with resolution (1280x720) is also used. I know that for the registration process, making the resolution of both image same using methods such as interpolation or else is necessary. Then, how can I use the same resolution? I tried the argument of the resolution to other values such as -2 or -3 but it didn't work. 

Thank you for reading the beginner's question..!

MartyG-RealSense, 2023-06-22:
Hi @kenloumixx  'Registering' depth and RGB is a way of describing the process of depth-color alignment.  If you are using the ROS1 wrapper's rs_camera.launch then you can align depth and color by adding the command **align_depth:=true** to your roslaunch instruction.

`roslaunch realsense2_camera rs_camera.launch align_depth:=true`

The align_depth parameter will use the RealSense SDK's **align processing block** to map depth and color together and automatically adjust for differences in resolution between the two streams.

If you wish to use the same resolution for both streams instead of having align_depth adjust for different resolution sizes then you can define a custom configuration in the launch instruction.

`roslaunch realsense2_camera rs_camera.launch align_depth:=true depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30`

*******

An alternative means of aligning depth to color is to use the **rs_rgbd** launch file, which publishes an RGB-textured pointcloud to depth/image/proc.  

First, install support for an RGBD launch by inputting the instruction below into the Ubuntu terminal.

`sudo apt-get install ros-melodic-rgbd-launch`

Then perform an rs_rgbd launch.

`roslaunch realsense2_camera rs_rgbd.launch depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30`

You do not need to enable align_depth as it is already enabled by default in the rs_rgbd launch file.

kenloumixx, 2023-06-22:
Thank you for the response :) As I want to use this topics to ROS, after I run the command `roslaunch realsense2_camera rs_camera.launch align_depth:=true`, is there any method that I can check if the depth-color is well-aligned? can I know it by visualization or any metrics?

MartyG-RealSense, 2023-06-22:
You can view an image in the **RViz** interface by using the command `rosrun rviz rviz`

RealSense ROS1 users sometimes find that an rs_rgbd launch performs better on their computer than rs_camera.launch when generating a pointcloud, especially for users of Nvidia Jetson computing boards.

The main differences between rs_camera and rs_rgbd were that rs_rgbd published its pointcloud to depth/image/proc whilst rs_camera publishes it to the topic /camera/depth/color/points.  The other key difference is that by default rs_camera publishes an **unordered** pointcloud and rs_rgbd publishes an **ordered** pointcloud.  

There is though a parameter for rs_camera called **ordered_pc** that publishes the point cloud as ordered instead of unordered, so the differences between the two launch files are fewer now.

Information about the meaning of unordered and ordered pointclouds can be found at https://github.com/IntelRealSense/realsense-ros/issues/1563#issuecomment-742439647

kenloumixx, 2023-06-22:
Thank you for the answer :) Then I should probably use the `rs_rgbd.launch` command. Meanwhile, to compare two launch results I run the command `roslaunch realsense2_camera rs_camera.launch align_depth:=true` and check rostopics but there was no /camera/depth/color/points. Below is the result of `rostopic list` and `rs_camera.launch`
rostopic list 
```
/camera/align_to_color/parameter_descriptions
/camera/align_to_color/parameter_updates
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/color/camera_info
/camera/color/image_raw
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static

```

rs_camera.launch align_depth:=true
```
... logging to /home/unitree/.ros/log/357eedb0-10e4-11ee-aee9-48b02d3d9c60/roslaunch-nx-18859.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nx:38123/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
[... 102 lines cut ...]
[ INFO] [1687428291.824472427]: Align Depth: On
[ INFO] [1687428291.824531340]: Sync Mode: On
[ INFO] [1687428291.824693773]: Device Sensors: 
[ INFO] [1687428292.018183120]: Stereo Module was found.
[ INFO] [1687428292.272253792]: RGB Camera was found.
[ INFO] [1687428292.274904475]: Motion Module was found.
[ INFO] [1687428292.275934950]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1687428292.283152080]: num_filters: 1
[ INFO] [1687428292.288710569]: Setting Dynamic reconfig parameters.
 22/06 18:04:54,720 WARNING [547398607232] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[ INFO] [1687428297.296702771]: Done Setting Dynamic reconfig parameters.
[ INFO] [1687428297.302884243]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1687428297.312694135]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1687428297.322771135]: setupPublishers...
[ INFO] [1687428297.370161157]: Expected frequency for depth = 30.00000
[ INFO] [1687428297.434015956]: Expected frequency for color = 30.00000
[ INFO] [1687428297.487292151]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1687428297.508909557]: setupStreams...
 22/06 18:04:58,133 WARNING [547398607232] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1687428298.184104678]: SELECTED BASE:Depth, 0
 22/06 18:04:58,229 WARNING [547398607232] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1687428298.293509737]: 
 22/06 18:04:58,381 WARNING [547398607232] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/06 18:04:58,432 WARNING [547398607232] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

MartyG-RealSense, 2023-06-22:
rs_rgbd.launch publishes a pointcloud automatically by default but on the rs_camera.launch file you need to define a **pointcloud filter** by adding` filters:=pointcloud` to the launch instruction.

`roslaunch realsense2_camera rs_camera.launch align_depth:=true depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30 filters:=pointcloud`

More information about this can be found here:

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#point-cloud

kenloumixx, 2023-06-22:
Thank you for the response :) May I ask another question?? 
I am trying to run this package in jetson computing board, and therefore I will run rs_rgbd.launch as you mentioned. I want xyz pointcloud, not the entire rgbxyz which is the result of rs_rgbd.launch. In this case, is there any way that I can only utilize the xyz of pointcloud? or do I have to modify the code to receive rgbxyz and cut off the rgb part? Thank you so much for your answers!

kenloumixx, 2023-06-22:
I figured out that I should [change xyzrgb to xyz](https://github.com/IntelRealSense/realsense-ros/blob/9398c98d522dbc40f7c06cde5188e78645603d37/realsense2_camera/launch/rs_rgbd.launch#L184), thank you so much for your responses :)

closed 2023-06-22 by kenloumixx

MartyG-RealSense, 2023-06-22:
You could adding `pointcloud_texture_stream:=RS2_STREAM_ANY` to your launch instruction so that the pointcloud tries to use the infrared stream for pointcloud texturing instead of RGB.  The infrared streams are disabled by default in the ROS wrapper.

Depth-color alignment is a processing-expensive operation, but on Jetson boards you can significantly reduce the processing burden by installing the librealsense SDK with **CUDA** support enabled.  This support is automatically included if librealsense is installed from packages, or on a source code build with CMake it can be added manually by including the flag **-DBUILD_WITH_CUDA=ON** in the CMake build instruction.

kenloumixx, 2023-06-29:
@MartyG-RealSense Thank you for the answer :) Since I use ros2 galactic, I cloned this repo with [`ros2-hkr` branch](https://github.com/IntelRealSense/realsense-ros/tree/ros2-hkr/realsense2_camera/launch) and try to use the command `ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true`. If I require xyz instead of xyzrgb for faster processing, do I need to just change [this line](https://github.com/IntelRealSense/realsense-ros/blob/6c1ae906d675c4c45f8edf61da3579bf346c29fa/realsense2_camera/scripts/rs2_listener.py#L40C12-L40C12) or is there any preferred way to faster the alignment process? TIA

MartyG-RealSense, 2023-06-29:
Setting `allow_no_texture_points `to false does not prevent a pointcloud from being textured with RGB.  It just colors outer regions of the image where depth information has been cut off from the aligned image on D435 / D435i RealSense camera models where the color sensor FOV is smaller than the depth sensor FOV.  

In regard to pointcloud performance when using the ROS2 wrapper: if you already have CUDA enabled then you could try setting `pointcloud.ordered_pc:=true` to publish the pointcloud as an **ordered** pointcloud instead of the default unordered cloud to see how it affects performance.