realsenseai/realsense-ros #2789: 0 pointcloud depth on an object

TapleFlib, 2023-06-22:
Is it normal for depth to be 0 on an object that is not supposed to has 0 depth value? I use yolo to detect object and sometimes the center point of the object has 0 value.

MartyG-RealSense, 2023-06-22:
Hi @TapleFlib  A value of '0' on a surface could indicate that this particular area of the object's surface was not readable by the camera.  An example of such a surface could be one that is colored black, is reflective or has a strong light source shining on it.

TapleFlib, 2023-06-23:
got it , thanks so much , do you know if there's realsense scientific research on why black colored object, reflective or has a strong light source shining on it isn't readable by the camera?

MartyG-RealSense, 2023-06-23:
The problem with reading black surfaces is due to a general physics principle (not specific to RealSense) that dark grey or black absorbs light and so makes it more difficult for depth cameras to read depth information from such surfaces.  The darker the color shade, the more light that is absorbed and so the less depth detail that the camera can obtain.

Reflections create glare that can block out details of a surface.  The picture below demonstrates how fitting a **linear polarization filter** product over the lenses on the front of the camera can negate glare to provide a clear image.

[image]

If a strong light source is cast on a surface then the area where the light is focused may just be a blob of plain bright white on an RGB image that can provide no useful information.  If the area is not readable then on the depth image it may appear as a plain black area that has no depth values in it.

TapleFlib, 2023-06-23:
I got it, thanks so much !

closed 2023-06-23 by TapleFlib