realsenseai/realsense-ros #2790: How to visualize fov

TapleFlib, 2023-06-23:
H:87±3 / V:58±1 /D:95±3 this is the FOV of Realsense D455 , but I'm not sure how to visualize it , I draw an abstract of how I think it is visualized : 
[image]
is this correct?

MartyG-RealSense, 2023-06-23:
Hi @TapleFlib  The shape of your D455 drawing is consistent with one provided by another RealSense user for the D435i model.

[image]

An FOV drawing from the official data sheet documented is presented like this:

[image]

TapleFlib, 2023-06-23:
I see thankyou so much !

closed 2023-06-23 by TapleFlib

tonynajjar, 2025-08-06:
Hey @MartyG-RealSense, this is probably not the right place to ask for this but I don't know where else. Could you put in a feature request to include the accurate FOV visualization in the CAD files? It would help people position their cameras on their robots  for maximum "coverage"