realsenseai/realsense-ros #2791: Low  performance on Xavier NX

dev10110, 2023-06-24:
Hi @MartyG-RealSense 

I've been trying to use the realsense d455 with a Jetson Xavier NX 16GB. I've mounted an nvme ssd as my main drive. 

I've been struggling to get high framerate on depth images. My main requirement is to be able to generate pointclouds at ~30fps although 60 would be nicer. Once I have the pointcloud I intend to do a voxelGrid filtering with a resolution of about 2.5cm. 

I've been able to get ~15fps if I use the nvidia Isaac ros docker files but when I use my own docker files (where I've installed the package through apt-get ros-humble-realsense-ros) the performance is about 8-10 fps. This is the same fps for depth/image_rect_raw and the pointcloud 

On a laptop, I've tested my entire pipeline and it works even at 60fps. Now im just trying to get this performance on the Xavier. 

I've also noticed 0 GPU usage while this is running. Perhaps that gives a clue? 

Unrelated question: when I do ppointcloud.enable = true, is the pointcloud construction happening in the Xavier cpu, the Xavier gpu, or inside the d455? Ie, would it be beneficial to do the pointcloud constructing and voxelGrid filtering in a separate node myself and only subscribing to the depth image?  

Sorry for the many questions, 
Dev

MartyG-RealSense, 2023-06-24:
Hi @dev10110  the '0' GPU usage could indicate that **CUDA** support is not enabled.  You can significantly reduce CPU usage when using pointclouds with a Jetson board by building the librealsense SDK with CUDA support enabled if it is not already.  

CUDA support is automatically enabled when building librealsense from Jetson packages, or manually by including the build flag **-DBUILD_WITH_CUDA** if building from source code with CMake.

Pointclouds are calculated on the CPU if CUDA support is not enabled, and on the GPU when it is enabled.

dev10110, 2023-06-26:
Hi @MartyG-RealSense 

Thanks for your reply. I think you were right about the GPU usage - I suspect that when I constructed the dockerfile, it did not have access to the GPU. 

For anyone else coming here in the future, do yourself a favor and look at the dockerfiles provided by Nvidia: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox and in particular try running the demo (https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/main/docs/tutorial-realsense.md) 

Using the dockerfiles in this  repo, I was able to get 30 and 60fps streaming of depth images from the realsense on the xavier into ROS.

closed 2023-06-26 by dev10110

MartyG-RealSense, 2023-06-27:
Thanks very much @dev10110 for the update about your success and the sharing of your solution!