realsenseai/realsense-ros #2796: process has died

bardpedersen, 2023-06-29:
When i run the test launch:
```
roslaunch realsense2_camera rs_camera.launch
```
The process shuts down. It works on another computer with the same camera.

Specs:
Ubuntu: 16.04
Ros: Kinetic
Kernal: 4.15.0-52-generic
```

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://thorvald-005.local:11311

process[camera/realsense2_camera_manager-1]: started with pid [28354]
process[camera/realsense2_camera-2]: started with pid [28355]
[ INFO] [1688024158.405759108]: Initializing nodelet with 4 worker threads.
[ INFO] [1688024158.483976919]: RealSense ROS v2.2.6
[ INFO] [1688024158.484008088]: Running with LibRealSense v2.22.0
[ INFO] [1688024158.964823304]: getParameters...
[ INFO] [1688024159.058758695]: setupDevice...
[ INFO] [1688024159.058790087]: JSON file is not provided
[ INFO] [1688024159.058816369]: ROS Node Namespace: camera
[ INFO] [1688024159.058850301]: Device Name: Intel RealSense D435
[ INFO] [1688024159.058875386]: Device Serial No: 825312073170
[ INFO] [1688024159.058899149]: Device FW version: 05.11.01.100
[ INFO] [1688024159.058923913]: Device Product ID: 0x0B07
[ INFO] [1688024159.058941690]: Enable PointCloud: Off
[ INFO] [1688024159.058961809]: Align Depth: Off
[ INFO] [1688024159.058984295]: Sync Mode: Off
[ INFO] [1688024159.059265293]: Device Sensors: 
[ INFO] [1688024159.059466232]: Stereo Module was found.
[ INFO] [1688024159.059520200]: RGB Camera was found.
[ INFO] [1688024159.059580752]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1688024159.059604965]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1688024159.059625621]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1688024159.059644747]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1688024159.059662968]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1688024159.059682662]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1688024159.059869185]: setupPublishers...
[ INFO] [1688024159.063337446]: Expected frequency for depth = 30.00000
[ INFO] [1688024159.104702608]: Expected frequency for infra1 = 30.00000
[ INFO] [1688024159.138988777]: Expected frequency for infra2 = 30.00000
[ INFO] [1688024159.173329751]: Expected frequency for color = 30.00000
[ INFO] [1688024159.209648562]: setupStreams...
 29/06 09:35:59,211 WARNING [140167930218240] (sensor.cpp:365) Unregistered Media formats : [ UYVY ]; Supported: [ GREY UYVY Y12I Y8I  Z16  ]
[ INFO] [1688024159.215216536]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1688024159.215611136]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1688024159.215887606]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1688024159.225945996]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1688024159.228667798]: insert Depth to Stereo Module
[ INFO] [1688024159.228986428]: insert Color to RGB Camera
[ INFO] [1688024159.229065348]: insert Infrared to Stereo Module
[ INFO] [1688024159.229118971]: insert Infrared to Stereo Module
[ INFO] [1688024159.250847170]: num_filters: 0
[ INFO] [1688024159.251076656]: SELECTED BASE:Depth, 0
[ INFO] [1688024159.260779052]: RealSense Node Is Up!
[ INFO] [1688024159.260881324]: Setting Dynamic reconfig parameters.
/opt/ros/kinetic/lib/nodelet/nodelet: symbol lookup error: /home/thorvald/pick_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN20ddynamic_reconfigure19DDynamicReconfigureC1ERKN3ros10NodeHandleE
[camera/realsense2_camera_manager-1] process has died [pid 28354, exit code 127, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/thorvald/.ros/log/217b0064-1644-11ee-b92c-7d8ddbccc605/camera-realsense2_camera_manager-1.log].
log file: /home/thorvald/.ros/log/217b0064-1644-11ee-b92c-7d8ddbccc605/camera-realsense2_camera_manager-1*.log
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/thorvald/.ros/log/217b0064-1644-11ee-b92c-7d8ddbccc605/camera-realsense2_camera-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

MartyG-RealSense, 2023-06-29:
Hi @bardpedersen  The exit seems to be happening when attempting to access the ROS package **ddynamic_reconfigure**.  The error` _ZN20ddynamic_reconfigure19DDynamicReconfigure` can indicate that ddynamic _reconfigure is missing or not installed.

Is the error resolved if you install ddynamic_reconfigure for kinetic using the commands below, please?

```
sudo apt-get update
sudo apt-get install ros-kinetic-ddynamic-reconfigure
```

bardpedersen, 2023-06-29:
The package is ready installed, so when i run the commands i get
```
0 upgraded, 0 newly installed, 0 to remove and 937 not upgraded.
```

MartyG-RealSense, 2023-06-29:
Looking at the wrapper release notes, 2.2.6 has no notes accompanying it.  This can sometimes be a warning flag to avoid using that version. 

[image]

If it is compulsory to use librealsense SDK 2.22.0 then I would recommend using wrapper **2.2.5** instead of 2.2.6.

Using such an old wrapper is not ideal because ROS Kinetic has been end of life (retired) for a while now and is no longer updated, but I can understand that you may have to use it because of Ubuntu 16.04.

closed 2023-07-01 by bardpedersen