realsenseai/realsense-ros #2797: Aligned Depth to Color Image not being published

hello-jackson, 2023-06-29:
Hey team,

I am using a D435i camera with LibRealSense v2.54.1 and Device FW version: 5.15.0.2 with ROS 2 Galactic and the realsense-ros package built from source with RealSense ROS v4.54.1. 

When I launch the camera with the parameter align_depth.enable=true, the RGB feed on /camera/color/image_raw feed starts to flicker and the topic /camera/aligned_depth_to_color topic is not published. This worked for me until recently, and the same setup for the firmware and SDK works in ROS Noetic. Has something changed in the ROS 2 driver that causes the aligned_depth_to_color topic to fail?

Thank you,
Chintan Desai

hello-chintan, 2023-06-29:
Oops. I'll be following up with this account. Please tag me when you reply.

Chintan Desai

MartyG-RealSense, 2023-06-30:
Hi @hello-chintan  There have not been any other reports yet of the problem that you described occurring with ROS wrapper 4.54.1.

Does the problem still occur if you launch with **ros2 run** instead of ros2 launch ?

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p align_depth.enable:=true`

hello-chintan, 2023-06-30:
Hey @MartyG-RealSense, yes, I can confirm that this happens with the `ros2 run realsense2_camera realsense2_camera_node --ros-args -p align_depth.enable:=true` command as well. The issue is reproducible while working on ROS 2 Humble as well with the most recent realsense_ros drivers built from source in my workspace.

MartyG-RealSense, 2023-07-01:
I have highlighted this issue to my Intel RealSense colleagues on the ROS team to discuss it with them.  Thanks very much for your patience.

hello-chintan, 2023-07-03:
Glad that the bug was found. Thanks for working on it, @MartyG-RealSense!

hello-chintan, 2023-07-04:
The issue is fixed. Closing this issue. Thanks again!

hello-chintan, 2023-07-04:
I can confirm, the issue is now resolved with the new updates. Thanks again!

MartyG-RealSense, 2023-07-04:
As your issue is resolved, I will close it.  Thanks for the confirmation!

closed 2023-07-04 by MartyG-RealSense