realsenseai/realsense-ros #2798: Limiting RGB exposure

john-maidbot, 2023-06-30:
Hello,

My setup is:
D435 camera
LibRealSense v2.50.0
ROS 2 Humble
realsense-ros package built from source

There is currently (As far as i know) no way to limit the exposure of the rgb image when using autoexposure other than increasing the rgb camera fps and setting autoexposure_priority to false.
I am not able to stream the rgb image at a high rate, but i would like the benefits of a low exposure limit (reduced motion blur) while still having autoexposure enabled for when lighitng conditions change slightly.

Is there a way to limit the rgb camera's autoexposure with librealsense atm? If not, is there some technical reason that the rgb autoexposure does not have a limit like the IR autoexposure does? I am considering just implementing a ROS2 node that does this (i.e. controls the exposure from realsense-ros where i can set a limit), but that isnt ideal :sweat_smile: So wanted to ask first.

Thank you for your time. :smiley:

MartyG-RealSense, 2023-06-30:
Hi @john-maidbot  The librealsense SDK, and the ROS wrapper, does have an auto_exposure_limit parameter for the depth sensor that enables an upper bound value to be set for auto-exposure.  This parameter is described by `/camera/stereo_module/auto_exposure_limit`

This parameter is not supported for RGB though.  This may be because the RGB sensor does not provide exposure metadata when auto-exposure is enabled, whilst the depth sensor does.

john-maidbot, 2023-06-30:
Ah ok, thank you for confirming my suspicion. 

It does seem odd though because i wonder how they are implementing the RGB auto_exposure_priority setting without access to the rgb exposure metadata when autoexposure is enabled. :shrug: 

Anyhow, It seems like the host-side auto exposure control is my best bet to achieve an RGB autoexposure limit then :slightly_smiling_face: 

Thanks!

closed 2023-06-30 by john-maidbot