realsenseai/realsense-ros #2803: infra_rgb parameter not working

pavllan90, 2023-07-03:
Hi, i am using d455 camera with 4.51.1 realsense-ros version with LibRealSense v2.54.1 and ROS2 Galactic. 
I am trying to get rgb image from left infra sensor, so I am setting 'infra_rgb' param to True. I am also setting 'enable_infra' param to true. But /camera/infra1/image_rect_raw remains mono and there is no topic /camera/infra/image_rect_raw. It appears, that these both parameters take absolutely no effect on process. There are also no such params in output of commands ros2 params list /camera/camera. I can obtain rgb infra frame only using realsense-viewer.
What can I do to get RGB Infra frame in ROS2 Node?

MartyG-RealSense, 2023-07-03:
Hi @pavllan90  The best match for ROS wrapper version 4.51.1 is librealsense 2.51.1.  The recommended librealsense version to use with a particular wrapper version is usually listed in the release notes for that wrapper.

If you have librealsense 2.54.1 installed then the best match for it is wrapper **4.54.1** 

https://github.com/IntelRealSense/realsense-ros/releases/tag/4.54.1

If infra_rgb and enable_infra are both set to True then `enable_infra1 `and `enable_infra2 `should **not** be enabled and they should be left in their default state of False.

Can you confirm please that you have not set enable_infra1 and enable_infra2 to True, please?  Setting these parameters to True will prevent the infra_rgb 'color from left infrared sensor' stream from working.

pavllan90, 2023-07-04:
Yes, `enable_infra1` and `enable_infra2` are set to False.

MartyG-RealSense, 2023-07-04:
What happens if you launch with **ros2 run** instead of ros2 launch?

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p infra_rgb:=true -p enable_infra:=true`

pavllan90, 2023-07-04:
By default I get these image topics:
`/infra1/image_rect_raw`
`/infra2/image_rect_raw'
`/color/image_raw`
`/depth/image_rect_raw`
because I set `enable_infra*` params in my own launch files. 
If I run command:
`ros2 run realsense2_camera realsense2_camera_node --ros-args -p infra_rgb:=true -p enable_infra:=true -p enable_infra1:=false -p enable_infra2:=false -p enable_color:=false -p enable_depth:=false` I got no Image topics at all. 
By the way, I have installed `4.54.1` wrapper version.

MartyG-RealSense, 2023-07-04:
Do you have the option to upgrade to ROS Humble to test whether it makes a difference?  

Support for Galactic (and also Foxy) is being removed in the next version of the RealSense ROS wrapper as those ROS branches have reached their End of Life (EOL).

https://docs.ros.org/en/galactic/Releases.html

pavllan90, 2023-07-04:
Unfortunately, I have other dependencies that can't be transferred to Humble. But I tried this option, commenting out all my other nodes and Realsense node shows exactly the same behavior with `ros2 run` and `ros2 launch`

MartyG-RealSense, 2023-07-04:
I will highlight your issue to my Intel RealSense colleagues on the ROS team.

MartyG-RealSense, 2023-07-09:
I just wanted to update you that a member of the RealSense ROS team is currently investigating the infra_rgb issue that you reported and agreed that the feature may not be working at present.

Arun-Prasad-V, 2023-07-19:
Hi @pavllan90 ,
A PR for 'Enabling RGB8 Infrared stream' https://github.com/IntelRealSense/realsense-ros/pull/2806 has been merged.
You can now use the infrared stream with RGB8 format by enabling the param: `enable_infra:=true`.

And kindly note that the param `infra_rgb` is not in use for ROS2.

MartyG-RealSense, 2023-07-29:
Hi @pavllan90  Do you have an update about this case that you can provide, please?  Thanks!

pavllan90, 2023-07-31:
It is working now, thanks!
I have one more question. Is there any way to get `left_infra` in `rgb8` format and `right_infra` in any format at the same time? Now it seems impossible even in `realsense-viewer` software. Am I missing something? If not and it is impossible now, do you have any plans for implementing this feature? It could be extremely useful for my project. I can open separate issue for this problem.

MartyG-RealSense, 2023-07-31:
Hi @pavllan90  When color from the left infrared sensor is being used, Y8 infrared from the right-hand sensor is not supported.  I am not aware of any future plans to implement support for it.  The only workaround would be to use the RGB sensor (rgb8 color) and the right-side infrared stream **infra2** simultaneously.

pavllan90, 2023-07-31:
Okay, thank you

closed 2023-07-31 by pavllan90