realsenseai/realsense-ros #2808: How to get camera_color_frame instead of camera_optical_color_frame?

ydzat, 2023-07-07:
os: Ubuntu 22.04
but ros is in a docker: Ubuntu 20.04
 * /rosdistro: noetic
 * /rosversion: 1.16.0

--------------------------------------
I have a D455, I use it combined with rtabmap to generate a .bag file, and then reconstruct the point cloud .ply file, put the ply into Blender and use the /tf information to reconstruct the motion of the virtual camera. In theory, the rendered image generated from Blender should be consistent with the rgb image (for example, if there is a cup in an rgb image, then the cup in the corresponding rendered image should be in the same position). However, the fact is that the position of the object in the rendered image has a certain degree of offset compared to the real position.

[image]

For example, in this image, I merged the rendered image with the rgb image, and set the transparency of the rendered image to 50%, and found that the objects were not aligned. 

--------------------------------------

Related Information:

.bag's tf tree
[image]

Camera's tf tree
[image]

rtabmap rviz looks like:
[image]

Observing the offset direction and the tf tree, I guess that there may be two types of offset:
1. Because the position corresponding to the camera_optical_color_frame used is not the position of the rgb camera, but the position of the left sensor (looking forward from the back).
2. The saved rgb image is the image obtained by the left sensor, not the generated interpolation image (pure guesswork).

So I'd like to start with the first guess. If I can get other tf information, maybe the problem of offset can be solved. However, I am a ros newbie and I have made quite a few attempts in camera driven launch and rtabmap launch, however unsuccessful, so here is my current launch file, please help me, how should I modify them?

handheld_realsense_driver.launch
```
<?xml version="1.0"?>
<launch>

    <arg name="l515" value="false"/>
    <arg name="d435i" value="false"/>
    <arg name="d455" value="true"/>

    <!-- https://desk.zoho.com/portal/dotproduct/en/kb/articles/d455connection -->
    <!-- Make sure USB-C connector at camera is upright for SuperSpeed mode -->
    <group if="$(arg d455)">
        <include file="$(find realsense2_camera)/launch/rs_camera.launch">
            <arg name="clip_distance" value="4.0"/>
            <arg name="align_depth" value="true"/>
            <!-- <arg name="frame_id" value="camera_color_frame"/> -->
            <!-- <arg name="filters" value="decimation,temporal"/> -->
            
            <arg name="unite_imu_method" value="linear_interpolation"/>
            <arg name="enable_gyro" value="true"/>
            <arg name="enable_accel" value="true"/>
            <arg name="enable_sync" value="true"/>
            <arg name="color_width" value="640"/>
            <arg name="color_height" value="480"/>
            <arg name="color_fps" value="30"/>
            <arg name="depth_width" value="640"/>  <!--L515: 1024x768, 640x480, 320x240-->
            <arg name="depth_height" value="480"/>
            <arg name="depth_fps" value="30"/>
            <!-- <param name="camera_info_url" type="string" value="file:///$(find cronos_rtabmap)/launch/ost.yaml"/> -->
            <!-- <remap from="camera_info_url" to="/rec/info" /> -->
         </include>
    </group>

    <node name="imu_filter_node" pkg="imu_filter_madgwick" type="imu_filter_node" output="screen">
        <param name="use_mag" type="bool" value="false"/>
        <param name="publish_tf" type="bool" value="false"/>
        <!-- <param name="publish_tf" type="bool" value="true"/> -->
        <param name="world_frame" type="string" value="enu"/>
        <remap from="/imu/data_raw" to="/camera/imu"/>
        <remap from="/imu/data" to="/rtabmap/imu" />
    </node>

</launch>
```

handheld_mapping.launch
```
<?xml version="1.0"?>
<launch>
    <node pkg="tf" type="static_transform_publisher" name="tf_publisher" args="0 0 0 0 0 0 camera_link camera_color_frame 100" />

    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
        <arg name="rtabmap_args"        value= "--delete_db_on_start --Rtabmap/DetectionRate 2  RGBD/LinearUpdate 0.01  RGBD/AngularUpdate 0.01" />
        <arg name="use_sim_time"        value="false"/>
        <arg name="rtabmapviz"          value="true" />
        <arg name="rviz"                value="false" />
       
        <arg name="depth_topic"         value="/camera/aligned_depth_to_color/image_raw" />
        <arg name="rgb_topic"           value="/camera/color/image_raw" />
        <arg name="camera_info_topic"   value="/camera/color/camera_info" />
        <arg name="depth_camera_info_topic" value="/camera/aligned_depth_to_color/camera_info" />

        <arg name="approx_sync"         value="false" />
        <arg name="wait_imu_to_init"    value="true"  />
        <arg name="imu_topic"           value="/rtabmap/imu" />
        <!-- <arg name="imu_topic"           value="/camera/imu" /> -->

        <arg name="frame_id"            value="camera_link"/>
        <arg name="map_frame_id"        value="map"/>
    </include>
</launch>

```

MartyG-RealSense, 2023-07-07:
Hi @ydzat  The Intel RealSense team member who created the ROS1 wrapper describes the difference between camera_color_frame and camera_optical_color_frame at https://github.com/IntelRealSense/realsense-ros/issues/743

After the RealSense launch (rs_camera.launch) has completed, you should be able to retrieve the **camera_color_frame** TF values by inputting the **echo** instruction below.

`rosrun tf tf_echo /camera_link /camera_color_frame`

When importing depth information into Blender via a ply, the measurement scale of Blender (or any other 3D modelling tool that a RealSense ply is imported into) should be set to the same **depth unit scale** that RealSense uses, otherwise the ply may be mis-scaled after import.  The default depth unit scale of the D455 camera is **0.001**.

ydzat, 2023-07-07:
> Hi @ydzat The Intel RealSense team member who created the ROS1 wrapper describes the difference between camera_color_frame and camera_optical_color_frame at #743
>
> After the RealSense launch [...]

Thank you. I think what I need (probably) is not camera_color_frame anymore, instead, it is a frame corresponding to the rgb camera located to the right of the camera_link. 
[image]

Here's a image depicting a few different coordinate systems in D455 (I don't know the source of this image)

As you can see, the camera_link is on the far left, and as the image in my previous problem description shows, "the rendered image is slightly offset relative to the rgb image", so I'm guessing that perhaps the camera_color_frame in this image is the one I should be getting.

So how do I subscribe to that coordinate information in the launch file?

MartyG-RealSense, 2023-07-07:
**camera_link** is at the center-line of the left infrared sensor, as this is the 0,0,0 origin coordinate of depth.  This is on the right side of the camera when viewing it from the front like in the image above.  

The RGB sensor is offset horizontally from the left infrared sensor's position.  On the D455 model it is on the opposite side of the camera beside the right infrared sensor, second to the left when viewing the camera from the front).

**base_link** corresponds to the tripod screw-thread hole on the bottom of the camera.

My knowledge of TF is admittedly limited.  In the guide at the link below for connecting a D455 to ROS though the author seems to be having a similar situation to yours and states the following:

********

I was looking for TF from camera_color_frame to base_link, and found that the point group coordinate system is camera_depth_optical_frame, and the xyz direction is different after calculation (I didn't notice the difference, I was confused) so x = t_z, y = - t_x, z = - t_y

************

https://www.hackster.io/Elephant-Robotics-Official/operate-mycobot-with-the-spatial-recognition-of-d455-7f298e#toc-6----connecting-mycobot-to-realsense-d455-using-ros-12

The author's solution was apparently to rewrite the temporary value in a file called **tf_broadcaster.cpp**, though this may not be applicable to your particular project. 

```
# Put the obtained values in t_x,y,z and theta_1,_2,_3 respectively 
    transformStamped.transform.translation.x = t_z; 
    transformStamped.transform.translation.y = -t_x; 
    transformStamped.transform.translation.z = -t_y; 
     
    tf2::Quaternion q; 
    q.setEuler(theta_1, theta_2, theta_3); 
```

This filename is referenced in the transformation tutorial below.

https://husarion.com/tutorials/ros-tutorials/6-transformation-in-ROS/#broadcasting-transformation

Various tutorials online describe that tf_broadcaster.cpp is a file that ROS users can create themselves in their catkin workspace's **/src** folder.

ydzat, 2023-07-08:
> **camera_link** is at the center-line of the left infrared sensor, as this is the 0,0,0 origin coordinate of depth. This is on the right side of the camera when viewing it from the front like in the image above.
> 
> The RGB sensor is offset horizontally from the left infrared sensor's position. On the D455 model it is on the opposite side of the camera beside the right infrared sensor, second to the left when viewing the camera from the front).
> 
> **base_link** corresponds to the tripod screw-thread hole on the bottom of the camera.
> 
> My knowledge of TF is admittedly limited. So in regard to your question "how do I subscribe to that coordinate information in the launch file", I unfortunately do not know.
> 
> In the guide at the link below for connecting a D455 to ROS though, the author seems to be having a similar situation to yours and states the following:
> 
> I was looking for TF from camera_color_frame to base_link, and found that the point group coordinate system is camera_depth_optical_frame, and the xyz direction is different after calculation (I didn't notice the difference, I was confused) so x = t_z, y = - t_x, z = - t_y
> 
> https://www.hackster.io/Elephant-Robotics-Official/operate-mycobot-with-the-spatial-recognition-of-d455-7f298e#toc-6----connecting-mycobot-to-realsense-d455-using-ros-12
> 
> The author's solution was apparently to rewrite values in a file called **tf_broadcaster.cpp**, though this may not be applicable to your particular project.
> 
> ```
> # Put the obtained values in t_x,y,z and theta_1,_2,_3 respectively 
>     transformStamped.transform.translation.x = t_z; 
>     transformStamped.transform.translation.y = -t_x; 
>     transformStamped.transform.translation.z = -t_y; 
>      
>     tf2::Quaternion q; 
>     q.setEuler(theta_1, theta_2, theta_3); 
> ```
> 
> This filename is referenced in the transformation tutorial below.
> 
> https://husarion.com/tutorials/ros-tutorials/6-transformation-in-ROS/#broadcasting-transformation
> 
> Various tutorials online describe that tf_broadcaster.cpp is a file that ROS users can create themselves in their catkin workspace's **/src** folder.
> 
> I apologize that I could not be of more assistance on this particular question.

I've noticed issues with different xyz orientations, but apparently that's not what's causing the image to appear offset...
[image]

I use rviz to view the tf after the subscriber's launch starts, as shown in the image. The one on the left is camera_link, the one on the right is camera_color_optical_frame. (camera_color_frame actually coincides with camera_color_optical_frame origin, the direction is different, as you said). There are several other frames in rviz's tf, but the positions are between camera_color_optical_frame and camera_link. This shows that the subscribed camera_color_optical_frame should be exactly the frame I need in theory, but there is a problem when writing the .bag, which causes the written tf information to be incorrect. In order to verify this guess, I recorded a small .bag file. 

1. Start the camera driver
2. Start the launch of rtabmap
3. `rostopic echo /tf`
```
frame_id: map
child_frame_id: odom

frame_id: odom
child_frame_id: camera_link
```

4. python script, used to save as .bag file
5. `rosrun rqt_tf_tree rqt_tf_tree` & `rosbag play rgbdp_data.bag`
[image]

6. `rostopic echo /tf` & `rosbag play rgbdp_data.bag`
```
frame_id: "map"
child_frame_id: "camera_color_optical_frame"
```

You can see that there is no camera_color_optical_frame in the 3 step, so I guess there is a problem in this step?

The launch file of the second step is:
```
<?xml version="1.0"?>
<launch>
    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
        <arg name="rtabmap_args"        value= "--delete_db_on_start --Rtabmap/DetectionRate 2  RGBD/LinearUpdate 0.01  RGBD/AngularUpdate 0.01" />
        <arg name="use_sim_time"        value="false"/>
        <arg name="rtabmapviz"          value="true" />
        <arg name="rviz"                value="false" />
       
        <arg name="depth_topic"         value="/camera/aligned_depth_to_color/image_raw" />
        <arg name="depth_camera_info_topic" value="/camera/aligned_depth_to_color/camera_info" />
        <!-- <arg name="depth_topic"         value="/camera/depth/image_raw" />
        <arg name="depth_camera_info_topic" value="/camera/depth/camera_info" /> -->

        <arg name="rgb_topic"           value="/camera/color/image_raw" />
        <arg name="camera_info_topic"   value="/camera/color/camera_info" />

        <arg name="approx_sync"         value="false" />
        <arg name="wait_imu_to_init"    value="true"  />
        <arg name="imu_topic"           value="/rtabmap/imu" />
        <!-- <arg name="imu_topic"           value="/camera/imu" /> -->

        <arg name="frame_id"            value="camera_link"/>
        <arg name="map_frame_id"        value="map"/>
    </include>
</launch>
```

MartyG-RealSense, 2023-07-08:
Rather than use a Python script to record the TF to a bag file, a more common approach may be to use the instruction `rosbag record /tf` to record the TF to a bag with ROS' own rosbag record tool.

ydzat, 2023-07-09:
> Rather than use a Python script to record the TF to a bag file, a more common approach may be to use the instruction `rosbag record /tf` to record the TF to a bag with ROS' own rosbag record tool.

right....but for reasons of my project, I only have the option to use python for writing. When I used realsense + rtabmap to record before, I noticed that the terminal corresponding to the launch of the driver output several warnings, so I wondered if this was the problem, so I completely reinstalled it according to the realsense tutorial--finally, Now there are no errors or warnings when running the camera, however the result remains the same. 

Here are the steps I took to install the driver:
1. For hardware compatibility reasons, my laptop can only install Ubuntu 22.04, and for realsense kernel patching requirements, I downgrade the kernel to 5.15.0-75-generic;
2. But I need ROS Noetic as a development environment, so Docker is the final choice;
3. Docker installs librealsense/librealsense:latest, and the container is named ros04
4. Install ros noetic in ros04
5. Install realsense-ros, because the driver version in librealsense/librealsense:latest is lower, so realsense-ros-2.2.24 must be downloaded;
6. According to the installation steps, `git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`` is skipped;
7. `root@ros03host:/# roslaunch cronos_rtabmap handheld_realsense_driver.launch`

``` 
... logging to /root/.ros/log/a9690b02-1e54-11ee-bc29-0242acac0165/roslaunch-ros03host-220.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ros03host:33843/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: 4.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
[... 363 lines cut ...]
[ INFO] [1688906081.134820019]: Odom: quality=75, std dev=0.018262m|0.020880rad, update time=0.036654s
[ INFO] [1688906081.168953847]: Odom: quality=80, std dev=0.017897m|0.023049rad, update time=0.033488s
[ INFO] [1688906081.206107999]: Odom: quality=87, std dev=0.026377m|0.021630rad, update time=0.036493s
[ INFO] [1688906081.242651338]: Odom: quality=83, std dev=0.020107m|0.019282rad, update time=0.035908s
[ INFO] [1688906081.278485159]: Odom: quality=88, std dev=0.024180m|0.021909rad, update time=0.035069s
[ INFO] [1688906081.311256849]: Odom: quality=97, std dev=0.016986m|0.022044rad, update time=0.032091s
[ INFO] [1688906081.344587135]: Odom: quality=93, std dev=0.016652m|0.024041rad, update time=0.032689s
[ INFO] [1688906081.379352647]: Odom: quality=99, std dev=0.017517m|0.019282rad, update time=0.034122s
[ INFO] [1688906081.417273250]: Odom: quality=77, std dev=0.014180m|0.022308rad, update time=0.037299s
[ INFO] [1688906081.451039870]: Odom: quality=92, std dev=0.015356m|0.020880rad, update time=0.033181s
[ INFO] [1688906081.485669638]: Odom: quality=88, std dev=0.012739m|0.023506rad, update time=0.033967s
[... cut ...]
[ INFO] [1688906081.689800207]: Odom: quality=88, std dev=0.016959m|0.026054rad, update time=0.031780s
[ INFO] [1688906081.722289341]: Odom: quality=94, std dev=0.022223m|0.030639rad, update time=0.031873s
[ INFO] [1688906081.756606047]: Odom: quality=91, std dev=0.021259m|0.019282rad, update time=0.033681s
...
```

Can you help me to find out what I have done wrong(Put ply in Blender, use the virtual camera to reconstruct the path based on /tf, the rendered image is offset from the actual rgb image) from the above information?

MartyG-RealSense, 2023-07-09:
I cannot see any problems in your RealSense ROS launch logs.  However, I note from the logs that you ae using firmware **5.15.0.2** with SDK 2.44.0.  The 5.15.0.2 firmware should only be used with the current latest SDK **2.54.1**.  Using this firmware with an older SDK may result in errors.

For the 2.44.0 SDK, the old firmware version **5.12.12.0** from March 2021 is the recommended version to use with it.

ydzat, 2023-07-09:
> I cannot see any problems in your RealSense ROS launch logs and your installation procedure seems fine. However, I note from the logs that you are using camera firmware driver version **5.15.0.2** with SDK 2.44.0. The 5.15.0.2 firmware should only be used with the current latest SDK **2.54.1**. Using this firmware with an older SDK may result in errors.
> 
> For the 2.44.0 SDK, the old firmware version **5.12.12.0** from March 2021 is the recommended version to use with it.

Thank you very much for your help, at least now I can say that the problem is not with realsense. I should have asked over at rtabmap for example.

MartyG-RealSense, 2023-07-09:
You are very welcome.  :)

ydzat, 2023-07-10:
> You are very welcome. :)

Sorry, I have problems with the realsense driver:
As mentioned before, I am now using the Docker version of the realsense SDK, which is version 2.44.0, and its realsense-ros counterpart is version 2.2.24, which does not contain the urdf file for D455. So, is there any way I can upgrade the realsense SDK in Docker (to the latest version)?

MartyG-RealSense, 2023-07-10:
There should not be a problem with changing the librealsense version installed in your Docker container.  However, if you upgrade to a different librealsense version such as 2.50.0 or 2.51.1 then you will also have to rebuild the ROS wrapper and use a newer wrapper version (2.3.2). As you are using ROS1 Noetic you will not be able to upgrade to a newer SDK version such as 2.53.1 or 2.54.1 as the ROS1 wrapper has ceased development and was not updated for use with SDKs beyond 2.51.1.

**********

If you build the wrapper from source code then an alternative to upgrading librealsense that you could explore would be to download the **_d455_urdf.xacro** URDF file from the **realsense2_description/urdf** folder of the most recent ROS1 wrapper (2.3.2) and copy them into the description folder of wrapper 2.2.24.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy/realsense2_description/urdf

ydzat, 2023-07-10:
> There should not be a problem with changing the librealsense version installed in your Docker container. However, if you upgrade to a different librealsense version such as 2.50.0 or 2.51.1 then you will also have to rebuild the ROS wrapper afterwards and use a newer wrapper version (2.3.2). As you are using ROS1 Noetic you will not be able to upgrade to a newer SDK version such as 2.53.1 or 2.54.1 as the ROS1 wrapper has ceased development and was not updated for use with SDKs beyond 2.51.1.
> 
> If you build the wrapper from source code then an alternative to upgrading librealsense that you could explore would be to download the **_d455_urdf.xacro** URDF file from the **realsense2_description/urdf** folder of the most recent ROS1 wrapper (2.3.2) at the link below and copy it into the realsense2_description folder of wrapper 2.2.24.
> 
> https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy/realsense2_description/urdf

According to the [SDK build guidelines](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md), Ubuntu's kernel needs to be patched before building the SDK, i.e. the following code needs to be executed:

>Run Intel Realsense permissions script from librealsense root directory:
>`./scripts/setup_udev_rules.sh`
>
>Notice: One can always remove permissions by running: ./scripts/setup_udev_rules.sh --uninstall
>
>Build and apply patched kernel modules for:
>
>**Ubuntu 20/22 (focal/jammy) with LTS kernel 5.13, 5.15 **
>`./scripts/patch-realsense-ubuntu-lts-hwe.sh`

Do I still need to perform this step if I choose to upgrade the current version of the SDK?

MartyG-RealSense, 2023-07-10:
If you have not changed your kernel version from the **5.15.0-75-generic** kernel that you installed and patched earlier then you should not need to patch the kernel again when installing a new librealsense version.

If your preference would be to use a more modern kernel than 5.15 then you can do so if librealsense is built from source code with CMake with the flag **-DFORCE_RSUSB_BACKEND=TRUE** included in the CMake build instruction.  An RSUSB build of librealsense bypasses the kernel, so that it is not dependent on Linux versions or kernel versions and does not require a kernel patch script to be applied.

ydzat, 2023-07-10:
@MartyG-RealSense Sorry, I had to reopen this, I still have some questions.

Firstly, I always start the Stereo Module first, then the RGB Camera. Here are two screenshots:

[image]
In the right rgb image, the eyedrop bottle in the centre can be seen to overlap with the side of the thin wall behind it. However, on the left, the silhouette of the eyedrop bottle in the depth image  is clearly to the right of the thin wall.

[image]
In this image, even though you can see the eyedropper bottle in the middle being correctly aligned to the depth image, look closely, there is still a small eyedropper bottle in between the cup and the eyedropper bottle in the middle, and I think this is where the error lies.

[image]
Here's a screenshot of the rtabmap, you can see that the eyedropper bottle fails to line up with the thin wall on the back side, what causes this?

MartyG-RealSense, 2023-07-10:
It's no trouble at all.

Looking at the image, it appears to me that the depth image may be showing the wide section of the wall rather than its thin side, with this thin edge not showing up on the depth image.

[image]

ydzat, 2023-07-10:
> It's no trouble at all.
> 
> Looking at the image, it appears to me that the depth image may be showing the wide section of the wall rather than its thin side, with this thin edge not showing up on the depth image.

you are right. 
I have written `<arg name="align_depth" value="true"/>` in the launch file, does this correspond to the second image just now? And as shown in that image, showing the large eyedrops is supposed to be the correct result?

MartyG-RealSense, 2023-07-10:
When align_depth is True, the depth image will be aligned to the color image, resulting in a image resembling the one below that you provided.

[image]

When you ask if the 'large eyedrops' are correct, do you mean the large black shadow surrounding the eyedropper bottle, please?

ydzat, 2023-07-10:
> When align_depth is True, the depth image will be aligned to the color image, resulting in a image resembling the one below that you provided.

What I expect, eyedrop bottle should be in the green box. The red box is in the wrong position.
[image]

[image]

I.e., the eyedrop in the red box in the 3D Map on the right side of this image is located in the wrong location, it should be located in the green box

ydzat, 2023-07-10:
> Okay, I see what you mean about the alignment of the red-boxed dropper to the edge of the wall compared to the green-boxed ones. The same seems to be true of the cup beside the dropper compared to the positions of the objects on the shelves and door behind. It looks as though the green-box dropper images are RGB images and the red-box dropper is an aligned depth and RGB image. It is as though the foreground of the image is horizontally offset from the background.
> 
> Please try using the instructions at [IntelRealSense/librealsense#10182 (comment)](https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487) to reset the camera to its factory-new calibration settings in the RealSense Viewer tool to see whether it improves the aligned image.

I ticked "I know what I'm doing" and clicked reset. But the write button is greyed out and cannot be clicked.

After upgrading the SDK to the latest version and installing the corresponding realsense-ros, I ran the programme again, and after careful observation, I found that the range of the "points" for generating the 3D Map is smaller than that of the rgb images, as shown in the figure, the part of the two rgb images on the left side, which I framed with a red line, is used to build the 3D map. This does not match the size of the rgb.

[image]

My launch file:
```
<?xml version="1.0"?>
<launch>
    <arg name="l515" value="false"/>
    <arg name="d435i" value="false"/>
    <arg name="d455" value="true"/>
    <group if="$(arg d455)">
        <include file="$(find realsense2_camera)/launch/rs_camera.launch">
            <arg name="clip_distance" value="4.0"/>
            <arg name="align_depth" value="true"/>
            <arg name="unite_imu_method" value="linear_interpolation"/>
            <arg name="enable_gyro" value="true"/>
            <arg name="enable_accel" value="true"/>
            <arg name="enable_sync" value="true"/>

            <arg name="infra_width"         default="640"/>
            <arg name="infra_height"        default="480"/>
            <arg name="enable_infra"        default="true"/>

            <arg name="color_width" value="640"/>
            <arg name="color_height" value="480"/>
            <arg name="color_fps" value="30"/>
            <arg name="depth_width" value="640"/>
            <arg name="depth_height" value="480"/>
            <arg name="depth_fps" value="30"/>
         </include>
    </group>

    <node name="imu_filter_node" pkg="imu_filter_madgwick" type="imu_filter_node" output="screen">
        <param name="use_mag" type="bool" value="false"/>
        <param name="publish_tf" type="bool" value="false"/>
        <param name="world_frame" type="string" value="enu"/>
        <remap from="/imu/data_raw" to="/camera/imu"/>
        <remap from="/imu/data" to="/rtabmap/imu" />
    </node>
</launch>
```

MartyG-RealSense, 2023-07-10:
You must enable the depth stream first before opening the camera data window in order for the Write Table button to be unlocked.

**infra_rgb** could likely be set to False instead of True as it is not related to depth to color alignment. Instead, it generates an RGB stream from the left infrared sensor, which is unnecessary if you already have the color stream enabled.

ydzat, 2023-07-10:
> You must enable the depth stream first before opening the camera data window in order for the Write Table button to be unlocked.
>
> **infra_rgb** could likely be set to False instead of True as it is [...]

Ok, I did this and successfully reset the D455.
I deleted the lines in the startup file.
Then I ran the programme, but the results remained as before.

MartyG-RealSense, 2023-07-10:
As you are using rtabmap. how does it look if you launch with the **opensource_tracking.launch** launch file instead of rs_camera.launch.  The opensource_tracking.launch file is suited for use with rtabmap and imports the settings from rs_camera.launch

MartyG-RealSense, 2023-07-11:
You could test generating a simple pointcloud in the ROS wrapper with:

`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#point-cloud

If the pointcloud appears correct then the problem is likely to be with rtabmap rather than the ROS wrapper.

ydzat, 2023-07-11:
> You could try generating a simple depth-color aligned pointcloud in the ROS wrapper with:
> 
> `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`
> 
> https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#point-cloud
> 
> If the pointcloud's depth-color alignment appears correct then the problem is likely to be either with the wrapper's align_depth function or with rtabmap.

the pointcloud's depth-color alignment appears correct. I guess the problem really isn't with the camera itself. One last question, what is the best distance to use the D455? Since I'm trying to record indoor objects and the objects are small, like apples bananas something else, what should I set the rgb/depth/fov to to get the best results? As well, how far should I hold the camera from these objects?

MartyG-RealSense, 2023-07-11:
A D455 will have a minimum depth sensing distance of around 0.4 meters / 40 cm, with the depth image progressively breaking up when moving the camera closer to the object than that minimum.

A method for reducing the minimum distance in the ROS wrapper is to define a **disparity shift** value in a json camera configuration file and then configure the roslaunch instruction or edit the launch file to load that json during launch to apply the new value.  More information about this can be found at https://github.com/IntelRealSense/realsense-ros/issues/2012#issuecomment-890390205

With the RGB image, having the camera close to an object usually will not matter.

ydzat, 2023-07-11:
> A D455 will have a minimum depth sensing distance of around 0.4 meters / 40 cm, with the depth image progressively breaking up when moving the camera closer to the object than that minimum.
> 
> A method for reducing the minimum distance in the ROS wrapper is to define a **disparity shift** value in a json camera configuration file and then configure the roslaunch instruction or edit the launch file to load that json during launch to apply the new value. More information about this can be found at [#2012 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/2012#issuecomment-890390205)
> 
> With the RGB image, having the camera close to an object usually will not matter and it will not require a special setting.

Thank you very much for your continued help!