realsenseai/realsense-ros #2809: displaying range information from data point

Daya-AWR1642, 2023-07-10:
[image]

Hi, I need one help in displaying range information of lidar and camera .

Daya-AWR1642, 2023-07-11:
Ohh I'm sorry I forgot to give background information.

I'm working on RLC (Radar,camera,lidar) sensor fusion using ROS, I need range information from each sensor.
I knew how to get range with radar(TI's AWR1642) but not with L515 LiDAR . I need range information link 

[image]

MartyG-RealSense, 2023-07-11:
Hi @Daya-AWR1642  https://github.com/IntelRealSense/realsense-ros/issues/1790#issuecomment-814097810 provides information about ways to retrieve depth-stream distance in the ROS1 wrapper.

Daya-AWR1642, 2023-07-18:
I need one help. I wanted to know where data of camera and lidar is getting stored during real-time running. I need to perform some analysis from that data

MartyG-RealSense, 2023-07-18:
RealSense camera data is stored in the computer's memory during real-time running.  Specifically, it is held in a **frame queue** that by default can hold 16 frames of each type of stream that is enabled.  The queue is like a constantly moving conveyor belt where old frames that reach the end of the queue are dropped out of the queue and lost as new frames enter the queue.  

It is possible to increase the **frame queue capacity** to have a greater number of frames in the queue simultaneously, but this uses a larger amount of the computer's available memory.

I assume that your non-L515 lidar's frames are also held in the computer's memory but not in the RealSense frame queue with the other RealSense camera frames.

Daya-AWR1642, 2023-07-20:
ohh Thanks for your information.I wanted to know how can I access that data(since I'm undergraduate student  I don't have that much idea )

MartyG-RealSense, 2023-07-20:
As you are using the ROS1 wrapper, you could obtain the distance in meters of the coordinate at the center of the L515 camera's view by using a ROS **node script** provided by the RealSense ROS wrapper.  It uses intrinsics from "/camera/depth/camera_info" topic to calculate the 3D position of a pixel whose range is obtained from the depth image, received from the "/camera/depth/image_rect_raw" topic.

https://github.com/IntelRealSense/realsense-ros/blob/ros1-legacy/realsense2_camera/scripts/show_center_depth.py

Daya-AWR1642, 2023-07-22:
Thanks a lot @MartyG-RealSense  this gave some good idea.
But one doubt is, 
`    depth_image_topic = '/camera/depth/image_rect_raw'  `
`   depth_info_topic = '/camera/depth/camera_info'             `

in my case it topics are showing,
` /camera/color/image_raw `
` /camera/depth/image_rect_raw `

am I in right direction ? 

do I have to customize the code corresponding to L515 ?

MartyG-RealSense, 2023-07-22:
So the topic `/camera/depth/camera_info` is not listed if you use the **rostopic list** command to generate a list of the ROS topics that are available from your L515?   An official Intel list of supported topics on L515 at the link below shows that the camera should be able to provide the /camera/depth/camera_info topic.

https://dev.intelrealsense.com/docs/ros-wrapper#l500-lidar-depth-cameras

MartyG-RealSense, 2023-07-27:
Hi @Daya-AWR1642  Do you require further assistance with this case, please?  Thanks!

Daya-AWR1642, 2023-07-27:
No, Thank you

closed 2023-07-27 by Daya-AWR1642