realsenseai/realsense-ros #2814: multiple camera issue

taeyeonggim, 2023-07-18:
ASAP 💨 
--- 
Whenever I turn on the 3 cameras with the launch file below, the computer freezes, or 
only 2 of the 3 cameras publish rostopic, and the other one doesn't publish topic even 
though the rosnode is alive. Can I get some help ASAP?

For reference, the hardware spec is RTX3080 on an i9 10th Gen.

```
<launch>
  <arg name="serial_no_camera1"         default=""/> 			<!-- Note: D435i -->
  <arg name="serial_no_camera2"    	default=""/> 			<!-- Note: D415 -->
  <arg name="serial_no_camera3"         default=""/>      <!-- Note: L515 -->
  <arg name="camera1"              			default="camera"/>		<!-- Note: Replace with camera name -->
  <arg name="camera2"              			default="camera01"/>		<!-- Note: Replace with camera name -->
  <arg name="camera3"                   default="camera02"/>
  <arg name="tf_prefix_camera1"         default="$(arg camera1)"/>
  <arg name="tf_prefix_camera2"         default="$(arg camera2)"/>
  <arg name="tf_prefix_camera3"         default="$(arg camera3)"/>
  <arg name="initial_reset"             default="true"/>

  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>

  <!-- Camera device specific arguments -->

  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"       default="true"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="400"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>

  <arg name="enable_pointcloud"   default="true"/>
  <arg name="enable_sync"         default="true"/>
  <arg name="align_depth"         default="true"/>
  <arg name="filters"             default="temporal"/>
  <arg name="unite_imu_method"    default=""/>

  <!-- rgbd_launch specific arguments -->

  <!-- Arguments for remapping all device namespaces -->
  <arg name="rgb"                             default="color" />
  <arg name="ir"                              default="infra" />
  <arg name="depth"                           default="depth" />
  <arg name="depth_registered_pub"            default="depth_registered" />
[... 238 lines cut ...]
      </node>

      <!-- Publish registered XYZRGB point cloud with software registered input -->
      <node pkg="nodelet" type="nodelet" name="points_xyzrgb_sw_registered"
            args="load depth_image_proc/point_cloud_xyzrgb $(arg manager) $(arg bond)" respawn="$(arg respawn)">
        <remap from="rgb/image_rect_color"        to="$(arg rgb)/image_rect_color" />
        <remap from="rgb/camera_info"             to="$(arg rgb)/camera_info" />
        <remap from="depth_registered/image_rect" to="$(arg depth_registered_filtered)/sw_registered/image_rect_raw" />
        <remap from="depth_registered/points"     to="$(arg depth_registered)/points" />
      </node>
    </group>

    <group if="$(eval depth_registered_processing and hw_registered_processing)">
      <!-- Publish registered XYZRGB point cloud with hardware registered input (ROS Realsense depth alignment) -->
      <node pkg="nodelet" type="nodelet" name="points_xyzrgb_hw_registered"
            args="load depth_image_proc/point_cloud_xyzrgb $(arg manager) $(arg bond)" respawn="$(arg respawn)">
        <remap from="rgb/image_rect_color"        to="$(arg rgb)/image_rect_color" />
        <remap from="rgb/camera_info"             to="$(arg rgb)/camera_info" />
        <remap from="depth_registered/image_rect" to="$(arg depth_registered)/image_raw" />
        <remap from="depth_registered/points"     to="$(arg depth_registered_pub)/points" />
      </node>
    </group>
  </group>
</launch>
```

MartyG-RealSense, 2023-07-18:
Hi @taeyeonggim  As you are using the **align_depth** parameter and not **align_depth.enable**, it looks as though you are using the ROS1 wrapper (Kinetic, Melodic, Noetic).  Can you confirm if you are using ROS1 please?

If you are using ROS1, which version of the RealSense SDK (librealsense) and RealSense ROS wrapper are you using?  Because development of the ROS1 wrapper has ceased, the most recent librealsense / wrapper combination that should be used with it is librealsense 2.51.1 and wrapper 2.3.2 as the ROS1 wrapper has not been updated for newer librealsense versions such as 2.53.1 and 2.54.1.

Can three cameras be used with the ROS1 wrapper if they are launched from three separate ROS terminals instead of with a single launch, as described at the ROS wrapper documentation section linked to below.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#work-with-multiple-cameras

taeyeonggim, 2023-07-19:
@MartyG-RealSense, Thank you for your quick reply. 👍 
I'm currently using ROS1 melodic.  And librealsense, wrapper are using the most recent versions such as 2.51.1 and 2.3.2.

I am having issues with running 3 at once on ROS1, but not when running them one after the other. 

Is there any way to get around this error by running them all at once?

MartyG-RealSense, 2023-07-19:
Have you tried the ROS1 launch file **rs_multiple_devices.launch** which supports three cameras, please?

https://github.com/IntelRealSense/realsense-ros/blob/ros1-legacy/realsense2_camera/launch/rs_multiple_devices.launch

roslaunch realsense2_camera rs_multiple_devices.launch serial_no_camera1:=<serial number of the first camera> serial_no_camera2:=<serial number of the second camera> serial_no_camera3:=<serial number of the third camera>

The ROS2 wrapper has its own multiple camera launch file called **rs_multi_camera_launch.py**, though this one is designed for only two cameras.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/launch/rs_multi_camera_launch.py

taeyeonggim, 2023-07-19:
I've already tried running rs_multiple_devices.launch, and it didn't fix the topic not coming up. 
ROS2 also doesn't officially provide launch files for more than 3 cameras. :cry:.

so I guess the only way to do that is to run the cameras sequentially, right?

Is librealsense the reason this is happening?

MartyG-RealSense, 2023-07-19:
The librealsense SDK is not a limitation on the number of cameras, as it is possible to list as many as 20 cameras in the RealSense Viewer (though not have 20 cameras enabled simultaneously due to the limitations of a computer's resources).  It can certainly have four cameras active simultaneously with a poerful computer specification such as an i7 processor.  Your i9 computer is easily a strong enough specification to process three cameras.

Problems with multiple camera setups do seem to occur more often in the ROS wrapper than they do in librealsense, such as the case at https://github.com/IntelRealSense/realsense-ros/issues/1515#issuecomment-735603814 where six D435i cameras were being used and at least one would not publish topics.

If the ROS1 wrapper is being used and it was installed from packages with the wrapper's **Method 1** instructions then there is the potential for multicam problems as the ROS1 packages are based on the **RSUSB backend**, which is not ideal for use with multiple cameras.

closed 2023-08-02 by MartyG-RealSense