realsenseai/realsense-ros #2816: [ROS2] Realsense Camera Node Error

jesusramondovale, 2023-07-18:
I receive the following error whily executing: 

`ros2 run realsense2_camera realsense2_camera_node` 

on my SBC **Debix Model A (2GB RAM**) with **Ubuntu 22.04 LTS (Kernel: 5.15.71) and ROS2 Humble Hawksbill** installed on it. 

```
[INFO] [1689686546.852992327] [camera]: RealSense ROS v4.54.1
[INFO] [1689686546.853310713] [camera]: Built with LibRealSense v2.54.1
[INFO] [1689686546.853378966] [camera]: Running with LibRealSense v2.54.1
 18/07 15:22:26,882 WARNING [281472727836896] (d400-factory.cpp:1195) DS5 group_devices is empty.
 18/07 15:22:27,259 ERROR [281472727836896] (librealsense-exception.h:52) xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
 18/07 15:22:27,283 WARNING [281472727836896] (rs.cpp:312) null pointer passed for argument "device"
[WARN] [1689686547.284061376] [camera]: Device 1/3 failed with exception: xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument

```

I've installed "succesfully" (no problems appeared) both the SDK and  RealSense libs (and dependencies).

MartyG-RealSense, 2023-07-18:
Hi @jesusramondovale  This type of warning is typically related to a conflict with the kernel.   Which method did you use to install the librealsense SDK, please?

1.  Built with packages using apt-get install
2. Built from source code with CMake
3. Built from source code with CMake and with the build flag **-DFORCE_RSUSB_BACKEND=TRUE** included in the CMake build instruction

If methods 1 and 3 are used then there is no need to apply a kernel patch script to patch the kernel.

If method 2 is used then a kernel patch script called **patch-realsense-ubuntu-lts-hwe.sh** should be run before librealsense is built in order to patch kernel 5.15.

***********

Is the ROS wrapper able to launch without this warning if instead of using ros2 run, you use ros2 launch?

`ros2 launch realsense2_camera rs_launch.py`

jesusramondovale, 2023-07-19:
Hi! Thank you for your response!

I've used the method number 1: install with apt-get, but.. the Linux Image that I'm using is not the "original" Ubuntu 22.04, it's the DEBIX image they provide at https://debix.io/Software/download.html (Ubuntu 22.04 - Model A&B)
________________________________________________________________________________________________________________________________________

The same occurs if I launch with ros2 launch instead of ros2 run

MartyG-RealSense, 2023-07-19:
librealsense installation method **1.  Built with packages using apt-get install** does not need to have the kernel patched as the kernel patch is already included in the packages.  So if you have previously installed librealsense from packages and also applied a patch script to the kernel then this could be a potential cause of problems.

If you are using a non-standard Linux version then building librealsense from source code with the **RSUSB backend** installation method and then building the ROS2 wrapper from source afterwards may work better for you than the alt-get package installation.  This is because an RSUSB-based build of librealsense is not dependent on Linux versions or kernel versions and does not require kenel patching.

jesusramondovale, 2023-07-20:
Hi! 
 
Could you please tell me exactly what must I do to build from source with "the RSUSB backend installation method", or where can I find some information about that?  Thank you so much!

MartyG-RealSense, 2023-07-20:
Instructions for using the **CMake** tool to build the SDK from source code with the RSUSB backend method can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

jesusramondovale, 2023-07-21:
It did literally the same
Builded using the method of your comment in a thread, it gave the same error message as I mentioned in [message #1]

jesusramondovale, 2023-07-21:
Hi again! The librealsense was succesfully installed using the method you mentioned, but now I can't execute any command, what must I do?

MartyG-RealSense, 2023-07-21:
If you earlier installed the SDK from packages then I recommend removing it from the computer by inputting the uninstall instruction below into the Ubuntu terminal.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

After that, use the RSUSB installation method at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692 again and ensure that you do **not** run a kernel patch script before or after performing the RSUSB installation.

jesusramondovale, 2023-07-24:
Hi! I've removed the SDK installed from packages as you said, and I re-installed the RSUSB using method you mention at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692, but I'm getting the output: 
`cmake: /usr/lib/libcurl.so.4: no version information available (required by cmake)`

MartyG-RealSense, 2023-07-24:
Hi @jesusramondovale  It appears that the build is failing on the error **Performing Test SUPPORTS_CXX14 - Failed**.  The current latest version 2.54.1 of librealsense introduced a change that requires C++14 support instead of the C++11 supported in versions before 2.54.1.  

You could try inputting into the CMake command line interface the instruction below:

**"set(CMAKE_CXX_STANDARD 14)"**

jesusramondovale, 2023-07-24:
Hi! Removed the /build content, added the set() you mentioned at start of CMake/lrs_macros.cmake file, but nothing changed :(

```
debix@DEBIX26:~/librealsense-2.54.1/build$ cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
cmake: /usr/lib/libcurl.so.4: no version information available (required by cmake)
-- The CXX compiler identification is unknown
-- The C compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.54.1
-- Setting Unix configurations
-- Building libcurl enabled
-- Found OpenSSL: /usr/lib/aarch64-linux-gnu/libcrypto.so (found version "3.0.2")  
-- Performing Test SUPPORTS_CXX14
-- Performing Test SUPPORTS_CXX14 - Failed
CMake Error at CMake/lrs_macros.cmake:19 (message):
  Project 'realsense2' requires C++14 or higher
Call Stack (most recent call first):
  CMakeLists.txt:53 (config_cxx_flags)

-- Configuring incomplete, errors occurred!
See also "/home/debix/librealsense-2.54.1/build/CMakeFiles/CMakeOutput.log".
See also "/home/debix/librealsense-2.54.1/build/CMakeFiles/CMakeError.log".

```

jesusramondovale, 2023-07-24:
Added "set(CMAKE_CXX_STANDARD 14)" at line 1 of CMakeLists.txt and CMake/lrs_macros.cmake files but nothing changed (same error output)

MartyG-RealSense, 2023-07-24:
The Debix Model A has an i.MX 8M Plus quad-core ARM Cortex-A53 processor.  There have been past problems with installing librealsense on IMX8 boards.  But use of `-DFORCE_RSUSB_BACKEND = true` usually fixed them.

Which version of CMake are you using, please?  Version 3.8 or newer is recommended for librealsense.

jesusramondovale, 2023-07-24:
I'm using CMake version 3.22.1, I'll try installing a newer one and doing everything again. Thank you!

jesusramondovale, 2023-07-24:
So, should it compile correctyly with CMake 3.22? I don't know what can I try next

MartyG-RealSense, 2023-07-24:
There have not been many past cases on this support forum site of boards with IMX8 so there are not many references about it, except that they had problems with installing librealsense and RSUSB usually solved the problem.

You said that you edited the CMakeLists.txt file to add **set(CMAKE_CXX_STANDARD 14)**.  Have you also tried entering it into the CMake command-line interface instead before inputting the CMake build instruction?

jesusramondovale, 2023-07-24:
I don't know what you mean, if you're asking if I added the SET command before executing cmake.. yes, I did.

MartyG-RealSense, 2023-07-24:
Instead of a source code installation with CMake, let's try installing from packages on Ubuntu 22.04 and kernel 5.15 using the RealSense ROS wrapper's Humble install instructions.

1.  First, set up the Ubuntu repositories using the instructions at the link below.

http://wiki.ros.org/Installation/Ubuntu/Sources

 2.  Then afterwards install all RealSense ROS packages with the command below.

`sudo apt install ros-humble-librealsense2*`

jesusramondovale, 2023-07-24:
· Succesfully setted the Ubuntu Respositories : 

 ```
debix@DEBIX26:~$ sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
[sudo] password for debix: 
debix@DEBIX26:~$ curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -
curl: /usr/lib/libcurl.so.4: no version information available (required by curl)
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
OK
´´´

· Succesfully installed the librealsense2 via apt install: 

```
debix@DEBIX26:~$ sudo apt install ros-humble-librealsense2*
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'ros-humble-librealsense2' for glob 'ros-humble-librealsense2*'
Note, selecting 'ros-humble-librealsense2-dbgsym' for glob 'ros-humble-librealsense2*'
The following NEW packages will be installed:
  ros-humble-librealsense2 ros-humble-librealsense2-dbgsym
0 upgraded, 2 newly installed, 0 to remove and 2 not upgraded.
Need to get 69.1 MB of archives.
After this operation, 87.3 MB of additional disk space will be used.
Get:1 http://packages.ros.org/ros2/ubuntu jammy/main arm64 ros-humble-librealsense2 arm64 2.54.1-1jammy.20230624.065425 [3307 kB]
Get:2 http://packages.ros.org/ros2/ubuntu jammy/main arm64 ros-humble-librealsense2-dbgsym arm64 2.54.1-1jammy.20230624.065425 [65.7 MB]
Fetched 69.1 MB in 34s (2011 kB/s)                                                                                                                                                                                
Selecting previously unselected package ros-humble-librealsense2.
(Reading database ... 158886 files and directories currently installed.)
Preparing to unpack .../ros-humble-librealsense2_2.54.1-1jammy.20230624.065425_arm64.deb ...
Unpacking ros-humble-librealsense2 (2.54.1-1jammy.20230624.065425) ...
Selecting previously unselected package ros-humble-librealsense2-dbgsym.
Preparing to unpack .../ros-humble-librealsense2-dbgsym_2.54.1-1jammy.20230624.065425_arm64.deb ...
Unpacking ros-humble-librealsense2-dbgsym (2.54.1-1jammy.20230624.065425) ...
Setting up ros-humble-librealsense2 (2.54.1-1jammy.20230624.065425) ...
Setting up ros-humble-librealsense2-dbgsym (2.54.1-1jammy.20230624.065425) ...
Processing triggers for libc-bin (2.35-0ubuntu3.1) ...
```

MartyG-RealSense, 2023-07-24:
What happens if you launch with **ros2 launch** instead of ros2 run?

`ros2 launch realsense2_camera rs_launch.py enable_color:=false spatial_filter.enable:=true temporal_filter.enable:=true`

jesusramondovale, 2023-07-24:
> What happens if you launch with **ros2 launch** instead of ros2 run?
> 
> `ros2 launch realsense2_camera rs_launch.py enable_color:=false spatial_filter.enable:=true temporal_filter.enable:=true`

Exactly the same (Illegal instruction: core dumped)

MartyG-RealSense, 2023-07-24:
No, you would not need to execute it from an exact folder.  ROS commands such as this should be input into the ROS terminal window and not an Ubuntu terminal window.

jesusramondovale, 2023-07-24:
> No, you would not need to execute it from an exact folder. ROS commands such as this should be input into the ROS terminal window and not an Ubuntu terminal window.

What is the **ROS terminal** ??

MartyG-RealSense, 2023-07-24:
1.  Open a new Terminal Window (Ctrl + Alt + T). 

2.  In the new terminal, input **roscore** to launch the 'ROS Master'.

[image]

3.  Open a second new terminal window with Ctrl - Alt - T.

4.  Input the launch command into this second terminal window.

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_color:=false -p spatial_filter.enable:=true -p temporal_filter.enable:=true`

jesusramondovale, 2023-07-24:
ROS2 distros don't use the slave/master system so I can't launch the "ROS Master" on my ROS2 - humble Hawskbill ROS distro...

MartyG-RealSense, 2023-07-24:
It seems that on ROS2 you should enter the command below to enable ROS2 Humble.

`source /opt/ros/humble/setup.bash`

This command needs to be entered every time that you are starting ROS2.

Then try entering the ROS2 launch command.

jesusramondovale, 2023-07-24:
I've got the ROS2 distro sourced every time I start a Linux session since I've written it in the ~./bashrc file, so... nothing changed.

Tried it and gave the exact same error (Illegal instruction: core dumped)

MartyG-RealSense, 2023-07-24:
As this error has occurred on both source code and package installations, it does not seem as though it is librealsense or the ROS wrapper causing the problem.  It is more likely to be the board's ARM processor.

Some have suggested inputting the command below to see whether it makes a difference.

`export OPENBLAS_CORETYPE=ARMV8`

jesusramondovale, 2023-07-24:
Did the export and nothing changed :(

```
debix@DEBIX26:~$ export OPENBLAS_CORETYPE=ARMV8
debix@DEBIX26:~$ ros2 launch realsense2_camera rs_launch.py enable_color:=false spatial_filter.enable:=true temporal_filter.enable:=true
```

MartyG-RealSense, 2023-07-24:
Do you have an Ubuntu PC computer that you can test installation on?  If it works on that but not on the Debix then it would likely confirm that the problem is with compatibility with Debix.

MartyG-RealSense, 2023-07-24:
As a swapfile is virtual memory created from setting aside drive storage space, it will be slower than real memory and only as fast as the access speed of the storage device.  Swapfile memory will start being used once the real memory is consumed.

I would rather trust real memory capacity for running an application than swapfile memory.  But there is not enough information about the use of IMX8 boards with RealSense to draw a firm conclusion about whether the low amount of real memory is responsible for the ROS wrapper error.

Would it be possible for your project to use Pi 4 instead of Debix or is Debix a compulsory project requirement?

jesusramondovale, 2023-07-24:
Using a Debix SBC is a compulsory project requirement, my friend ...

jesusramondovale, 2023-07-24:
They have been recently working on a Ubuntu 22.04 Image, they have released it a month ago aproximately, as you can see in their downloads webpage: https://debix.io/Software/download.html

MartyG-RealSense, 2023-07-24:
Even though installing librealsense and the wrapper from packages does not include graphical librealsense tools in the installation, are you able to run text-based tools such as **rs-enumerate-devices** and **rs-hello-realsense** on the Debix?

On Ubuntu, if they have been installed then they should be located in the **usr/local/bin** folder.

jesusramondovale, 2023-07-25:
I'm going to re-isntall the Debix's Ubuntu Image from zero, and then re-do all the steps you mention. I'll keep you updated, thank you so much

jesusramondovale, 2023-07-25:
Summarize: I can't build the librealsense from source, neither can I execute a single command as "rs-enumerate-devices" after installing librealsense with sudo apt-get... it says "command not found" ... this is now becoming embarrasing. 

Do you know what can I try? If not, you can close the issue if you want..

jesusramondovale, 2023-07-25:
Re-installed all dependencies, Realsense SDK and ROS Wrapper. 

I can use the rs-enumerate-devices command, and much more, but the error given is the same as first: 

```
debix@DEBIX26:~/Desktop/librealsense/build$ rs-enumerate-devices
 25/07 09:26:26,587 ERROR [281473799996512] (librealsense-exception.h:52) xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
Could not create device - xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument . Check SDK logs for details
Segmentation fault (core dumped)
```

I remind you that Intel is correctly plugged and recognised by the Debix board as you can see:  

´´´
debix@DEBIX26:~$ lsusb | grep "Intel"
Bus 002 Device 005: ID 8086:0b5c Intel Corp. Intel(R) RealSense(TM) Depth Camera 455 

´´´

MartyG-RealSense, 2023-07-25:
The directory that Linux stores logs in is **/var/log**

jesusramondovale, 2023-07-26:
There's no Realsense Logs in there. Does the Realsense SDK even write to logs?

MartyG-RealSense, 2023-07-26:
Yes, the SDK can write log files.

https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A#q-how-do-i-enable-librealsense-logs

The documentation implies that the SDK's logging is not on by default and has to be enabled.

Also, on a ROS2 launch log, a log location is provided at the top of the launch log on the line **All log files can be found below**, followed by the log path for your particular computer.

MartyG-RealSense, 2023-07-26:
In regards to libcurl, the CMake flag that depends on it can be set to false.

`cmake ../ -DCHECK_FOR_UPDATES=OFF`

jesusramondovale, 2023-07-27:
I achieved to get rs-enumerate-devices work correctly : it gives me a large list of video properties (output OK) but the same error persists when I try to run ROS2 camera node: 

```
debix@DEBIX26:~$ ros2 run realsense2_camera realsense2_camera_node
[INFO] [1690469333.859057130] [camera]: RealSense ROS v4.54.1
[INFO] [1690469333.859422552] [camera]: Built with LibRealSense v2.54.1
[INFO] [1690469333.859521736] [camera]: Running with LibRealSense v2.54.1
 27/07 16:48:53,888 WARNING [281472518645984] (d400-factory.cpp:1195) DS5 group_devices is empty.
 27/07 16:48:54,267 ERROR [281472518645984] (librealsense-exception.h:52) xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
 27/07 16:48:54,291 WARNING [281472518645984] (rs.cpp:312) null pointer passed for argument "device"
[WARN] [1690469334.291689435] [camera]: Device 1/3 failed with exception: xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
[ros2run]: Segmentation fault
```

MartyG-RealSense, 2023-07-27:
Please try resetting the camera at launch with **initial_reset** as it has not been mentioned yet in this case that it has been tried yet.

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p initial_reset:=true`

MartyG-RealSense, 2023-08-09:
Case closed due to no further comments received.

closed 2023-08-09 by MartyG-RealSense

jesusramondovale, 2023-09-27:
Hey. I've got the exact same problem again... 
Installed RealSense SDK succesfully (sudo rs-enumerate-devices works fine)
but when I execute the ROS2 Wrapper with 

`debix@DEBIX25:~/librealsense$ ros2 launch realsense2_camera rs_launch.py`

I get this error ouput: 

`[INFO] [launch]: All log files can be found below /home/debix/.ros/log/2023-09-27-06-45-57-373454-DEBIX25-14668
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [14682]
[realsense2_camera_node-1] [INFO] [1695797158.404652736] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1695797158.404968999] [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1695797158.405047252] [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1]  27/09 06:45:58,435 WARNING [281472476768480] (d400-factory.cpp:1195) DS5 group_devices is empty.
[realsense2_camera_node-1]  27/09 06:45:58,830 ERROR [281472476768480] (librealsense-exception.h:52) xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
[realsense2_camera_node-1]  27/09 06:45:58,837 WARNING [281472476768480] (rs.cpp:312) null pointer passed for argument "device"
[realsense2_camera_node-1] [WARN] [1695797158.837362369] [camera.camera]: Device 1/3 failed with exception: xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
[ERROR] [realsense2_camera_node-1]: process has died [pid 14682, exit code -11, cmd '/home/debix/ros2_ws/install/realsense2_camera/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_j9ko_vrh --params-file /tmp/launch_params_w3fbxgpj'].` 

I need help as soon as possible, please.

MartyG-RealSense, 2023-09-27:
Hi @jesusramondovale  **xioctl(VIDIOC_S_EXT_CTRLS)** errors typically indicate that there is a conflict with the Linux kernel.  Which kernel version are you using, please?

Do you experience the same problem if you launch with **ros2 run** instead of ro2 launch?

`ros2 run realsense2_camera realsense2_camera_node`