realsenseai/realsense-ros #2818: roslaunch problem (please help!!!):   what():  stop_streaming() failed. Hid device is not streaming!

loubron23, 2023-07-21:
My system is ubuntu 16.04lts, camera is realsense d435i, RealSense ROS v2.2.7, LibRealSense v2.54.1
and when i use the "roslaunch realsense2_camera rs_camera.launch" I got:

[ INFO] [1689951428.123701106]: RealSense Node Is Up!
terminate called after throwing an instance of 'rs2::wrong_api_call_sequence_error'
  what():  stop_streaming() failed. Hid device is not streaming!
[camera/realsense2_camera_manager-1] process has died [pid 14152, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/neousys/.ros/log/e6d6ff9c-279d-11ee-b7c9-003018050a5a/camera-realsense2_camera_manager-1.log].
log file: /home/neousys/.ros/log/e6d6ff9c-279d-11ee-b7c9-003018050a5a/camera-realsense2_camera_manager-1*.log
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/neousys/.ros/log/e6d6ff9c-279d-11ee-b7c9-003018050a5a/camera-realsense2_camera-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

This problem has been bothering me for three days now。

please help

MartyG-RealSense, 2023-07-21:
Hi @loubron23  You have a mismatch between your librealsense SDK and RealSense ROS wrapper versions.  It is important to match the wrapper version as closely as possible with the librealsense version listed in the release notes for that particular wrapper version.

As you are using the ROS1 wrapper, the best match is librealsense **2.51.1** and wrapper **2.3.2**.  You should also ensure that camera firmware driver version **5.13.0.50** is used with librealsense 2.51.1.  It is necessary to use 2.51.1 because the ROS1 wrapper is no longer updated and so is not designed for use with newer librealsense versions such as 2.53.1 and 2.54.1.

If you would prefer to continue with librealsense 2.54.1 then you should change from ROS1 to ROS2, update the firmware to **5.15.0.2** and build ROS2 wrapper **4.54.1**.

loubron23, 2023-07-22:
> Hi @loubron23 You have a mismatch between your librealsense SDK and RealSense ROS wrapper versions. It is important to match the wrapper version as closely as possible with the librealsense version [...]

Thank you very much. 
I have installed librealsense 2.51.1 and wrapper 2.3.2. But now there is a new problem that after roslaunch it will stay for a period of 10 seconds and then automatically shut down.

roslaunch realsense2_camera rs_camera.launch 
... logging to /home/neousys/.ros/log/e6d6ff9c-279d-11ee-b7c9-003018050a5a/roslaunch-neousys-Nuvo-7000-Series-4774.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://neousys-Nuvo-7000-Series:41565/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
[... 71 lines cut ...]
 22/07 15:26:24,643 WARNING [140222821062400] (backend-v4l2.cpp:1444) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[ INFO] [1690010784.673431045]: RGB Camera was found.
[ INFO] [1690010784.674065698]: Motion Module was found.
[ INFO] [1690010784.674092082]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1690010784.674113214]: num_filters: 0
[ INFO] [1690010784.674130529]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ WARN] [1690010784.850720380]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1690010784.879075402]: Done Setting Dynamic reconfig parameters.
[ INFO] [1690010784.881004098]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1690010784.884396778]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1690010784.884466814]: setupPublishers...
[ INFO] [1690010784.892068266]: Expected frequency for depth = 30.00000
[ INFO] [1690010784.969353556]: Expected frequency for color = 30.00000
[ INFO] [1690010785.038995596]: setupStreams...
[ INFO] [1690010785.093768376]: SELECTED BASE:Depth, 0
[ INFO] [1690010785.130113405]: RealSense Node Is Up!
[... cut ...] e. Please refer to the installation chapter for details.
[ WARN] [1690010785.214290314]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 22/07 15:26:25,536 WARNING [140222514386688] (ds5-timestamp.cpp:69) UVC metadata payloads not available. Please refer to the installation chapter for details.
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/neousys/.ros/log/e6d6ff9c-279d-11ee-b7c9-003018050a5a/camera-realsense2_camera-2*.log

MartyG-RealSense, 2023-07-22:
Does it make a difference if you add **initial_reset:=true** to your roslaunch instruction to reset the camera at launch, please?

loubron23, 2023-07-22:
> Does it make a difference if you add **initial_reset:=true** to your roslaunch instruction to reset the camera at launch, please?
Oops, It's still the same result.

loubron23, 2023-07-22:
> Does it make a difference if you add **initial_reset:=true** to your roslaunch instruction to reset the camera at launch, please?

@MartyG-RealSense  Oops, It's still the same result.

MartyG-RealSense, 2023-07-22:
```
likely requires patch for fourcc code RW16!
Frame metadata isn't available!
UVC metadata payloads not available.
```

The messages above indicate that librealsense was built from source code but the Linux kernel has not had a kernel patch script applied to it.  It is therefore not capable of providing camera **metadata** information.

If you did build librealsense from source code, on an Ubuntu 16.04 computer the librealsense kernel patch script **./scripts/patch-realsense-ubuntu-lts.sh** should be run if using a kernel version older than 5.13.

loubron23, 2023-07-22:
> ```
> likely requires patch for fourcc code RW16!
> Frame metadata isn't available!
> UVC metadata payloads not available.
> ```
> 
> The messages above indicate that librealsense was built from source code but the Linux kernel has not had a kernel patch script applied to it. It is therefore not capable of providing camera **metadata** information.
> 
> If you did build librealsense from source code, on an Ubuntu 16.04 computer then the librealsense kernel patch script **./scripts/patch-realsense-ubuntu-lts.sh** should be run if using a kernel version older than 5.13.

thank you! But for some reason, i can only try this after 1 week. Please wait, thanks~

MartyG-RealSense, 2023-07-22:
It's no trouble at all to keep this case open until you are ready.  Thanks!

loubron23, 2023-08-02:
hello, @MartyG-RealSense !
I run the command ./scripts/patch-realsense-ubuntu-lts.sh when installing the librealsense. And I reinstall the realsense2_camera, but now it still has problem.
After using the "roslaunch realsense2_camera rs_camera.launch" I got:

... logging to /home/neousys/.ros/log/271ba132-3038-11ee-a6ec-003018050a5a/roslaunch-neousys-Nuvo-7000-Series-18500.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://neousys-Nuvo-7000-Series:44193/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
[... 67 lines cut ...]
[ INFO] [1690957763.241310127]: Device Sensors: 
[ INFO] [1690957763.271803294]: Stereo Module was found.
[ INFO] [1690957763.296733607]: RGB Camera was found.
[ INFO] [1690957763.297300770]: Motion Module was found.
[ INFO] [1690957763.297327542]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1690957763.297344782]: num_filters: 0
[ INFO] [1690957763.297355027]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1690957763.461541879]: Done Setting Dynamic reconfig parameters.
[ INFO] [1690957763.463515421]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1690957763.466847707]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1690957763.466922327]: setupPublishers...
[ INFO] [1690957763.475068240]: Expected frequency for depth = 30.00000
[ INFO] [1690957763.566013129]: Expected frequency for color = 30.00000
[ INFO] [1690957763.639362593]: setupStreams...
[ INFO] [1690957763.687854924]: SELECTED BASE:Depth, 0
[ INFO] [1690957763.721261541]: RealSense Node Is Up!
[ WARN] [1690957763.808488594]: 
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/neousys/.ros/log/271ba132-3038-11ee-a6ec-003018050a5a/camera-realsense2_camera-2*.log

There is a warn without any content, and then it just finished. What's wrong?

loubron23, 2023-08-02:
@MartyG-RealSense  hello!

I find the prompt: hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready).
Then I downgrade the realsense firm version to 5. 13.0.50, then the prompt disappeared.
However, the result is the same.

... logging to /home/neousys/.ros/log/271ba132-3038-11ee-a6ec-003018050a5a/roslaunch-neousys-Nuvo-7000-Series-32529.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://neousys-Nuvo-7000-Series:44707/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
[... 63 lines cut ...]
[ INFO] [1690964337.616400660]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0
[ INFO] [1690964337.616460056]: Device FW version: 05.13.00.50
[ INFO] [1690964337.616507606]: Device Product ID: 0x0B3A
[ INFO] [1690964337.616573745]: Enable PointCloud: Off
[ INFO] [1690964337.616619293]: Align Depth: Off
[ INFO] [1690964337.616659489]: Sync Mode: Off
[ INFO] [1690964337.616866593]: Device Sensors: 
[ INFO] [1690964337.649853545]: Stereo Module was found.
[ INFO] [1690964337.680336458]: RGB Camera was found.
[ INFO] [1690964337.680967630]: Motion Module was found.
[ INFO] [1690964337.681003898]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1690964337.681023762]: num_filters: 0
[ INFO] [1690964337.681036489]: Setting Dynamic reconfig parameters.
[ INFO] [1690964337.916602207]: Done Setting Dynamic reconfig parameters.
[ INFO] [1690964337.918569211]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1690964337.921944491]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1690964337.922025427]: setupPublishers...
[ INFO] [1690964337.930696519]: Expected frequency for depth = 30.00000
[ INFO] [1690964338.030370813]: Expected frequency for color = 30.00000
[ INFO] [1690964338.111179456]: setupStreams...
[ INFO] [1690964338.157699992]: SELECTED BASE:Depth, 0
[ INFO] [1690964338.191568512]: RealSense Node Is Up!
[ WARN] [1690964338.278504096]: 
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/neousys/.ros/log/271ba132-3038-11ee-a6ec-003018050a5a/camera-realsense2_camera-2*.log

MartyG-RealSense, 2023-08-02:
I note from the top of the above log that you are using a **Neousys Nuvo 7000 Series** industrial PC with an Intel processor.  

Whilst RealSense 400 Series cameras can work with any Intel processor, they may work better with PCs with unusual hardware / software configurations such as industrial machines if librealsense is built from source code with CMake with the **-DFORCE_RSUSB_BACKEND=TRUE** flag included in the CMake build installation.  This flag causes the SDK to bypass the kernel, meaning that it is not dependent on Linux versions or kernel versions and does not require a kernel patch to be applied.

loubron23, 2023-08-02:
> I note from the top of the above log that you are using a **Neousys Nuvo 7000 Series** industrial PC with an Intel processor.
> 
> Whilst RealSense 400 Series cameras can work with any Intel processor, they may work better with PCs with unusual hardware / software configurations such as industrial machines if librealsense is built from source code with CMake with the **-DFORCE_RSUSB_BACKEND=TRUE** flag included in the CMake build installation. This flag causes the SDK to bypass the kernel, meaning that it is not dependent on Linux versions or kernel versions and does not require a kernel patch to be applied.
> 
> If librealsense is built again then the ROS wrapper must be built again afterwards.

@MartyG-RealSense Thank you for reply! Do you mean that use "cmake …/ -DFORCE_RSUSB_BACKEND=TRUE" instead of "cmake …/ -DBUILD_EXAMPLES=true" when build librealsense?

MartyG-RealSense, 2023-08-02:
The -DFORCE flag should be added to your existing CMake instruction.

cmake …/ -DBUILD_EXAMPLES=true -DFORCE_RSUSB_BACKEND=TRUE

If you need graphics support and graphical tools and graphical programs such as realsense-viewer then you should add **-DBUILD_GRAPHICAL_EXAMPLES=TRUE** otherwise leave it out if your industrial computer is a 'headless' one without a display screen.

I also recommend including the flag **-DCMAKE_BUILD_TYPE=Release** which should make the SDK build work better.

For example:

`cmake …/ -DBUILD_EXAMPLES=true -DFORCE_RSUSB_BACKEND=TRUE -DCMAKE_BUILD_TYPE=Release`

loubron23, 2023-08-02:
@MartyG-RealSense  I use the command "cmake …/ -DBUILD_EXAMPLES=true -DFORCE_RSUSB_BACKEND=TRUE -DCMAKE_BUILD_TYPE=Release -DBUILD_GRAPHICAL_EXAMPLES=TRUE " to build the librealsense, and rebuild the ros wrapper. Then it comes as (I was able to use realsense-ROS normally four months ago on the same PC. Does this indicate that it has nothing to do with the industrial PC?):

... logging to /home/neousys/.ros/log/271ba132-3038-11ee-a6ec-003018050a5a/roslaunch-neousys-Nuvo-7000-Series-5838.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://neousys-Nuvo-7000-Series:42265/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 79 lines cut ...]
 02/08 23:17:47,416 WARNING [139972875634432] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1690989467.466987918]: SELECTED BASE:Depth, 0
[ INFO] [1690989467.508097984]: RealSense Node Is Up!
[ WARN] [1690989467.534236112]: 
 02/08 23:17:47,535 WARNING [139972875634432] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/08 23:17:47,585 WARNING [139972875634432] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/08 23:17:47,637 WARNING [139972875634432] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/08 23:17:47,768 WARNING [139972875634432] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/08 23:17:47,818 WARNING [139972875634432] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/08 23:17:47,868 WARNING [139972875634432] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/08 23:17:48,021 WARNING [139972875634432] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/08 23:17:48,172 WARNING [139972875634432] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[... cut ...]
 02/08 23:17:49,031 WARNING [139972875634432] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/08 23:17:49,183 WARNING [139972875634432] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/08 23:17:49,233 WARNING [139972875634432] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/neousys/.ros/log/271ba132-3038-11ee-a6ec-003018050a5a/camera-realsense2_camera-2*.log

MartyG-RealSense, 2023-08-02:
If you were able to use the ROS1 wrapper successfully on the same PC several months ago then it would indicate that the industrial PC can run the ROS wrapper.  But the ROS1 wrapper has not been updated since then, so it also suggests that there is not a fault in the wrapper either. 

As you built with **-DBUILD_GRAPHICAL_EXAMPLES=TRUE** then the realsense-viewer librealsense program should be accessible.  Does the camera work normally in realsense-viewer?

If librealsense and the ROS wrapper have been installed additional times on the same computer then there may be a hidden glitch somewhere, as I would recommend removing existing previous installations of librealsense and the wrapper before performing a new installation.  A fast way to do that would be to simply wipe the computer's drive and install everything again from the beginning, including Ubuntu and ROS.  Doing so can remove hidden tangles from the system and suddenly resolve previous unsolvable errors.

loubron23, 2023-08-02:
> If you were able to use the ROS1 wrapper successfully on the same PC several months ago then it would indicate that the industrial PC can run the ROS wrapper. But the ROS1 wrapper has not been updated since then, so it also suggests that there is not a fault in the wrapper either.
> 
> As you built with **-DBUILD_GRAPHICAL_EXAMPLES=TRUE** then the realsense-viewer librealsense program should be accessible. Does the camera work normally in realsense-viewer?
> 
> If librealsense and the ROS wrapper have been installed additional times on the same computer then there may be a hidden glitch somewhere, as I would recommend removing existing previous installations of librealsense and the wrapper before performing a new installation. A fast way to do that would be to simply wipe the computer's drive and install everything again from the beginning, including Ubuntu and ROS. Doing so can remove hidden tangles from the system and suddenly resolve previous unsolvable errors.
> 
> Which version of ROS1 are you using, please? The most suitable ROS1 branch for Ubuntu 16.04 would be **ROS Kinetic** instead of Melodic or Noetic,

Yes, the camera has always been normal in the realsense viewer.
I have indeed reinstalled librealsense and the ROS wrapper many times. But my current conditions do not support me to reinstall the ubuntu and ros because there are many important things that have already been configured. I did consider this, so I tried using the 18.04 Docker on the PC, but it still reported an error. May I have a way to remove existing previous installations of librealsense and the wrapper without reinstalling ubuntu and ROS? Thank you!

loubron23, 2023-08-03:
> Source code builds of the ROS wrapper can be removed simply by completely deleting the entire **catkin_ws** catkin workspace folder and creating it again on your next wrapper installation.
> 
> Source code builds of librealsense can be removed by going to the 'build' directory of your librealsense source code folder and using the CMake command below.
> 
> **sudo make uninstall && make clean**
> 
> If you have ever installed librealsense or the ROS wrapper from packages on the computer instead of building from source code then you can remove all RealSense related packages from the computer by inputting the command below into the Ubuntu terminal.
> 
> `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`
OK, I removed existing previous installations of librealsense and the wrapper. The result is still the same.!!! To be honest, my mindset has already collapsed.

MartyG-RealSense, 2023-08-03:
After performing the launch and then stopping it, please try unplugging the camera and wait 10 seconds.  Then re-insert the camera into the USB port and perform the ROS launch again.

Does the  **control_transfer returned error, index: 768, error: Success, number: 0** warning still occur after the unplug, wait and re-insertion?

loubron23, 2023-08-03:
> control_transfer returned error, index: 768, error: Success, number: 0

control_transfer returned error, index: 768, error: Success, number: 0 warning doesn't occur now. But the result is still the same, warn and then finished cleanly.

loubron23, 2023-08-04:
> Let's next try a different kind of ROS1 launch - **rs_rgbd.launch** instead of rs_camera.
> 
> Before doing so, support for an RGBD launch in Kinetic will need to be installed with the instruction below.
> 
> `sudo apt-get install ros-kinetic-rgbd-launch`
> 
> After this installation is complete, perform the rs_rgbd launch.
> 
> `roslaunch realsense2_camera rs_rgbd launch`

Thank you for being so patient! 
If I use `roslaunch realsense2_camera rs_rgbd launch`, it will become:

... logging to /home/neousys/.ros/log/3efbd18c-31be-11ee-93fe-003018050a5a/roslaunch-neousys-Nuvo-7000-Series-28330.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://neousys-Nuvo-7000-Series:37973/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
[... 228 lines cut ...]
[ INFO] [1691082744.706444189]: setupPublishers...
[ INFO] [1691082744.711917740]: Expected frequency for depth = 30.00000
[ INFO] [1691082744.801287047]: Expected frequency for infra1 = 30.00000
[ INFO] [1691082744.886055864]: Expected frequency for aligned_depth_to_infra1 = 30.00000
[ INFO] [1691082744.973039816]: Expected frequency for infra2 = 30.00000
[ INFO] [1691082745.059484730]: Expected frequency for color = 30.00000
[ INFO] [1691082745.140040821]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1691082745.220511993]: setupStreams...
[ INFO] [1691082745.233177625]: insert Depth to Stereo Module
[ INFO] [1691082745.233703215]: insert Color to RGB Camera
[ INFO] [1691082745.233870728]: insert Infrared to Stereo Module
[ INFO] [1691082745.233987394]: insert Infrared to Stereo Module
[ INFO] [1691082745.234124778]: insert Gyro to Motion Module
[ INFO] [1691082745.234244162]: insert Accel to Motion Module
 04/08 01:12:25,247 WARNING [139700304074496] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ INFO] [1691082745.338291034]: num_filters: 0
[ INFO] [1691082745.338441380]: SELECTED BASE:Depth, 0
[ INFO] [1691082745.347373280]: RealSense Node Is Up!
terminate called without an active exception
[camera/realsense2_camera_manager-1] process has died [pid 28339, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/neousys/.ros/log/3efbd18c-31be-11ee-93fe-003018050a5a/camera-realsense2_camera_manager-1.log].
log file: /home/neousys/.ros/log/3efbd18c-31be-11ee-93fe-003018050a5a/camera-realsense2_camera_manager-1*.log
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/neousys/.ros/log/3efbd18c-31be-11ee-93fe-003018050a5a/camera-realsense2_camera-2*.log
[camera/points_xyzrgb_hw_registered-4] process has finished cleanly
log file: /home/neousys/.ros/log/3efbd18c-31be-11ee-93fe-003018050a5a/camera-points_xyzrgb_hw_registered-4*.log

loubron23, 2023-08-04:
Also, I would like to ask. I tried to install librealsense on another Industrial PC, but its kernel is 4.19.90+. Does this mean that it does not support librealsense？

MartyG-RealSense, 2023-08-04:
It is possible to install librealsense on a computer with an unsupported kernel but there may be unforeseen issues in regards to stability and **hardware timestamp** metadata will not be supported.

Kernel 4.19 is not supported.  However, if librealsense is built from source code with the **RSUSB backend = true** installation method then it bypasses the kernel and so it can be used with any kernel, with hardware metadata support included.

loubron23, 2023-08-04:
> It is possible to install librealsense on a computer with an unsupported kernel but there may be unforeseen issues in regards to stability and **hardware timestamp** metadata will not be supported. [...]

> 

> Thank you for being so patient! If I use `roslaunch realsense2_camera rs_rgbd launch`, it will become:
>
> ... logging to [...]

OK, I'll try 'cmake …/ -DBUILD_EXAMPLES=true -DFORCE_RSUSB_BACKEND=TRUE -DCMAKE_BUILD_TYPE=Release' on the other PC.
And could you help me with this issue?
Your help was very much appreciated!!!

MartyG-RealSense, 2023-08-04:
In your log it states that the launch is using librealsense 2.45.0 and wrapper 2.2.7.

```
[ INFO] [1691082743.511562452]: RealSense ROS v2.2.7
[ INFO] [1691082743.511588232]: Running with LibRealSense v2.45.0
```

Wrapper 2.2.7 would not be compatible for use with librealsense 2.45.0 as it was designed for the much older librealsense **2.25.0**.  It is important to match the wrapper version as closely as possible with a specific librealsense version listed in that particular wrapper version's release notes.  The correct ROS1 wrapper match for librealsense 2.45.0 is **2.3.0**.

Also, if you have firmware 5.15.0.2 installed in your camera then it is not fully compatible with librealsense versions older than 2.54.1.  The correct firmware match for 2.45.0 is **5.12.13.50**.  If you install this firmware in your camera though then you will not be able to use librealsense 2.54.1 / wrapper 4.51.1.

loubron23, 2023-08-04:
Oops. I may have accidentally made a mistake with the version of librealsense. I am using the correct version now, but the result is back to the beginning. Hahaha

... logging to /home/neousys/.ros/log/3efbd18c-31be-11ee-93fe-003018050a5a/roslaunch-neousys-Nuvo-7000-Series-26820.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://neousys-Nuvo-7000-Series:46865/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 58 lines cut ...]
[ INFO] [1691138054.328715810]: Device Serial No: 045322073426
[ INFO] [1691138054.328754526]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0
[ INFO] [1691138054.328811305]: Device FW version: 05.13.00.50
[ INFO] [1691138054.328861700]: Device Product ID: 0x0B3A
[ INFO] [1691138054.328894101]: Enable PointCloud: Off
[ INFO] [1691138054.328935026]: Align Depth: Off
[ INFO] [1691138054.328985945]: Sync Mode: Off
[ INFO] [1691138054.329184418]: Device Sensors: 
[ INFO] [1691138054.362899773]: Stereo Module was found.
[ INFO] [1691138054.391386753]: RGB Camera was found.
[ INFO] [1691138054.392041656]: Motion Module was found.
[ INFO] [1691138054.392071833]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1691138054.392091306]: num_filters: 0
[ INFO] [1691138054.392102892]: Setting Dynamic reconfig parameters.
[ INFO] [1691138054.640133270]: Done Setting Dynamic reconfig parameters.
[ INFO] [1691138054.642197326]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1691138054.645821705]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1691138054.645895399]: setupPublishers...
[ INFO] [1691138054.656679732]: Expected frequency for depth = 30.00000
[ INFO] [1691138054.751930249]: Expected frequency for color = 30.00000
[ INFO] [1691138054.855973085]: setupStreams...
[ INFO] [1691138054.876542564]: SELECTED BASE:Depth, 0
[ INFO] [1691138054.920724885]: RealSense Node Is Up!
[ WARN] [1691138054.997728425]: 
[camera/realsense2_camera-2] process has finished cleanly

MartyG-RealSense, 2023-08-04:
There is not an obvious reason for why the process is finishing as the launch proceeds fine up until the point where the launch completes at RealSense Node is Up. 

Perhaps it is an issue of insufficient power being provided to the camera by the Neousys computer's USB port and the launch is crashing as the streams are enabled and try to draw a greater amount of power from the port.

Do you have the option of trying the camera with a mains electricity powered USB 3 hub instead of plugging the camera directly into the computer or into a 'passive' hub (one that is powered by the computer instead of a mains plug socket).

loubron23, 2023-08-04:
> There is not an obvious reason for why the process is finishing as the launch proceeds fine up until the point where the launch completes at RealSense Node is Up.
> 
> Perhaps it is an issue of insufficient power being provided to the camera by the Neousys computer's USB port and the launch is crashing as the streams are enabled and try to draw a greater amount of power from the port.
> 
> Do you have the option of trying the camera with a mains electricity powered USB 3 hub instead of plugging the camera directly into the computer or into a 'passive' USB hub (one that is powered by the computer instead of a mains plug socket).

I've tried. Unfortunately, It doesn't work. 
But the strange thing is that I tried it on my other PC with kernel 4.19 and made a similar error(below). But I did have no problem trying on my classmate's PC.

... logging to /home/HwHiAiUser/.ros/log/3efbd18c-31be-11ee-93fe-003018050a5a/roslaunch-davinci-mini-9670.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://davinci-mini:45299/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
[... 73 lines cut ...]
[ INFO] [1691148036.946696826]: Stereo Module was found.
[ INFO] [1691148036.992738737]: RGB Camera was found.
[ INFO] [1691148036.994483671]: Motion Module was found.
[ INFO] [1691148036.994800231]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1691148036.995031479]: num_filters: 0
[ INFO] [1691148036.995206425]: Setting Dynamic reconfig parameters.
[ INFO] [1691148040.901009458]: Done Setting Dynamic reconfig parameters.
[ INFO] [1691148040.902871526]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1691148040.906111655]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1691148040.906346758]: setupPublishers...
[ INFO] [1691148040.913883674]: Expected frequency for depth = 30.00000
[ INFO] [1691148041.016883240]: Expected frequency for color = 30.00000
[ INFO] [1691148041.081394790]: setupStreams...
 04/08 11:20:41,300 WARNING [281469784146304] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1691148041.351847555]: SELECTED BASE:Depth, 0
[ INFO] [1691148041.394620430]: RealSense Node Is Up!
[ WARN] [1691148041.419597568]: 
 04/08 11:20:41,421 WARNING [281469784146304] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 04/08 11:20:41,711 WARNING [281469784146304] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 04/08 11:20:41,865 WARNING [281469784146304] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/HwHiAiUser/.ros/log/3efbd18c-31be-11ee-93fe-003018050a5a/camera-realsense2_camera-2*.log

MartyG-RealSense, 2023-08-04:
Does the same problem occur if you reset Ubuntu's **udev** device handling rules with the commands below?

```
sudo udevadm control --reload-rules
udevadm trigger
```

MartyG-RealSense, 2023-08-08:
It's great to hear that you have made significant progress.

There is not much information to refer to regarding multi machine communication with ROS1 and RealSense other than the guide at the link below.

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

In that guide, they recommend opening **rosscore** and setting a **ROS_MASTER_URI** for the ROS wrapper launch in two separate terminals.  Are you using multiple terminals or a single one, please?

loubron23, 2023-08-08:
> It's great to hear that you have made significant progress.
>
> There is not much information to refer to regarding multi machine communication with ROS1 and RealSense other than the guide at the link [...]

I tried his method, but the result did not change. I don't think this error was caused by a problem with my ROS communication settings. Because when I use other ros packages, there will not be such automatic node shutdown errors.

loubron23, 2023-08-09:
OK! I know why the error of node automatic shutdown occurred, and it is still because of the IP setting issue in ROS multi machine communication. I am very sorry for taking up so much of your time to answer my question, but in fact, it was not due to issues with librealsense or ROS wrapper. I took the wrong direction
But now I have a new problem, when I use 'roslaunch realsense2_camera rs_camera.launch  initial_reset:=true’, the node will start and maintain normally. But when I open rviz and try to open /camera/color/image, the node will suddenly shut down and report an error. However, I am now able to use librealsense and ros wrapper normally in a docker, and this error occurs outside docker.

... logging to /home/neousys/.ros/log/91fbfa5a-367e-11ee-9547-003018050a5a/roslaunch-neousys-Nuvo-7000-Series-10550.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://neousys-Nuvo-7000-Series:36261/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 577 lines cut ...]
7fdcd879c000-7fdcd879f000 r--p 00063000 08:02 9442291                    /opt/ros/kinetic/lib/libnodeletlib.so
7fdcd879f000-7fdcd87a0000 rw-p 00066000 08:02 9442291                    /opt/ros/kinetic/lib/libnodeletlib.so
7fdcd87a0000-7fdcd87c6000 r-xp 00000000 08:02 13371200                   /lib/x86_64-linux-gnu/ld-2.23.so
7fdcd87d6000-7fdcd8980000 rw-p 00000000 00:00 0 
7fdcd8991000-7fdcd8992000 rw-p 00000000 00:00 0 
7fdcd8992000-7fdcd89b0000 r-xp 00000000 08:02 13370047                   /lib/x86_64-linux-gnu/libudev.so.1.6.4
7fdcd89b0000-7fdcd89b1000 r--p 0001d000 08:02 13370047                   /lib/x86_64-linux-gnu/libudev.so.1.6.4
7fdcd89b1000-7fdcd89b2000 rw-p 0001e000 08:02 13370047                   /lib/x86_64-linux-gnu/libudev.so.1.6.4
7fdcd89b2000-7fdcd89c5000 rw-p 00000000 00:00 0 
7fdcd89c5000-7fdcd89c6000 r--p 00025000 08:02 13371200                   /lib/x86_64-linux-gnu/ld-2.23.so
7fdcd89c6000-7fdcd89c7000 rw-p 00026000 08:02 13371200                   /lib/x86_64-linux-gnu/ld-2.23.so
7fdcd89c7000-7fdcd89c8000 rw-p 00000000 00:00 0 
7ffdcef37000-7ffdcef56000 rwxp 00000000 00:00 0                          [stack]
[... cut ...]
[camera/realsense2_camera_manager-1] process has died [pid 10559, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/neousys/.ros/log/91fbfa5a-367e-11ee-9547-003018050a5a/camera-realsense2_camera_manager-1.log].
log file: /home/neousys/.ros/log/91fbfa5a-367e-11ee-9547-003018050a5a/camera-realsense2_camera_manager-1*.log
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/neousys/.ros/log/91fbfa5a-367e-11ee-9547-003018050a5a/camera-realsense2_camera-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2023-08-09:
There has only been one past reported RealSense case of the **corrupted size vs. prev_size** error at https://github.com/IntelRealSense/realsense-ros/issues/934 which was also on Kinetic and Ubuntu 16.04.

In that case, it was thought that it was also related to temperature, with one RealSense user commenting that their camera's ASIC temperature (the internal temperature inside the camera) was in the high 40's and the camera casing was hot when touched.  

Does your own camera feel hot to the touch?

MartyG-RealSense, 2023-08-09:
Thank you.  After the error occurs and the launch exits, is the camera able to launch if you repeat the roslaunch instruction?

loubron23, 2023-08-09:
Yes, the camera can be launched. But it will still occur errors if I use rviz to show the image.

MartyG-RealSense, 2023-08-09:
I cannot find a reason why the crash would be occurring outside Docker.

Is it practical for your project to launch from within Docker, since you do not experience the errors when using Docker?

loubron23, 2023-08-09:
It's not a big problem. Next, I plan to use it in Docker. Thank you very much for your patient and friendly help!

closed 2023-08-22 by MartyG-RealSense