realsenseai/realsense-ros #2819: !Using the rqt_image_view command does not display the image

EGOIST5, 2023-07-23:
I use one terminal to enter "roslaunch realsense2_camera rs_camera.launch" command, and then another terminal to enter "rqt_image_view" command, but the expected results are not displayed.
[image]
[image]

Before I enter the rqt_image_view command, the first terminal only shows the information up to malloc(): invalid size(unsorted).

EGOIST5, 2023-07-23:
Here are the details：
wzy@ubuntu:~/Fast-Drone-250-master$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/wzy/.ros/log/656d624a-2967-11ee-9fb2-845cf326b7db/roslaunch-ubuntu-7185.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:39287/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: False
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 76 lines cut ...]
[ INFO] [1690123458.915362082]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ WARN] [1690123459.118051978]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1690123459.142008956]: Done Setting Dynamic reconfig parameters.
[ INFO] [1690123459.145867255]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1690123459.147070392]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1690123459.148351383]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1690123459.152430509]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1690123459.152613992]: setupPublishers...
[ INFO] [1690123459.166119849]: Expected frequency for depth = 30.00000
[ INFO] [1690123459.177305287]: Expected frequency for infra1 = 30.00000
[ INFO] [1690123459.184509195]: Expected frequency for infra2 = 30.00000
[ INFO] [1690123459.194210447]: Expected frequency for color = 30.00000
[ INFO] [1690123459.205493290]: setupStreams...
[ INFO] [1690123459.409072864]: SELECTED BASE:Depth, 0
[ INFO] [1690123459.452754161]: RealSense Node Is Up!
[ WARN] [1690123459.490632153]: 
malloc(): invalid size (unsorted)
[camera/realsense2_camera_manager-2] process has died [pid 7229, exit code -6, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/wzy/.ros/log/656d624a-2967-11ee-9fb2-845cf326b7db/camera-realsense2_camera_manager-2.log].
log file: /home/wzy/.ros/log/656d624a-2967-11ee-9fb2-845cf326b7db/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/wzy/.ros/log/656d624a-2967-11ee-9fb2-845cf326b7db/camera-realsense2_camera-3*.log

EGOIST5, 2023-07-23:
I did some more experiments：use "roslaunch realsense2_camera rs_camera.launch filters:=pointcloud" command,then also use  "rqt_image_view" command.
here are the details：wzy@ubuntu:~/Fast-Drone-250-master$ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
... logging to /home/wzy/.ros/log/e0b08896-2968-11ee-8ad7-845cf326b7db/roslaunch-ubuntu-10738.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:40803/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: False
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 77 lines cut ...]
[ INFO] [1690124095.174606867]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ WARN] [1690124095.422168468]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1690124095.464771187]: Done Setting Dynamic reconfig parameters.
[ INFO] [1690124095.474752615]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1690124095.476771333]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1690124095.480661724]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1690124095.484594741]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1690124095.484878494]: setupPublishers...
[ INFO] [1690124095.493085403]: Expected frequency for depth = 30.00000
[ INFO] [1690124095.504608605]: Expected frequency for infra1 = 30.00000
[ INFO] [1690124095.513521039]: Expected frequency for infra2 = 30.00000
[ INFO] [1690124095.522568454]: Expected frequency for color = 30.00000
[ INFO] [1690124095.537292779]: setupStreams...
[ INFO] [1690124095.720481072]: SELECTED BASE:Depth, 0
[ INFO] [1690124095.771990945]: RealSense Node Is Up!
[... cut ...]
[camera/realsense2_camera_manager-2] process has died [pid 10772, exit code -6, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/wzy/.ros/log/e0b08896-2968-11ee-8ad7-845cf326b7db/camera-realsense2_camera_manager-2.log].
log file: /home/wzy/.ros/log/e0b08896-2968-11ee-8ad7-845cf326b7db/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/wzy/.ros/log/e0b08896-2968-11ee-8ad7-845cf326b7db/camera-realsense2_camera-3*.log

MartyG-RealSense, 2023-07-23:
Hi @EGOIST5  The only previous report of the **malloc(): invalid size (unsorted)** problem with RealSense ROS is at https://github.com/IntelRealSense/realsense-ros/issues/2656

In that particular issue, I suggested performing a clean wipe and reinstall at https://github.com/IntelRealSense/realsense-ros/issues/2656#issuecomment-1467883440 in case the librealsense SDK had been installed on the computer more than once, and a clean reinstall made the malloc(): invalid size (unsorted) problem disappear.

EGOIST5, 2023-07-24:
Thanks for your answer, I have successfully solved the problem

closed 2023-07-24 by EGOIST5

MartyG-RealSense, 2023-07-24:
You are very welcome.  It's excellent to hear that you achieved a solution.  Thanks very much for the update!