realsenseai/realsense-ros #2831: Realsense d435 Bad Quality

gurselturkeri, 2023-08-01:
Hi, I would like to use d435 for object detection but quality is really bad. Is there any settings to better quality.
[image]

MartyG-RealSense, 2023-08-01:
Hi @gurselturkeri  You could try maximizing the RGB **Sharpness** to make the objects clearer, like in the example of Python code for doing so at https://github.com/IntelRealSense/librealsense/issues/12030#issuecomment-1647640046

gurselturkeri, 2023-08-01:
> Hi @gurselturkeri You could try maximizing the RGB **Sharpness** to make the objects clearer, like in the example of Python code for doing so at [IntelRealSense/librealsense#12030 (comment)](https://github.com/IntelRealSense/librealsense/issues/12030#issuecomment-1647640046)

How can I do that with using CLI? Is it correct?: `roslaunch realsense2_camera rs_camera.launch sharpness:=100`

MartyG-RealSense, 2023-08-01:
That instruction should work for setting the sharpness parameter.  You could also set it by loading a **json** camera configuration file into the launch, either by defining a **json_file_path** inside the launch file or in the roslaunch instruction.

closed 2023-08-16 by MartyG-RealSense