realsenseai/realsense-ros #2832: D435i RGB Problem when pointcloud.enable set to true

BlackFanta, 2023-08-01:
### **Infos**

    LibRealSense v2.54.1
    ROS2 Humble
    Ubuntu Mate Jammy 22.04.2
    Rapsberry Pi 4 4GB
    Kernel 5.15.0-1034-raspi
    D435i

### **Problem**

    
 I followed the instructions on installing librealsense and the ros wrapper for humble i've tested it on my local machine and it works fine, when I tried to test it on the raspberry I encountered problems when enabling pointcloud. I am running the ros2 realsense node with the following command :

 `ros2 launch realsense2_camera rs_launch.py rgb_camera.profile:=640x480x30 depth_module.profile:=640x480x30 pointcloud.enable:=true log_level:=debug initial_reset:=true`

The problem is that the pointcloud seems to only be published once, after that the color topic stops publishing and is no longer read and only the depth module keeps working and is constantly published.

Here are the full logs after the camera has been started.
[log.txt](https://github.com/IntelRealSense/realsense-ros/files/12231353/log.txt)

I noticed that after the first pointcloud is generated and displayed on rviz this error is produced

`[realsense2_camera_node-1] [WARN] [1690903147.334993962] [camera.camera]: XXX Hardware Notification:Depth stream start failure,1.6909e+12,Error,Hardware Error
[realsense2_camera_node-1] [WARN] [1690903147.335256884] [camera.camera]: Hardware Notification:Depth stream start failure,1.6909e+12,Error,Hardware Error`

The color is published only once more after the error and later on only the depth is published.

I've tried launching another node with pointcloud.enable set to false and the camera seems to be publishing all topics correctly.

BlackFanta, 2023-08-02:
Thanks for the reply!

I've tested using ros2 run, no pointcloud received, and the following error keeps getting generated.

` 02/08 10:16:22,556 ERROR [281473063839968] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
` 

I tried lowering the resolution of both the rgb_camera and depth_module down to 640x380x30 to see if it was a network bandwidth problem. I've gotten similar results, only now the /depth/image_rect_raw topic seems to not contain any messages and the frequency of /color/image_raw is very low compared to the desired publish rate.

```
ros2 topic hz /color/image_raw
average rate: 3.721
	min: 0.235s max: 0.399s std dev: 0.05831s window: 6
average rate: 3.887
	min: 0.235s max: 0.399s std dev: 0.04490s window: 11
average rate: 3.965
	min: 0.230s max: 0.399s std dev: 0.03812s window: 16
average rate: 3.913
	min: 0.230s max: 0.399s std dev: 0.04070s window: 20
average rate: 3.944
	min: 0.230s max: 0.399s std dev: 0.03678s window: 25
average rate: 3.965
	min: 0.230s max: 0.399s std dev: 0.03371s window: 30
average rate: 3.980
	min: 0.230s max: 0.399s std dev: 0.03135s window: 35
average rate: 3.988
	min: 0.230s max: 0.399s std dev: 0.02936s window: 40
average rate: 4.000
	min: 0.230s max: 0.399s std dev: 0.02786s window: 45
average rate: 4.008
	min: 0.230s max: 0.399s std dev: 0.02648s window: 50
```

This is what's shown on rviz

[image]

MartyG-RealSense, 2023-08-02:
Thanks very much for the information.  Does it make a difference if you add **auto_exposure_priority:=false** to your ros2 launch instruction?  When both depth and color streams are being published and auto_exposure_priority (an RGB option) is set to false then it forces the FPS speed to be constant for both streams instead of the FPS being permitted to vary.

BlackFanta, 2023-08-02:
Hello,

I've tried what you have suggested, it seems there is a slight difference from what I've been experiencing as it appears point clouds are being generated for about a second after I've started up the camera contrary to the single point cloud I used to receive before. Only this time, I get these two following warnings 
```
[realsense2_camera_node-1] [WARN] [1690987093.943378564] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1] [WARN] [1690987099.916837486] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
```
Afterwards, color is no longer published and only depth is being published.

These are the full logs:
```

ros2 launch realsense2_camera rs_launch.py rgb_camera.profile:=640x480x15 depth_module.profile:=640x480x15 pointcloud.enable:=true initial_reset:=true auto_exposure_priority:=false
[INFO] [launch]: All log files can be found below /home/digitin/.ros/log/2023-08-02-15-38-06-033388-digitin-desktop-4409
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [4410]
[realsense2_camera_node-1] [INFO] [1690987086.864784031] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1690987086.865087198] [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1690987086.865159827] [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1690987086.949428142] [camera.camera]: Device with serial number 233622071919 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1690987086.949614272] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1690987086.949675994] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1690987086.951017661] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1690987086.951167790] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1690987086.951220235] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1690987093.098510231] [camera.camera]: Device with serial number 233622071919 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1690987093.098674602] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1690987093.098733935] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1690987093.099808694] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1690987093.099957657] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1690987093.104002694] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1690987093.107015398] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1690987093.107138916] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1690987093.107193787] [camera.camera]: Device Serial No: 233622071919
[realsense2_camera_node-1] [INFO] [1690987093.107244916] [camera.camera]: Device physical port: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1690987093.107294194] [camera.camera]: Device FW version: 5.13.0.55
[realsense2_camera_node-1] [INFO] [1690987093.107341972] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1690987093.107387324] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1690987093.166124342] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1690987093.178661546] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1]  02/08 15:38:13,180 WARNING [281473013639392] (backend-v4l2.cpp:1855) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[realsense2_camera_node-1] [WARN] [1690987093.202402324] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1690987093.205301676] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1690987093.205673713] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1690987093.248634713] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1690987093.259993657] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1690987093.291026398] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1690987093.291986138] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1690987093.298075546] [camera.camera]: Starting Sensor: RGB Camera
[... cut ...]
[realsense2_camera_node-1] [INFO] [1690987134.275324373] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [INFO] [1690987134.281122744] [camera.camera]: Close Sensor - Done. 
[realsense2_camera_node-1] [INFO] [1690987134.281213484] [camera.camera]: Stop Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1690987134.281245207] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [INFO] [1690987135.370898002] [camera.camera]: Close Sensor - Done. 
[INFO] [realsense2_camera_node-1]: process has finished cleanly [pid 4410]

```

MartyG-RealSense, 2023-08-03:
**No stream match for pointcloud chosen texture Process - Color** means that the color frames are missing at the particular moment in time that the message is occurring.

Let's try using the **infrared** stream for the pointcloud texture instead of the color stream using the parameter **pointcloud.stream_filter** and see if it makes a difference.

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x15 rgb_camera.profile:=640x480x15 pointcloud.enable:=true log_level:=debug initial_reset:=true auto_exposure_priority:=false enable_infra1:=true pointcloud.stream_filter:=1`

BlackFanta, 2023-08-07:
Hello again,

I have tried your suggestion, I'm now getting this warning:
`
[WARN] [1691401229.556725538] [camera.camera]: No stream match for pointcloud chosen texture Process - Depth`

The thing is now depth and color are being published but infrared is not and no pointcloud is generated whatsoever.

This is the list of topics

```
ros2 topic list
/camera/color/camera_info
/camera/color/image_raw
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_depth
/camera/extrinsics/depth_to_infra1
/camera/imu
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra1/metadata

```
/camera/color/image_raw and /camera/depth/image_rect_raw are being published correctly and visualized in rviz while /camera/infra1/image_rect_raw is not. It seems that there is always a problem with the selected texture of pointclouds that stops their respective topics of being sent, such was the case with color and now with infra.

MartyG-RealSense, 2023-08-07:
This seems to be an issue specific to Raspberry Pi as it works on your local machine. Usually when there is a Pi-specific issue such as this, there is unfortunately not a solution other than to use a single-board computer other than a Pi.

MartyG-RealSense, 2023-08-13:
Hi @BlackFanta  Bearing in mind the advice in the comment above, do you require further assistance with this case please?  Thanks!

MartyG-RealSense, 2023-08-20:
Case closed due to no further comments received.

closed 2023-08-20 by MartyG-RealSense

SwiftGust, 2023-12-18:
I also have very similar color stream problem in Jetson Xavier NX with ROS Noetic, SDK 2.50.0
with pointcloud such as examples of demo_pointcloud.launch or rs_d435_camera_with_model.launch
rs_camera.launch seems doesn't have issue so i'm suspecting pointcloud

RGB camera streaming drop fps down to 0.0, diagnostic says "Frequency too low" to "No events recorded" for color stream, 
while depth stream seems unaffected

I have attached diagnostics log from demo_pointcloud.launch
[rs-ros-rgb-issue.txt](https://github.com/IntelRealSense/realsense-ros/files/13699297/rs-ros-rgb-issue.txt)

MartyG-RealSense, 2023-12-18:
Hi @SwiftGust  **rs_camera.launch** does not have the pointcloud enabled by default.  It can be enabled in the roslaunch instruction if you have not done so already.

`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`

*********

If the rs_camera launch file has problems with the pointcloud after enabling it, you could try am **RGBD** launch.  This is a 2-step process.

1.  Enable support for an RGBD launch using the command below.  You only need to input this command once and not on subsequent RGBD launches.

`sudo apt-get install ros-noetic-rgbd-launch`

2.  Initiate an RGBD launch with pointcloud enabled by default using the roslaunch instruction below.

`roslaunch realsense2_camera rs_rgbd.launch`

rs_rgbd.launch should generate a point cloud that is published at **depth_image_proc**

MartyG-RealSense, 2023-12-19:
Hi @SwiftGust  When rs_camera has pointcloud problems but rs_rgbd works, then using rs_rgbd is usually the best available option.  This is because development of the ROS1 wrapper has ceased and so is not updated with fixes.  Development is now focused on the ROS2 wrapper.