realsenseai/realsense-ros #2835: reset camera

hfdlwt, 2023-08-02:
Hello everyone ！ I have a question about multiple camera such as D435 and T265 ， Someone knows how to restart one of the cameras when they are all running ？

MartyG-RealSense, 2023-08-02:
Hi @hfdlwt  A C++ hardware reset script at the link below will cycle through all attached cameras.

If a specific camera in a set of multiple cameras needs to be reset then https://github.com/IntelRealSense/librealsense/issues/5428 describes how to perform this in Python using a camera serial number.

hfdlwt, 2023-08-02:
Thank you for your prompt reply @MartyG-RealSense ，Can the camera continue to work after hardware restart, or does it need to restart the launch file （realsense_camera manger node）

MartyG-RealSense, 2023-08-02:
Although a reset will disconnect the camera and then re-connect it, my understanding is that a current session will continue if reconnection is achieved within a time period of 5 seconds.  If it takes longer than 5 seconds to re-establish the connection after disconnection then the connection 'times out' (hence the well known error message 'Frame didn't arrive within 5000').

It is certainly possible to reset the camera during a ROS launch without restarting the launch when using the **initial_reset:=true** parameter.

hfdlwt, 2023-08-02:
Thanks again @MartyG-RealSense , Are you talking about manually restarting the camera? What I want is to automatically restart the camera through a trigger condition in the program.

MartyG-RealSense, 2023-08-02:
That code will automatically restart the camera when the script is launched.  You can though nest a **hardware_reset()** instruction within an If code block so that if certain conditions are met then the reset command within the If condition is triggered.

closed 2023-08-16 by MartyG-RealSense