realsenseai/realsense-ros #2838: Malloc(): memory corruption

qiyueliuhuo23, 2023-08-09:
I was using `roslaunch realsense2_camera rs_camera.launch initial_reset:=true`，the node was fine. But when I start my own script with cmds `rosrun yolox yolox_hat.py`, in yolox_hat.py I get the raw img from realsense camera and process them to detect safety helmet, and **Malloc(): memory corruption** occurred.
I am using :
realsense D435i
Ubuntu 18.04 5.4.0-150-generic
ros-melodic

Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://zt-System-Product-Name:45477/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters:
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
[... 113 lines cut ...]
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/threadpool.py", line 218, in run
    result = cmd(*args)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/master_api.py", line 210, in publisher_update_task
    ret = xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1311, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1459, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 1095, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 898, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 860, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 837, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 575, in create_connection
    raise err
error: [Errno 111] Connection refused

MartyG-RealSense, 2023-08-09:
Hi @qiyueliuhuo23  ROS1 wrapper 2.3.2 should be used with librealsense version **2.51.1** and camera firmware version **5.13.0.50** instead of librealsense 2.54.1 and firmware 5.12.15.50.   

I note though that you were able to launch successfully with 2.54.1 / 5.12.15.50.

Are you starting your yolox script first and launching the ROS wrapper secondly afterwards?  If you are, does it make a difference if you launch the ROS wrapper first and your yolox script second?

MartyG-RealSense, 2023-08-10:
Your librealsense source code installation procedure looks fine.

To remove a previous librealsense source code installation, go to the 'build' folder of that installation and input the instruction below.

`sudo make uninstall && make clean`

*********

In regard to changing the firmware version, if the flag `-DBUILD_GRAPHICAL_EXAMPLES=true `is not included in your CMake build instruction then graphics support and graphics-based tools and examples will not be built, meaning that the RealSense Viewer tool cannot be used to change the firmware (which is the easiest method of doing it).  

Instead, you will need to use the text-based **rs-fw-update** tool which should be included in your librealsense build because you included the `-DBUILD_EXAMPLES=true` flag to include text-based tools and examples.

https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update

The first step would be to visit the firmware archive page and download to your computer the firmware **.bin** file for the particular firmware version to update to (5.13.0.50, which has a .bin file named **Signed_Image_UVC_5_13_0_50.bin**.

https://dev.intelrealsense.com/docs/firmware-releases

Then launch the rs-fw-update tool with the command **rs-fw-update -l**

This will list the details of your camera, including its serial number which must be noted down.

Once you have the serial number, input the command below to update the firmware version of the camera to 5.13.0.50.

`rs-fw-update -s [serial number] -f Signed_Image_UVC_5_13_0_50.bin`

For example, if the serial number was 725112060411:

`rs-fw-update -s 725112060411 -f Signed_Image_UVC_5_13_0_50.bin`

qiyueliuhuo23, 2023-08-10:
Thanks for teaching me how to uninstall librealsense and change fireware version. Follow your comments, I got the ROS1 wrapper  2.3.2 with librealsense version 2.51.1 and camera firmware version 5.13.0.50, but I got the same Malloc(): memory corruption as before.
Here is how I start my yolox script:
1. First terminal: `ros core`
2. Second terminal: `roslaunch realsense2_camera rs_camera.launch initial_reset:=true`
3. Third terminal: `rosrun yolox_ros yolox_hat.py`

The second terminal works fine before I type the command in third terminal.
After I type the command `rosrun yolox_ros yolox_hat.py`, the second terminal got  **Malloc(): memory corruption** 
Here are the log in second terminal:

> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
>
> started roslaunch server [...]

>NODES
> /camera/
>   realsense2_camera (nodelet/nodelet)
>    realsense2_camera_manager (nodelet/nodelet)
>
>ROS_MASTER_URI=http://zt-System-Product-Name:11311
>
>process[camera/realsense2_camera_manager-1]: started with pid [23923]
>process[camera/realsense2_camera-2]: started with pid [23924]
>[ INFO] [1691670080.640062880]: Initializing nodelet with 12 worker threads.
>[ INFO] [1691670080.716349953]: RealSense ROS v2.3.2
>[ INFO] [1691670080.716373915]: Built with LibRealSense v2.51.1
>[ INFO] [1691670080.716382074]: Running with LibRealSense v2.51.1
>[ INFO] [1691670080.730833353]:  
>[ INFO] [1691670080.847351938]: Device with serial number 108322071393 was found.
>
>[ INFO] [1691670080.847404495]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2->1:1.0/video4linux/video0 was found.
>[ INFO] [1691670080.847435931]: Device with name Intel RealSense D435I was found.
>[ INFO] [1691670080.847868396]: Device with port number 2-1 was found.
>[ INFO] [1691670080.847900245]: Device USB type: 3.2
>[ INFO] [1691670080.847938380]: Resetting device...
>[ INFO] [1691670086.938909205]:  
>[ INFO] [1691670087.053365423]: Device with serial number 108322071393 was found.
>
>[ INFO] [1691670087.053409254]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2->1:1.0/video4linux/video0 was found.
>[ INFO] [1691670087.053419639]: Device with name Intel RealSense D435I was found.
>[ INFO] [1691670087.053690394]: Device with port number 2-1 was found.
>[ INFO] [1691670087.053705543]: Device USB type: 3.2
>[ INFO] [1691670087.057145165]: getParameters...
>[ INFO] [1691670087.078606894]: setupDevice...
>[ INFO] [1691670087.078619145]: JSON file is not provided
>[ INFO] [1691670087.078625583]: ROS Node Namespace: camera
>[ INFO] [1691670087.078636238]: Device Name: Intel RealSense D435I
>[ INFO] [1691670087.078641563]: Device Serial No: 108322071393
>[ INFO] [1691670087.078648106]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2->1:1.0/video4linux/video0
>[ INFO] [1691670087.078654937]: Device FW version: 05.13.00.50
>[ INFO] [1691670087.078661358]: Device Product ID: 0x0B3A
>[ INFO] [1691670087.078671691]: Enable PointCloud: On
>[ INFO] [1691670087.078677463]: Align Depth: On
>[ INFO] [1691670087.078683781]: Sync Mode: On
>[ INFO] [1691670087.078741356]: Device Sensors:
>[ INFO] [1691670087.083375407]: Stereo Module was found.
> 10/08 20:21:27,083 WARNING [140580471084800] (backend-v4l2.cpp:1444) Pixel format 36315752-1a66-a242-9065-d01814a >likely requires patch for fourcc code RW16!
>[ INFO] [1691670087.094883396]: RGB Camera was found.
>[ INFO] [1691670087.095241269]: Motion Module was found.
>[ INFO] [1691670087.095258044]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
>[ INFO] [1691670087.096097515]: Add Filter: pointcloud
>[ INFO] [1691670087.097959540]: num_filters: 2
>[ INFO] [1691670087.097977212]: Setting Dynamic reconfig parameters.
>[ WARN] [1691670087.403065453]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { >>{50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[> INFO] [1691670087.411086089]: Done Setting Dynamic reconfig parameters.
>[ INFO] [1691670087.411676717]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
>[ INFO] [1691670087.412837660]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
>[ INFO] [1691670087.414236381]: setupPublishers...
>[ INFO] [1691670087.416081785]: Expected frequency for depth = 30.00000
>[ INFO] [1691670087.441834841]: Expected frequency for color = 30.00000
>[ INFO] [1691670087.449782549]: Expected frequency for aligned_depth_to_color = 30.00000
>[ INFO] [1691670087.456295817]: setupStreams...
>[ INFO] [1691670087.519406711]: SELECTED BASE:Depth, 0
>[ INFO] [1691670087.539117316]: RealSense Node Is Up!
>[ WARN] [1691670087.603168344]:
>malloc(): memory corruption
>[camera/realsense2_camera_manager-1] process has died [pid 23923, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet >manager __name:=realsense2_camera_manager __log:=/home/zt/.ros/log/c0d8fb8a-3777-11ee-9672-581122d9af9c/camera->realsense2_camera_manager-1.log].
>log file: /home/zt/.ros/log/c0d8fb8a-3777-11ee-9672-581122d9af9c/camera-realsense2_camera_manager-1*.log
>[camera/realsense2_camera-2] process has finished cleanly
>log file: /home/zt/.ros/log/c0d8fb8a-3777-11ee-9672-581122d9af9c/camera-realsense2_camera-2*.log
>all processes on machine have died, roslaunch will exit
>shutting down processing monitor...
[... cut ...] essing monitor complete
>done

After I change the librealsense SDK to 2.51.1 and camera firmware version to 5.13.0.50, I cannot got the right execution of `roslaunch realsense2_camera rs_camera.launch initial_reset:=true` with a different error. Then I open the folder of ROS1 wrapper and type `catkin_make`, then I can got the **Malloc(): memory corruption** as before.
So do you have any suggestion for me to solve the error of **Malloc(): memory corruption** ? Any suggestion will be very helpful!

qiyueliuhuo23, 2023-08-12:
Thanks for giving me several suggestions. After 2 days effort, I rollback the version of my project with no errors.  Because the project works fine with no errors at first, but someone not me changed the msg and the Malloc error appears. 
**This experience told us the importance of version controll.** 
Anyway, thanks for your answers @MartyG-RealSense, I appreciate those advices and wish you have a good day!