realsenseai/realsense-ros #2843: D455 : using RGB stream increase pointcloud publication latency

Milnaye, 2023-08-10:
| Required Info   | |
|----------|:-------------:|
| Camera Model | D455 |
| Firmware Version | 5.14.0.0 |
| Operating System & Version | custom Linux |
| Kernel Version (Linux Only) | 4.9.253 |
| Platform | NVIDIA Jetson nano |
| SDK Version | version 2.53.1 |
| ROS platform | Humble |
| Language | C and python |
| Segment | Robot | 

We enabled rgb, infra1 and pointcloud (depth) stream, at 5hz. 

### Issue description :

While working on the latency between the cameras and a node receiving the pointclouds, we found that using the rgb stream led the latency to increase.

### Reason :

Enabling the pointcloud stream automaticaly activate the "enable_sync" option : 
```
enable_sync:
    gathers closest frames of different sensors, infra red, color and depth, to be sent with the same timetag.
    This happens automatically when such filters as pointcloud are enabled.
```

This options make the different frames (color, infrared and depth) to move from librealsense to realsense-ros as a set. and every frame in the set take the timestamp of the color one (if present).
It's made like this intentionally, the timestamp uniformity helps user to sync frames, and the frames moving as a set is needed for the "align_depth" option.

However, this leads to two side effect that we don't want : 
-First is that the pointcloud messages doesn't have a precise timestamp (header.timestamp of the published pointclouds) as the color and infra1 sensors are not synchonized. (infra1 and depth share the same timestamp and depth is calculated from infrared, so having the infra1 stream enabled does not cause us issue).
For this we could manually change the realsense-ros library to use the frame.timestamp of the pointcloud stream and have the correct timestamp.
-The second issue is that librealsense wait for both depth and rgb frame to be ready to send them, and at 5 Hz, the wait can be long (values comes below).
We tried to simply prevent the use of enable_sync inside realsense-ros, which breaks things.

### Values

For thoses graphs, we added log inside the realsense-ros library and gathered the results.
All the results are in milliseconds, the small vertical lines represent min and max, the bigger vertical line the median, and the box represents the first and last quartile.

First side effect :
[image]
_difference between assigned vs true timestamp, which is the difference between frame.timestamp of color and depth stream (depth - color)_
Without color stream, the assigned timestamp (header.stamp of the ros messages) and true timestamp (frame.timestamp of the rs2::frame object) are the same.

Second side effect : 
[image]
_time between the pointcloud timestamp and its reception by a subscribing node with color stream enabled (using the true timestamp)_

[image]
_time between the pointcloud timestamp and its reception by a subscribing node with color stream disabled (using the true timestamp)_

Our system needs to be reactive, we can't afford to wait this extra time. We don't want neiher to disable the color stream, as we use it in parallele.  

### Possibles solutions : 
Remove the automatic use of "enable_sync" when using pointcloud (make it to be enable when using align depth option).
We also checked in librealsense library to see if it was possible to sort of change the waiting time for a set of frame, with no success.

MartyG-RealSense, 2023-08-10:
Hi @Milnaye  Do you experience the same problem if you add the instruction below to your `ros2 launch` command so that the pointcloud is textured with the infrared stream (which is perfectly synced with depth) instead of the color stream?  

`pointcloud.stream_filter:=1`

The color stream should still continue to be published even if it is not being used for texturing the pointcloud.

You could also try disabling **global time**, another timestamp synching mechanism that is enabled by default on D455.

`global_time_enabled:=false`

MartyG-RealSense, 2023-08-16:
Hi @Milnaye  Do you require further assistance with this case, please?  Thanks!

Milnaye, 2023-08-16:
Hi @MartyG-RealSense,

I tried the two commands you proposed. 
For the `pointcloud.stream_filter:=1`, as we initially don't even use the color stream to add texture to our pointclouds it doesn't solve our issue. We are already using the `pointcloud.stream_filter:=0`.
It's actually part of the issue as the added latency comes from the fact that librealsense try to send frames as a set in anticipation of a possible texturing, which cause side the effects we don't want. 

And for the `global_time_enabled:=false`, it doesn't change the first side effect which is that the timestamp considered for the pointclouds stream use the one from the color stream.
The latency (difference between the time of reception in another node and the considered pointcloud timestamp) increased by a lot and is quite unstable.
It certainly isn't the true latency but an error due to an incorrect timestamp correction, the actual latency shouldn't have changed. We need to estimate an accurate latency, so we can't disable `global_time`.

MartyG-RealSense, 2023-08-16:
Having global_time enabled is not as important in ROS as it is in librealsense because ROS has its own drift compensation mechanism that still operates if global_time is disabled or it is not supported, as ROS times are usually based on a time-stable machine.

Is enable_sync still active if you disable it in the launch **after** the pointcloud is enabled?

`ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true enable_sync:=false`

Milnaye, 2023-08-16:
The parameter enable_sync is already disabled, but it doesn't matter as the sync activation is overridden by the use of pointcloud.enable anyways, and we need the pointcloud stream. Here is the part of the code where it happen, if I understand correctly: https://github.com/IntelRealSense/realsense-ros/blob/25a1191187de8b2d0587ab987ce36b6288a5a0ee/realsense2_camera/src/rs_node_setup.cpp#L114C7-L114C7

What we need is to enable pointcloud and color stream, without having sync mechanism automatically activated. We weren't able to just bypass the sync usage, it should be easily reproducible on your end if you want to try it.

MartyG-RealSense, 2023-08-16:
My suggestion for putting enable_sync after the pointcloud command was to see if enable_sync was made true by the pointcloud command and then immediately made false again by the subsequent sync disable command.

The only other suggestion I can think of is to launch and then after launch has completed, set the enable_sync parameter to false with a **ros2 param set** command during runtime to see if it disables sync of the pointcloud.

https://github.com/IntelRealSense/realsense-ros#parameters-that-can-be-modified-during-runtime

For example, something like:

`ros2 param set /camera/camera depth_module.enable_sync false`

Milnaye, 2023-08-16:
Doing a `ros2 param get /camera/camera enable_sync` already return `False`, the issue isn't here.
The parameter enable_sync stay at `false` even during runtime.

When doing a `ros2 param list /camera/camera`, I don't see the "depth_module.enable_sync" you're mentioning.

MartyG-RealSense, 2023-08-16:
The `camera/camera` command was a best-guess suggestion about the path of the enable_sync parameter (I do not have ROS on my support computer and so cannot run tests).  To see the entire parameter list, you can simply do **ros2 param list**.  But I note that you have confirmed that enable_sync is already false after launch by using `param get`.

Your remainng option may be to edit the wrapper code as you suggested yourself.  I apologize that I cannot be of more help on this particular question.

Milnaye, 2023-08-16:
We already tried to modify the wrapper ros and search for the root cause in librealsense with no success. The help from the development team is indeed required.
Thanks for your help.

MartyG-RealSense, 2023-08-16:
I will submit a message to the team in the coming week.  Thanks so much for your patience.

MartyG-RealSense, 2023-08-30:
Hi @Milnaye   Thanks very much for your patience.  The ROs development team responded: 

"We are using sync frames for filters (even if enable_sync parameter is false), because when align_depth and pointcloud are running, we need both depth and color frames to be synced, in order to texture the pointcloud, and in order to align the depth to color."

They asked if you are using any other filters in addition to the pointcloud filter.

closed 2023-09-11 by MartyG-RealSense