realsenseai/realsense-ros #2848: Point cloud not published in rs_camera file

furkancetiny, 2023-08-15:
I can't get pointcloud output even though I set `enable_pointcloud=true` in rs_camera file. Can you help about the reason?

MartyG-RealSense, 2023-08-15:
Hi @furkancetiny  The ROS1 wrapper parameter `enable_pointcloud` has been deprecated for years.

Is a pointcloud generated if you set it as a **filter** in the rs_camera.launch roslaunch instruction, please?

`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`

furkancetiny, 2023-08-15:
Unfortunately, it did not form a point cloud. Issue `/camera/color/image_raw`  stopped publishing

furkancetiny, 2023-08-15:
A topic named `camera/depth_registered/points` has been published, but I couldn't get a printout in the rviz environment.

MartyG-RealSense, 2023-08-15:
So you are unable to view `camera/depth_registered/points` if you set it as the topic in RViz like in https://github.com/IntelRealSense/realsense-ros/issues/553

What is your librealsense SDK and RealSense ROS wrapper versions please?  It should be librealsense **2.51.1** and ROS1 wrapper **2.3.2**

furkancetiny, 2023-08-15:
I'm working on the rtabmap package. For this package I need the `/camera/color/image_raw` and `/camera/depth/color/points` topics.

ROS version: 2.3.2
Librealsense version: 2.50.0
Driver version: 5.13.00.50

MartyG-RealSense, 2023-08-20:
Hi @furkancetiny  Do you require further assistance with this case, please?  Thanks!

furkancetiny, 2023-08-21:
I'm sorry for the late reply. `roslaunch realsense2_camera opensource_tracking.launch filters:=pointcloud` When I ran this command, I was able to get pointcloud output via `/voxel_cloud`. 

I don't need help right now. Thank you.

MartyG-RealSense, 2023-08-21:
It's no problem at all.  I'm pleased to hear that you were able to achieve a pointcloud.  As you do not requre further assistance, I will close this case.  Thanks again!

closed 2023-08-21 by MartyG-RealSense