realsenseai/realsense-ros #2851: ROS-Noetic-Navigation using move_base

Murali25111993, 2023-08-21:
Hello I am having a 4 wheel differential drive robot using Intel realsense d455 camera. I have done SLAM using this [link](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i). It is successful. I have created a map of my working environment. Now I want to do Navigation/Path planning using the generated map.

Setup:
Differential wheel drive
Jetson Tx2
Arduino (Roboclaw)(/cmd_vel)<-----> Jetson (Rosserial)
Rover is responding to rostopic /cmd_vel.

**Note: I am not using any 360 degree LIDAR**

Can someone who did the same setup achieved Navigation, if so Please guide me.

MartyG-RealSense, 2023-08-21:
Hi @Murali25111993  The link below has a navigation project that has similarities with your setup, as it uses diffential drive, Arduino, Rosserial and cmd_vel.  The main differences are that they are using Jetson Nano instead of TX2, ROS Melodic instead of Noetic and RealSense T265 instead of D455.  The project may be adaptable for your needs though.

https://github.com/XRobots/Navigation-Unit

Murali25111993, 2023-08-28:
Hi, 

I believe the link you have provided also has RPlidar which provides 2d map. On top of gmapping, realsense camera is used for generating map cloud. But my sole purpose is to do navigation only using d455 realsense, where I have completed slam.

Murali25111993, 2023-08-28:
After executing **roslaunch realsense2_camera opensource_tracking.launch**,

I created a map, 

[image]

[image]

Now, as per the documentation, for localizing i have to run the command **roslaunch realsense2_camera opensource_tracking.launch localization:=true**. After this, I could able to load from rtabmap.db and able to give 2D pose estimate from Rviz and 2DNavGoal.

[image]
[image]

[image]
[image]

But for navigating, I have added a line in opensource_tracking.launch

 
    <arg name="offline"          default="false"/>
    <include unless="$(arg offline)" 
        file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="align_depth" value="true"/>
        <arg name="linear_accel_cov" value="1.0"/>
        <arg name="unite_imu_method" value="linear_interpolation"/>
    </include>  
`   
    <node pkg="imu_filter_madgwick" type="imu_filter_node" name="ImuFilter">
        <param name="use_mag" type="bool" value="false" />
        <param name="_publish_tf" type="bool" value="false" />
        <param name="_world_frame" type="string" value="enu" />
        <remap from="/imu/data_raw" to="/camera/imu"/>
      </node>
    
     <include file="$(find rur_navigation)/launch/move_base.launch" />
    
    
    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
        <arg name="rgb_topic" value="/camera/color/image_raw"/>
        <arg name="depth_topic" value="/camera/aligned_depth_to_color/image_raw"/>
        <arg name="camera_info_topic" value="/camera/color/camera_info"/>
        <arg name="depth_camera_info_topic" value="/camera/depth/camera_info"/>
        <arg name="rtabmapviz" value="false"/>
        <arg name="rviz" value="true"/>
    </include>

    <include file="$(find robot_localization)/launch/ukf_template.launch"/>
    <param name="/ukf_se/frequency" value="300"/>
    <param name="/ukf_se/base_link_frame" value="base_link"/>
    <param name="/ukf_se/odom0" value="rtabmap/odom"/>
    <rosparam param="/ukf_se/odom0_config">[true,true,true,
                                            true,true,true,
                                            true,true,true,
                                            true,true,true,
                                            true,true,true]
    </rosparam>
    <param name="/ukf_se/odom0_relative" value="true"/>
    <param name="/ukf_se/odom0_pose_rejection_threshold" value="10000000"/>
    <param name="/ukf_se/odom0_twist_rejection_threshold" value="10000000"/>

    <param name="/ukf_se/imu0" value="/imu/data"/>
    <rosparam param="/ukf_se/imu0_config">[false, false, false,
                                           true,  true,  true,
                                           true,  true,  true,
                                           true,  true,  true,
                                           true,  true,  true]
    </rosparam>
    <param name="/ukf_se/imu0_differential" value="true"/>
    <param name="/ukf_se/imu0_relative" value="false"/>
    <param name="/ukf_se/use_control" value="false"/>
    
  </launch>`

continued...

Murali25111993, 2023-08-28:
In the attached zip folder below, I have made changes in local and global costmap yaml in param folder ,, also made changes in move_base.launch.

[rur_navigation.zip](https://github.com/IntelRealSense/realsense-ros/files/12451182/rur_navigation.zip)

Kindly guide me/ point me what mistake I have committed. will provide you if any documents required. 

My rover is working with /cmd_vel

Thanks

Murali

MartyG-RealSense, 2023-08-28:
It's great to hear that you made such significant progress!

My knowledge of ROS navigation is limited though unfortunately and so I do not have advice to offer about your configuration.  I do apologize.

Murali25111993, 2023-08-28:
It will be grateful ,if you direct me to the correct person who can help?

MartyG-RealSense, 2023-08-28:
I will ask my Intel RealSense colleagues on the ROS team about your question.  Thanks very much for your patience.

Murali25111993, 2023-08-31:
Any leads?

MartyG-RealSense, 2023-08-31:
My colleagues have your request but it may take some time to respond to your request due to their existing workload.

Murali25111993, 2023-08-31:
Got it, Thanks for the update.

MartyG-RealSense, 2023-09-19:
There are no updates to provide at the moment, unfortunately.  My colleagues on the RealSense ROS team responded that they are not familiar with the move_base package (navigation is not a core part of the RealSense ROS wrapper), so they would have to explore it when they have the opportunity to do so.

Murali25111993, 2023-12-09:
No worry, I have figured it out. Now I can navigate using VSLAM with realsense. Thanks

closed 2023-12-09 by Murali25111993

MartyG-RealSense, 2023-12-09:
You are very welcome.  I'm pleased to hear that you have achieved a solution.  Thanks very much for the update!