realsenseai/realsense-ros #2852: tf2_ros_py not available for Foxy so wrapper cannot be installed

jidnyesha-patil, 2023-08-21:
I am trying to install the ROS wrapper for Realsense cameras on Jetson Orin NX. The Orin NX supports only Ubuntu 20.04 so I installed ROS Foxy. The rosdep install step gives error saying 
ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies:
realsense2_camera: Cannot locate rosdep definition for [tf2_ros_py]
I realize that rosdep cannot find the package named tf2_ros_py but the there is no version of the package for foxy. 
Any explanation and solutions will be appreciated.

MartyG-RealSense, 2023-08-22:
Hi @jidnyesha-patil  Support for ROS2 Foxy and Galactic was recently removed from the RealSense ROS2 wrapper (3 weeks ago at the time of writing this), as described at https://github.com/IntelRealSense/librealsense/pull/11953 and at the wrapper documentation extract below.  The wrapper now supports ROS2 Humble and Iron.

[image]

According to an announcement from Nvidia at the link below, JetPack 6 will support Orin and Ubuntu 22.04.  JetPack 6 is due for release in Q3 2023 (Q3 is the quarter of the year between July and September) so it should be available soon.

https://forums.developer.nvidia.com/t/jetson-software-roadmap-for-2023/238889

********

If you need to use Foxy and Ubuntu 20.04 then a workaround may be to build a 'forked' version of the 4.54.1 wrapper created by a RealSense ROS user that pre-dates the 3-weeks-ago changes in the official wrapper, and use the Git address of that fork in the **git clone** installation command.  The fork for 4.54.1 at the link below dates from last month at the time of writing this.

https://github.com/MichaelKerley/realsense-ros

********

You could also build from a fork of **4.51.1**, which still has support for Foxy and Ubuntu 20.04, and match it with librealsense **2.51.1**.  Here is an example of a 4.51.1 fork (there are others available).

https://github.com/HedgeHao/realsense-ros

jidnyesha-patil, 2023-08-22:
Thank you for the explanation and solutions @MartyG-RealSense. Really appreciate the help!

closed 2023-08-22 by jidnyesha-patil

MartyG-RealSense, 2023-08-22:
You are very welcome.  I'm pleased that I could help.  :)

privvyledge, 2023-08-25:
@MartyG-RealSense's suggestion works, but if you don’t want to clone another repository (fork), you can download the last Release/Tag from this repository, i.e., 4.54.1. I have verified it is working with Foxy, and it also fixed the rosdep issue.

You could either download the zip using wget or just run `git clone --depth 1 https://github.com/IntelRealSense/realsense-ros.git -b 4.54.1`. Another option, if you have already cloned the main branch, is to git checkout that specific tag or any past tag.

MartyG-RealSense, 2023-08-26:
Thanks so much @privvyledge for your advice to @jidnyesha-patil  :)