realsenseai/realsense-ros #2854: Post-processing filters in ros launch file?

ydzat, 2023-08-22:
I want to run the camera driver file (.launch) along with the filter. But I don't know what should be the name of this filter that should be filled in, please help me!

This is the corresponding line in the launch file:

```
<arg name="filters" value=""/>
```

I got two value by guessing: `decimation`, `temporal`. But that's not enough. Refer to [this](https://dev.intelrealsense.com/docs/post-processing-filters), it gives a recommended order: 

> Depth Frame >> Decimation Filter >> Depth2Disparity Transform** -> Spatial Filter >> Temporal Filter >> Disparity2Depth Transform** >> Hole Filling Filter >> Filtered Depth.

What should be the name of these filters that corresponds to the value parameter?

MartyG-RealSense, 2023-08-24:
Hi @ydzat  The names of the filters in the ROS1 wrapper are listed below.

**disparity** - convert depth to disparity before applying other filters and back.
**spatial** - filter the depth image spatially.
**temporal** - filter the depth image temporally.
**hole_filling** - apply hole-filling filter.
**decimation** - reduces depth scene complexity.

The launch format is to put the word **filters:=** and then the filter name that you want to launch.  If there is more than one filter name, separate them with commas.  For example:

`roslaunch realsense2_camera rs_camera.launch filters:=decimation, temporal`

ydzat, 2023-08-24:
> Hi @ydzat The names of the filters in the ROS1 wrapper are listed below.
>
> **disparity** - convert depth to disparity before applying other filters and back. **spatial** - filter the depth image [...]

Thank you so much!

closed 2023-08-24 by ydzat