realsenseai/realsense-ros #2855: How can I execute the box_dimensioner_multicam_demo.py for ROS communication?

kevinDrawn, 2023-08-23:
My Ros environment is Ubuntu 18.04 Melodic. And Depth Camera Model is D405.

I want to run the code "box_dimensioner_multicam_demo.py" for Box dimensions calculation.

And I want to communicate the data about 3d object dimension. (Width, Length, Height) 

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam

#11157

MartyG-RealSense, 2023-08-24:
Hi @kevinDrawn  Pyrealsense2 code can be run from ROS as a **node script**.  An example of such a script provided by the ROS wrapper is **show_center_depth.py**, which displays the depth value at the center of the camera's view.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/scripts/show_center_depth.py

The command below demonstrates how to run the node script.

`python src/realsense/realsense2_camera/scripts/show_center_depth.py --all`

kevinDrawn, 2023-08-27:
Hello! Thanks your reply.
I'm struggling to try the python code "the box_dimensioner_multicam_demo.py"
I did the command you suggested.
```
cd ~/Documents/librealsense-2.51.1/wrappers/python/examples/box_dimensioner_multicam
python3 box_dimensioner_multicam_demo.py --all
```

And then, the below command was came out.

```
1 devices have been found
Place the chessboard on the plane where the object needs to be detected..
Place the chessboard on the plane where the object needs to be detected..
Place the chessboard on the plane where the object needs to be detected..
Place the chessboard on the plane where the object needs to be detected..
Place the chessboard on the plane where the object needs to be detected..
Place the chessboard on the plane where the object needs to be detected..
Place the chessboard on the plane where the object needs to be detected..
Place the chessboard on the plane where the object needs to be detected..
Place the chessboard on the plane where the object needs to be detected..
Place the chessboard on the plane where the object needs to be detected..

```

The Camera frame did not appear. So I don't know where the camera is looking.
Even though I positioned the checker board in front of the camera lens, the message "Place the chessboard on the plane where the object needs to be detected.." was still appeared.   

How can I watch the Camera frame?

MartyG-RealSense, 2023-08-27:
The box_dimensioner_multicam has a 'sweet spot' distance to place the camera away from the chessboard.  The program may not work as expected outside of that range.  Placing the camera between 0.5 and 1 meters from the chessboard is likely to work best.

kevinDrawn, 2023-08-27:
Is it acceptable to set the distance to objects from 50cm to 100cm, even though the ideal depth measurement range for the d405 camera is 7 to 50cm? I'm conducting an experiment, and the camera only seems to recognize objects when fingers are brought close, as shown in the image below.

[image]

MartyG-RealSense, 2023-08-27:
box_dimensioner_multicam was written years before the D405 model was released.  Whilst most 400 Series cameras use the same **depth unit scale** of '0.001', the D405 uses a different depth scale of **0.01** because it is designed for short-range capture.  It is possible to change the default depth scale of the camera with program scripting.  Python code for doing so with the **rs.option.depth_units** instruction is at https://github.com/IntelRealSense/librealsense/issues/10973

The camera should be in a fixed position, yes.

It is possible to use box_dimensioner_multicam without the chessboard when only using one camera with it.  Details about doing so can be found at https://github.com/IntelRealSense/librealsense/issues/9951#issuecomment-966039933

kevinDrawn, 2023-08-29:
Where do i modify the code for setting the depth scale of the camera?? 

box_dimensioner_multicam_demo.py ? Or realsense_device_manager.py ?

Thank you a lot.

MartyG-RealSense, 2023-08-29:
**realsense_device_manager.py** would be the appropriate file to make the edit in, yes, as it handles the camera device.

The scale would be set to **0.001** rather than 0.01, as the D405's scale is already 0.01 by default and so you would not be changing anything by using 0.01.

kevinDrawn, 2023-09-02:
Thank you Bro!! I solve my problem!

MartyG-RealSense, 2023-09-02:
That's great to hear that you have a solution.  Thanks very much for the update!

closed 2023-09-07 by MartyG-RealSense