realsenseai/realsense-ros #2856: Questions about using the camera

develiberta, 2023-08-24:
I haven't bought a camera yet, but I'm trying to test a virtual camera on a digital twin using urdf files in the realsense2_description folder in this repository. **Is this possible?**
I am having a hard time testing in many ways as follows.

1. The command guided by README.md does not show the input of the camera. Specifically, the following results are obtained.
```
$ ros2 run realsense2_camera realsense2_camera_node
[INFO] [1692839084.755977797] [camera]: RealSense ROS v4.54.1
[INFO] [1692839084.756074406] [camera]: Built with LibRealSense v2.54.1
[INFO] [1692839084.756090156] [camera]: Running with LibRealSense v2.54.1
[WARN] [1692839084.759219004] [camera]: No RealSense devices were found!
[WARN] [1692839090.763100072] [camera]: No RealSense devices were found!
[WARN] [1692839096.765315352] [camera]: No RealSense devices were found!
...
```

2. Tried to see if I could see the camera on RViz using the following command.
`$ ros2 launch realsense2_description view_model.launch.py model:=test_l515_camera.urdf.xacro`
The shape of the camera appears normal, but I don't know how to see the information coming into the camera sensor. Does the model include camera sensor information? How can I check on RViz?

3. The xacro file in realsense2_description was converted to an urdf file. Looking at the urdf file, it seems that this file does not contain camera sensor information. This is because it consists of only links and points that represent the shape of the camera. Did I get it right? If the urdf file also contains camera sensor information, where is it contained?

4. If there is no camera sensor-related information in the urdf, how can the urdf and the camera sensor be combined later? I need to define which part of the camera model the sensor will be attached to, and I need to define the input/output properties of the sensor, so where and how should I define this?

Thank you for reading.

MartyG-RealSense, 2023-08-24:
Hi @develiberta  I must emphasize upfront that my knowledge of URDF is limited.

However, RealSense URDFs are typically used with **Gazebo** simulation, as demonstrated in the YouTube video at the link below which makes use of the RealSense ROS2 wrapper, RViz, the **Intel RealSense Gazebo ROS Plugin** and the **test_d435_camera.urdf.xacro** file.

https://www.youtube.com/watch?v=hpUCG6K5muI

I hope that this video will be helpful in answering your questions.

develiberta, 2023-08-28:
@MartyG-RealSense 

Thank you.
I checked the data you gave me well.
But it seems to be explaining it in the ROS1 version. Is there any reference material for ROS2 version?

MartyG-RealSense, 2023-08-28:
https://github.com/IntelRealSense/realsense-ros/issues/2283 discusses using Gazebo with ROS2.

MartyG-RealSense, 2023-09-04:
Hi @develiberta  Do you require further assistance with this case, please?  Thanks!

develiberta, 2023-09-05:
@MartyG-RealSense It was helpful enough. Thank you.

MartyG-RealSense, 2023-09-05:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2023-09-05 by MartyG-RealSense