realsenseai/realsense-ros #2858: Unable to Find <stream_name>_frame_id Parameters in ROS 2

kishoreparanthaman, 2023-08-28:
I'm working on a project that involves copying RGB frames from RealSense cameras and passing them to different virtual frames using v4l2loopback and FFmpeg. In this process, I need to manually update the frame IDs for the different streams to ensure proper synchronization and alignment. According to the documentation, the parameters I need to modify are <stream_name>_frame_id, <stream_name>_optical_frame_id, and aligned_depth_to_<stream_name>_frame_id.

However, upon searching through the ROS 2 parameter list, I cannot locate any parameters that end with _frame_id or _optical_frame_id. This is hindering my progress as I am unable to set the desired frame IDs for the streams.

I would greatly appreciate it if someone could provide guidance on how to change these frame ID parameters manually, or if there's an alternative approach I should take to achieve the desired synchronization and alignment of frames from multiple cameras.

Expected Behavior:
I expect to find the <stream_name>_frame_id and related parameters in the ROS 2 parameter list, which would allow me to manually update the frame IDs for different streams.

MartyG-RealSense, 2023-08-28:
Hi @kishoreparanthaman  Defining the **base_frame_id** launch parameter should set the frame_id that all static transformations refers to.  What happens if the launch instruction below is used, please?

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p base_frame_id:=camera_link`

On RealSense 400 Series cameras though, the **Global Time** parameter that helps to sync the timestamps of multiple cameras is already enabled by default - see https://github.com/IntelRealSense/librealsense/pull/3909

kishoreparanthaman, 2023-08-28:
I'm currently using a single RealSense D435i camera with ROS 2 for my project. My goal is to launch the ROS 2 camera node while simultaneously streaming the RGB feed from the RealSense camera to a separate Web GUI. I initially used RQT for this purpose, but now I'm aiming to move the GUI to a separate process outside of the ROS interface since there is latency. 

### **Issue:**

Upon connecting the RealSense camera, I've noticed that the camera streams are assigned video IDs ranging from _video1_ to _video7_ (confirmed using _**ls /dev/video***_). I specifically want to reassign the default _video6_ (RGB stream) to create two new virtual streams. The intention behind this is to ensure that I can run the **_ros2 run realsense2_camera realsense2_camera_node_** command without encountering the "device is busy" error.

How can I change the default video stream ID, currently video6, to a new stream ID that I can utilize for running the realsense2_camera_node without device conflicts?

### **Additional Information:**

RealSense camera model: D435i
ROS 2 foxy version being used.

MartyG-RealSense, 2023-08-28:
Would the **-map** function of ffmpeg meet your needs?

https://trac.ffmpeg.org/wiki/Map

kishoreparanthaman, 2023-08-28:
@MartyG-RealSense , I want to modify the following parameter
[image]

but when I check in ROS 2 parameter list it isn't available

MartyG-RealSense, 2023-08-29:
Checking the ROS wrapper release notes, it appears that the parameters were removed from the ROS2 wrapper a long time ago back in wrapper 4.0 and frame_ids are instead based on **camera_name** 

[image]

MartyG-RealSense, 2023-08-29:
My colleagues on the RealSense ROS team advised that the ffmpeg **-map** suggestion that I made earlier was a suitable answer and there were not any other foreseeable options for achieving your goal through the ROS wrapper.

They said though that another possibility may be to create your own subscriber and then to publish two topics on your side.  You could use **intra-process** communication to reduce copies and optimize the performance of your nodes.

MartyG-RealSense, 2023-08-29:
In regard to the absent frame_id parameters, my colleagues also informed me that they are not used any more and were left in the documentation by mistake.  The documentation will be updated to correct this.

closed 2023-09-15 by MartyG-RealSense