realsenseai/realsense-ros #2864: Multiple cameras on same topic

chumpblocckami, 2023-08-29:
Hi, i'm new to ROS and i'm trying to set up two Intel(R) RealSense(TM) Depth Ca on a computer (Lenovo Legion 5 15ARH05H) on two different USB port.

I launch the cameras as:
```
launch/cameras:
	ros2 launch realsense2_camera rs_launch.py camera:=camera_1 serial_no:=_$(CAMERA_1_SERIAL_NUMBER) &\
 	ros2 launch realsense2_camera rs_launch.py camera:=camera_2 serial_no:=_$(CAMERA_2_SERIAL_NUMBER)
```

The output on RViz is: 
[image]

The cameras are up and running but they share the same topic, which triggers the flickering on the video.
How can I run the cameras in two different topic?

Thanks a lot

MartyG-RealSense, 2023-08-29:
Hi @chumpblocckami  Does the flickering occur if you use the ROS2 wrapper's multiple camera launch file for a two-camera setup?

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/launch/rs_multi_camera_launch.py

`ros2 launch realsense2_camera rs_multi_camera_launch.py`

chumpblocckami, 2023-08-29:
Thanks for the quick response, 

I tried to launch both 
`ros2 launch realsense2_camera rs_multi_camera_launch.py`
and 
`ros2 launch realsense2_camera rs_multi_camera_launch.py camera_name1:=camera_1 camera_name2:=camera_2 device_type2:=d4. device_type1:=d4.`
defining both camera_names and device_types (that are equal because the cameras are the same).

I had the same result:

[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [44]
[INFO] [realsense2_camera_node-2]: process started with pid [46]
[INFO] [static_transform_publisher-3]: process started with pid [48]
[static_transform_publisher-3] [WARN] [1693324128.381845552] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-3] [INFO] [1693324128.398060717] [static_transform_publisher_K26PpZct8iZ1azue]: Spinning until stopped - publishing transform
[static_transform_publisher-3] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-3] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-3] from 'camera1_link' to 'camera2_link'
[realsense2_camera_node-1] [0m[INFO] [1693324128.464760937] [camera1.camera1]: RealSense ROS v4.54.1[0m
[realsense2_camera_node-1] [0m[INFO] [1693324128.464883718] [camera1.camera1]: Built with LibRealSense v2.54.1[0m
[realsense2_camera_node-1] [0m[INFO] [1693324128.464903344] [camera1.camera1]: Running with LibRealSense v2.54.1[0m
[realsense2_camera_node-2] [0m[INFO] [1693324128.466044349] [camera2.camera2]: RealSense ROS v4.54.1[0m
[realsense2_camera_node-2] [0m[INFO] [1693324128.466156724] [camera2.camera2]: Built with LibRealSense v2.54.1[0m
[realsense2_camera_node-2] [0m[INFO] [1693324128.466172439] [camera2.camera2]: Running with LibRealSense v2.54.1[0m
[realsense2_camera_node-1] [0m[INFO] [1693324128.492523041] [camera1.camera1]: Device with serial number 318122301365 was found.
[realsense2_camera_node-1] [0m
[realsense2_camera_node-1] [0m[INFO] [1693324128.492593302] [camera1.camera1]: Device with physical ID /sys/devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb3/3-2/3-2:1.0/video4linux/video20 was found.[0m
[realsense2_camera_node-1] [0m[INFO] [1693324128.492609156] [camera1.camera1]: Device with name Intel RealSense D455 was found.[0m
[realsense2_camera_node-1] [0m[INFO] [1693324128.492834535] [camera1.camera1]: Device with port number 3-2 was found.[0m
[realsense2_camera_node-1] [0m[INFO] [1693324128.492862891] [camera1.camera1]: Device USB type: 2.1[0m
[realsense2_camera_node-1] [33m[WARN] [1693324128.492876231] [camera1.camera1]: Device 318122301365 is connected using a 2.1 port. Reduced performance is expected.[0m
[realsense2_camera_node-1] [0m[INFO] [1693324128.493400254] [camera1.camera1]: getParameters...[0m
[realsense2_camera_node-2] [0m[INFO] [1693324128.493745132] [camera2.camera2]: Device with serial number 318122301365 was found.
[realsense2_camera_node-2] [0m
[realsense2_camera_node-2] [0m[INFO] [1693324128.493802403] [camera2.camera2]: Device with physical ID /sys/devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb3/3-2/3-2:1.0/video4linux/video20 was found.[0m
[realsense2_camera_node-2] [0m[INFO] [1693324128.493820631] [camera2.camera2]: Device with name Intel RealSense D455 was found.[0m
[realsense2_camera_node-2] [0m[INFO] [1693324128.494152310] [camera2.camera2]: Device with port number 3-2 was found.[0m
[realsense2_camera_node-2] [0m[INFO] [1693324128.494196939] [camera2.camera2]: Device USB type: 2.1[0m
[realsense2_camera_node-2] [33m[WARN] [1693324128.494212793] [camera2.camera2]: Device 318122301365 is connected using a 2.1 port. Reduced performance is expected.[0m
[realsense2_camera_node-1] [0m[INFO] [1693324128.494567869] [camera1.camera1]: JSON file is not provided[0m
[realsense2_camera_node-1] [0m[INFO] [1693324128.494617526] [camera1.camera1]: Device Name: Intel RealSense D455[0m
[realsense2_camera_node-1] [0m[INFO] [1693324128.494630866] [camera1.camera1]: Device Serial No: 318122301365[0m
[realsense2_camera_node-1] [0m[INFO] [1693324128.494641342] [camera1.camera1]: Device physical port: /sys/devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb3/3-2/3-2:1.0/video4linux/video20[0m
[realsense2_camera_node-1] [0m[INFO] [1693324128.494656218] [camera1.camera1]: Device FW version: 5.13.0.55[0m
[realsense2_camera_node-1] [0m[INFO] [1693324128.494669349] [camera1.camera1]: Device Product ID: 0x0B5C[0m
[realsense2_camera_node-1] [0m[INFO] [1693324128.494683666] [camera1.camera1]: Sync Mode: Off[0m
[realsense2_camera_node-2] [0m[INFO] [1693324128.494881388] [camera2.camera2]: getParameters...[0m
[realsense2_camera_node-2] [0m[INFO] [1693324128.495384668] [camera2.camera2]: JSON file is not provided[0m
[realsense2_camera_node-2] [0m[INFO] [1693324128.495421964] [camera2.camera2]: Device Name: Intel RealSense D455[0m
[realsense2_camera_node-2] [0m[INFO] [1693324128.495436072] [camera2.camera2]: Device Serial No: 318122301365[0m
[realsense2_camera_node-2] [0m[INFO] [1693324128.495450529] [camera2.camera2]: Device physical port: /sys/devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb3/3-2/3-2:1.0/video4linux/video20[0m
[realsense2_camera_node-2] [0m[INFO] [1693324128.495465196] [camera2.camera2]: Device FW version: 5.13.0.55[0m
[realsense2_camera_node-2] [0m[INFO] [1693324128.495479373] [camera2.camera2]: Device Product ID: 0x0B5C[0m
[... cut ...]
[realsense2_camera_node-1]  29/08 11:48:49,534 ERROR [140594831660608] (backend-v4l2.cpp:2771) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 24 error: No such device
[realsense2_camera_node-1]  29/08 11:48:49,534 ERROR [140594831660608] (backend-v4l2.cpp:2771) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 24 error: No such device
...

My gut says that this error can be related to the fact that both cameras are the same, so the script cannot detect the correct 
serial number for each cameras, resulting in overlapping the same camera.

Any thoughts on this?

MartyG-RealSense, 2023-08-30:
It does appear that only the camera with serial number 318122301365 is being detected.

Do the cameras both work normally when launched on their own (with one of them unplugged) instead of together?

chumpblocckami, 2023-08-30:
Yes, cameras were detected correctly if they one is plugged and the other one is not.

MartyG-RealSense, 2023-08-30:
Have you tried launching the cameras in separate ROS terminals?

**Terminal 1**
ros2 launch realsense2_camera rs_launch.py camera:=camera_1 serial_no:=318122301365

**Terminal 2**
ros2 launch realsense2_camera rs_launch.py camera:=camera_2 serial_no:= [whatever the camera 2 serial number is]

SamerKhshiboun, 2023-08-31:
Hi @chumpblocckami, you can take a look on latest changes in ros2-development branch, which include support of camera_namespace and camera_name parameters. This will help creating two different topics for the two cameras.

See announcement by @MartyG-RealSense regarding these changes:
https://github.com/IntelRealSense/realsense-ros/issues/2866

chumpblocckami, 2023-08-31:
Hi @SamerKhshiboun , thank you for pinpoint me into ros2-development branch, it was exactly what i needed :) 
I was able to generate two topic related to different cameras using :

```
	ros2 launch realsense2_camera rs_launch.py camera_namespace:=topic_1 camera_name:=cam_1 serial_no:=_$(CAMERA_1_SERIAL_NUMBER) &
 	ros2 launch realsense2_camera rs_launch.py camera_namespace:=topic_2 camera_name:=cam_2  serial_no:=_$(CAMERA_2_SERIAL_NUMBER)
```
The output in rviz is the following (video and depth): 

[image]

I post the commands and the outputs for clarification and for future users that might be interested.

closed 2023-08-31 by chumpblocckami

MartyG-RealSense, 2023-08-31:
@chumpblocckami  It's excellent news that the wrapper changes enabled you to succeed.  Thanks very much for the update!

chieweng, 2025-02-04:
Hi, could you post the code to display both camera streams on rviz please? Thanks