realsenseai/realsense-ros #2867: Is a D455 depth image aligned with its corresponded undistorted color image or the original color image?

yanmingzou, 2023-08-31:
On D455, the published RGB color images are with distortions.  If the parameter align_depth.enable is set true, what are the images the depth images are aligned with? The published color image or the un-distorted color image? Thanks a lot.

yanmingzou, 2023-09-01:
Thanks Marty.
So if I undistort the color image, I also need to undistort the depth image to keep the pixel correspondence between color and depth image. Am I right?

MartyG-RealSense, 2023-09-01:
If align_depth is used then librealsense's **align processing block** will automatically adjust for differences between the depth and color streams when the two streams are mapped together into a single aligned image.

MartyG-RealSense, 2023-09-13:
Case closed due to no further commemts received.

closed 2023-09-13 by MartyG-RealSense