realsenseai/realsense-ros #2869: Error connecting to T265

biagio-trimarchi, 2023-09-07:
Greetings.

I am trying to use a tracking camera T265 with ROS2 Foxy inside a Docker Container.

I have tried to install the realsense packages both via `sudo apt install ros-foxy-realsense2-camera` and through git by cloning the ros2-legacy branch.

Whenever I try to launch either `rs_launch.py`(with the ros-foxy-realsense2-camera installation) or `rs_t265_launch.py` (with the ros2-legacy installation) I get the following error:

```
[realsense2_camera_node-1]  07/09 14:53:06,991 ERROR [140639401568000] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
[realsense2_camera_node-1]  07/09 14:53:06,991 ERROR [140639401568000] (tm-boot.h:42) Failed to open T265 zero interface
```

The problem seems to be related to the T265 camera only, since I have tested also a D435i and I was able to see its topic on ROS2 and display its image on RViz2.

biagio-trimarchi, 2023-09-08:
I fixed it.

I needed to set up the udev permission in the local machine.

In the end I download on my local machine the github repository of RealSense (https://github.com/IntelRealSense/librealsense/tree/master), switched to version 2.50.0 (I do not know if it is a necessary step) and run the `setup_udev_file.sh` in the `scripts` folder.

On my docker container, I have installed the ros-foxy-realsense2-camera package and it all works fine now.

closed 2023-09-08 by biagio-trimarchi

AllenIsaacJose, 2023-09-13:
> I fixed it.
>
> I needed to set up the udev permission in the local machine.
>
> In the end I download on my local machine the github repository of RealSense [...]

How did you installed the SDK??

AllenIsaacJose, 2023-09-15:
SOLVED

```
sudo sh -c "echo 141a0000.pcie > /sys/bus/platform/drivers/tegra194-pcie/unbind"
sudo sh -c "echo 141a0000.pcie > /sys/bus/platform/drivers/tegra194-pcie/bind"
```
Find the PCIe card to which the USB is connected. In my case `141a0000.pcie`. The above command will turn the power supply to the PCIe card off and then on thereby making my T265 tracker camera detectable.