realsenseai/realsense-ros #2870: D435i-crash occasionlly (types.h:310) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device

ZhixiongSun, 2023-09-09:
Hi, We have many camera crashed, and log below:

[component_container-5] [INFO] [1694255400.079498993] [RealSenseCameraNode]: RealSense ROS v3.1.2
[component_container-5] [INFO] [1694255400.079715353] [RealSenseCameraNode]: Built with LibRealSense v2.49.0
[component_container-5] [INFO] [1694255400.079736585] [RealSenseCameraNode]: Running with LibRealSense v2.49.0
[component_container-5] [INFO] [1694255400.114625530] [RealSenseCameraNode]: GetRsFwVersion srv name:/sensor/camera/realsense_0/get_rs_fw_version
[component_container-5] [INFO] [1694255400.140881908] [RealSenseCameraNode]: GET_DEVICE_BY_ORIGIN
[component_container-5] [INFO] [1694255400.154634587] [RealSenseCameraNode]: Device with serial number 134222077064 was found.
[component_container-5] 
[component_container-5] [INFO] [1694255400.154711200] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:15.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[component_container-5] [INFO] [1694255400.154726664] [RealSenseCameraNode]: Device with name Intel RealSense D435I was found.
[component_container-5] [INFO] [1694255400.156565259] [RealSenseCameraNode]: Device with port number 2-1 was found.
[component_container-5] [INFO] [1694255400.156659517] [RealSenseCameraNode]: Device USB type: 3.2
[component_container-5] [INFO] [1694255400.156678409] [RealSenseCameraNode]: Resetting device...
[component_container-5] [INFO] [1694255406.187916488] [RealSenseCameraNode]: GET_DEVICE_BY_ORIGIN
[component_container-5] [INFO] [1694255406.223327054] [RealSenseCameraNode]: Device with serial number 134222077064 was found.
[component_container-5] 
[component_container-5] [INFO] [1694255406.223408232] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:15.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[component_container-5] [INFO] [1694255406.223423993] [RealSenseCameraNode]: Device with name Intel RealSense D435I was found.
[component_container-5] [INFO] [1694255406.223924599] [RealSenseCameraNode]: Device with port number 2-1 was found.
[component_container-5] [INFO] [1694255406.223970518] [RealSenseCameraNode]: Device USB type: 3.2
[component_container-5] [INFO] [1694255406.251414991] [RealSenseCameraNode]: getParameters...
[component_container-5] [INFO] [1694255406.264407330] [RealSenseCameraNode]: emitter_control: 1
[component_container-5] [INFO] [1694255406.264475847] [RealSenseCameraNode]: emitter_enabled: 0
[component_container-5] [INFO] [1694255406.264487061] [RealSenseCameraNode]: emitter_on_off: 0
[component_container-5] [INFO] [1694255406.264496476] [RealSenseCameraNode]: global_time_enabled: 1
[component_container-5] [INFO] [1694255406.264505930] [RealSenseCameraNode]: visual_preset: 3
[component_container-5] [INFO] [1694255406.432116614] [RealSenseCameraNode]: setupDevice...
[component_container-5] [INFO] [1694255406.432625954] [RealSenseCameraNode]: Not calibrated or no flag! If json provided will load json file
[component_container-5] [INFO] [1694255406.433364559] [RealSenseCameraNode]: JSON file is not provided or not calibrated
[component_container-5] [INFO] [1694255406.433405172] [RealSenseCameraNode]: Device Name: Intel RealSense D435I
[component_container-5] [INFO] [1694255406.433420695] [RealSenseCameraNode]: Device Serial No: 134222077064
[component_container-5] [INFO] [1694255406.433436073] [RealSenseCameraNode]: Device physical port: /sys/devices/pci0000:00/0000:00:15.0/usb2/2-1/2-1:1.0/video4linux/video0
[component_container-5] [INFO] [1694255406.433451901] [RealSenseCameraNode]: Device FW version: 05.12.06.00
[component_container-5] [INFO] [1694255406.433544709] [RealSenseCameraNode]: Device Product ID: 0x0B3A
[component_container-5] [INFO] [1694255406.433562438] [RealSenseCameraNode]: Enable PointCloud: Off
[component_container-5] [INFO] [1694255406.433576752] [RealSenseCameraNode]: Align Depth: Off
[component_container-5] [INFO] [1694255406.433590224] [RealSenseCameraNode]: Sync Mode: On
[component_container-5] [INFO] [1694255406.433645997] [RealSenseCameraNode]: Device Sensors: 
[component_container-5] [INFO] [1694255406.444464551] [RealSenseCameraNode]: Stereo Module was found.
[component_container-5] [INFO] [1694255406.470915577] [RealSenseCameraNode]: RGB Camera was found.
[component_container-5] [INFO] [1694255406.471286089] [RealSenseCameraNode]: Motion Module was found.
[component_container-5] [INFO] [1694255406.471410622] [RealSenseCameraNode]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[component_container-5] [INFO] [1694255406.471432502] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[component_container-5] [INFO] [1694255406.471450285] [RealSenseCameraNode]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[component_container-5] [INFO] [1694255406.471475099] [RealSenseCameraNode]: num_filters: 0
[component_container-5] [INFO] [1694255406.471490072] [RealSenseCameraNode]: Setting Dynamic reconfig parameters.
[component_container-5] [INFO] [1694255406.765807809] [RealSenseCameraNode]: Done Setting Dynamic reconfig parameters.
[component_container-5] [INFO] [1694255406.772462556] [RealSenseCameraNode]: depth stream is enabled - width: 1280, height: 720, fps: 15, Format: Z16
[component_container-5] [INFO] [1694255406.773402065] [RealSenseCameraNode]: infra1 stream is enabled - width: 1280, height: 720, fps: 15, Format: Y8
[component_container-5] [INFO] [1694255406.773473732] [RealSenseCameraNode]: setupPublishers...
[component_container-5] [INFO] [1694255406.818673946] [RealSenseCameraNode]: setupStreams...
[component_container-5] [INFO] [1694255407.080390656] [RealSenseCameraNode]: Depth stereo module mode: High Accuracy!
[component_container-5] [INFO] [1694255409.426057941] [RealSenseCameraNode]: SELECTED BASE:Depth, 0

[component_container-5]  09/09 11:26:23,973 ERROR [140308981470976] (types.h:310) acquire: Cannot open '/dev/video0 Last Error: No such file or directory
[component_container-5]  09/09 11:26:24,003 ERROR [140308981470976] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: acquire: Cannot open '/dev/video0 Last Error: No such file or directory
[component_container-5]  09/09 11:26:24,101 ERROR [140309075396352] (types.h:310) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
[component_container-5]  09/09 11:26:24,101 ERROR [140309075396352] (error-handling.cpp:78) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
[component_container-5]  09/09 11:26:25,004 ERROR [140308981470976] (types.h:310) acquire: Cannot open '/dev/video0 Last Error: No such file or directory

As you can see we used RealSense ROS v3.1.2  LibRealSense v2.49.0 and Device FW version: 05.12.06.00
most cameras ran 1 hour or more and then some of them crashes with same bugs.
Do you have any ideas about this issue or if version of LibRealSense  and FW  are mismatched.
Thanks a lot for your help in advance

MartyG-RealSense, 2023-09-09:
Hi @ZhixiongSun   It is important to match the ROS wrapper version as closely as possible with a specific librealsense version listed in that wrapper's release notes.  Wrapper 3.1.2 was designed for use with librealsense **2.40.0**.  There was not a wrapper created for librealsense 2.49.0 though.  The closest match is wrapper **3.2.2** which was designed for use with 2.48.0.

The 3.x series of ROS2 wrappers are now obsolete though and were replaced by the current 4.x series of ROS2 wrappers.  If possible, I would recommend that you use wrapper **4.54.1** and librealsense **2.54.1**.  But if you have to use librealsense 2.49.0 then the closest available wrapper match in the 4-series would be **4.0.4** which was designed for 2.50.0.

The firmware version 5.12.6.0 is also too old for use with librealsense 2.49.0 as it was designed for the very old 2.36.0.  The best match for 2.49.0 is **5.12.15.50**.

ZhixiongSun, 2023-09-11:
Hi, @MartyG-RealSense I have tried FW 5.12.15.50 with  librealsense 2.49.0, still keep ROS2 wrapper  3.1.2. And this bug still exist. I read some Issue 
https://github.com/IntelRealSense/realsense-ros/issues/2004#issuecomment-889843031
https://github.com/IntelRealSense/librealsense/issues/7893#issuecomment-736560284
you mentioned this mau because of conflict of kernel. But Do you have any idea how can I solve this issue?

MartyG-RealSense, 2023-09-11:
If you are able to build the librealsense SDK from source code using the **RSUSB backend** installation method then the SDK will bypass the kernel.  This can be a solution for kernel conflicts, as an RSUSB build of the SDK is not dependent on Linux versions or kernel versions and does not need to have a patch script applied to the kernel.  Instructions for an RSUSB build can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

ZhixiongSun, 2023-09-11:
Hi @MartyG-RealSense , Thank for your fast response. I have read #9931
I have two questions about this bug. 1) Can we say this bug of (UVCIOC_CTRL_QUERY) is because of kernel conflicts? and 2) Is that means I just build librealsense from source code  in my environment and then I can bypass the kernel issue according to https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692.

closed 2023-09-28 by MartyG-RealSense