realsenseai/realsense-ros #2873: No point cloud in generated but depth image seems working

xuyanwen2012, 2023-09-11:
Hello,

I did some research before opening this issue. Many others have similar issues; I followed their solutions, but it still did not work. 

- I am using `Ubuntu 20.04` on _Jetson Orin Nano_, I have a _D435_
- I installed ROS2 foxy using `apt install ros-foxy-desktop.`
- Realsense SDK version `2.54.1`
- Wrapper installed with `apt install ros-foxy-realsense2-*`, should be version `4.51.1`

Running this command
```
ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true
```

Topic list seems right
```
/camera/color/camera_info
/camera/color/image_raw
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/imu
/parameter_events
/rosout
/tf_static
```

Nothing shows with
```
ros2 topic echo /camera/depth/color/points
```
, and no point cloud shows up in rviz2 too

```
depth/image_rect_raw
```
seems working. I can visualize the depth image in rviz2

It just doesn't have any point cloud. 

[image]

Any possible reasons? This is a freshly installed machine.
I really appreciate any help you can provide.

MartyG-RealSense, 2023-09-12:
Hi @xuyanwen2012  It is important to match the ROS wrapper version as closely as possible with a specific librealsense version listed in a wrapper version's release notes. Wrapper 4.51.1 should be used with librealsense **2.51.1.

The 2.54.1 version of librealsense should be used with wrapper **4.54.1**.  This will not be practical in your particular situation though as support for Foxy has been removed from 4.54.1 and it now only supports ROS2 Humble and Iron.

xuyanwen2012, 2023-09-12:
> Hi @xuyanwen2012 It is important to match the ROS wrapper version as closely as possible with a specific librealsense version listed in a wrapper version's release notes. Wrapper 4.51.1 should be [...]

Thank you very much for the information! I will try now and will let you know if I have further questions!

xuyanwen2012, 2023-09-12:
I have just built 2.51.1 using RSUSB approach. 

I think the issue still exists, the depth image is shown, but the point cloud is now shown. Even when echo /camera/depth/color/point

MartyG-RealSense, 2023-09-13:
Building the wrapper again is likely not going to make a difference.

Is the topic published if you launch with **ros2 run** instead of ros2 launch?

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p pointcloud.enable:=true`

MartyG-RealSense, 2023-09-14:
It's no trouble at all.  It's great to hear that ros2 run worked for you.  Thanks very much for the update!

As you succeeded with the pointcloud, I will close this case.  Thanks again!

closed 2023-09-14 by MartyG-RealSense