realsenseai/realsense-ros #2876: Can't set camera resolution

chumpblocckami, 2023-09-14:
Hi, i want to use my rgb camera in 858x480 resolution, and i start it using: 

`realsense2_camera rs_launch.py camera_namespace:=$(TOPIC_NAME_1) camera_name:=$(CAMERA_NAME_1) serial_no:=_$(CAMERA_1_SERIAL_NUMBER) align_depth:=true color_width:=848 color_height:=480 color_fps:=30.0`

but it fails to change the resolution, resulting in this log: 
```
ros2 launch realsense2_camera rs_launch.py camera_namespace:=camera_1 camera_name:=dx serial_no:=_** align_depth:=true color_width:=848 color_height:=480 color_fps:=30.0
[INFO] [launch]: All log files can be found below /root/.ros/...
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [44]
[realsense2_camera_node-1] [INFO] [1694701176.161507562] [camera_1.dx]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1694701176.161606389] [camera_1.dx]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1694701176.161623430] [camera_1.dx]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1694701176.192258385] [camera_1.dx]: Device with serial number 313522303246 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1694701176.192317472] [camera_1.dx]: Device with physical ID /sys/devices/pci0000:00/0000:00:08.1/0000:05:00.4/usb6/6-2/6-2:1.0/video4linux/video2 was found.
[realsense2_camera_node-1] [INFO] [1694701176.192330462] [camera_1.dx]: Device with name Intel RealSense D455 was found.
[realsense2_camera_node-1] [INFO] [1694701176.192509049] [camera_1.dx]: Device with port number 6-2 was found.
[realsense2_camera_node-1] [INFO] [1694701176.192523436] [camera_1.dx]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1694701176.193083920] [camera_1.dx]: getParameters...
[realsense2_camera_node-1] [INFO] [1694701176.193405753] [camera_1.dx]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1694701176.193449265] [camera_1.dx]: Device Name: Intel RealSense D455
[realsense2_camera_node-1] [INFO] [1694701176.193462046] [camera_1.dx]: Device Serial No: 313522303246
[realsense2_camera_node-1] [INFO] [1694701176.193474478] [camera_1.dx]: Device physical port: /sys/devices/pci0000:00/0000:00:08.1/0000:05:00.4/usb6/6-2/6-2:1.0/video4linux/video2
[realsense2_camera_node-1] [INFO] [1694701176.193489284] [camera_1.dx]: Device FW version: 5.13.0.55
[realsense2_camera_node-1] [INFO] [1694701176.193499691] [camera_1.dx]: Device Product ID: 0x0B5C
[realsense2_camera_node-1] [INFO] [1694701176.193510307] [camera_1.dx]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1694701176.234623687] [camera_1.dx]: Set Exposure.1 to 7500
[realsense2_camera_node-1] [INFO] [1694701176.241090027] [camera_1.dx]: Set ROS param depth_module.profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1694701176.247511319] [camera_1.dx]: Set ROS param rgb_camera.profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1694701176.247906067] [camera_1.dx]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1694701176.247943363] [camera_1.dx]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1694701176.255447978] [camera_1.dx]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1694701176.257965300] [camera_1.dx]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1694701176.269434018] [camera_1.dx]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1694701176.269570909] [camera_1.dx]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1694701176.285984525] [camera_1.dx]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1694701176.304547953] [camera_1.dx]: Open profile: stream_type: Color(0), Format: RGB8, **Width: 1280, Height: 720, FPS: 30**
[realsense2_camera_node-1] [INFO] [1694701176.306495084] [camera_1.dx]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1694701176.533141829] [camera_1.dx]: 

```

I also tired to change fps and resolutions but it won't work. 
Any tips?

MartyG-RealSense, 2023-09-14:
Hi @chumpblocckami  If you are using one of the 4.x ROS2 wrappers such as **4.54.1** then it uses a different method for defining a custom stream configuration to the method for the ROS1 wrapper that you used.  Please remove the instructions below from your launch command:

`align_depth:=true color_width:=848 color_height:=480 color_fps:=30.0`

And use this instead:

`align_depth.enable:=true depth_module.profile:=848x480x30 rgb_camera.profile:=848x480x30`

chumpblocckami, 2023-09-14:
@MartyG-RealSense  - thanks for your quick response.
The solution worked as a charm! :)

closed 2023-09-14 by chumpblocckami

MartyG-RealSense, 2023-09-14:
You are very welcome.  I'm pleased that I could help.  :)