realsenseai/realsense-ros #2877: ROS Noetic D455 RGB image turns to white

doguscank, 2023-09-18:
Hi there,

We are using 3 D455 cameras on our robot. Left and right ones are running on 15 FPS rate and at (480, 270) RGB resolution, (424, 240) depth resolution. Middle camera is also running on 15 FPS rate but (1280, 720) RGB resolution, and (424, 240) depth resolution.

We are facing a randomly occurring problem when we start camera launch file. Sometimes the initial image comes very bright and it shifts to totally white (all pixels have 254 R, G, B values) in ~1 second (I can see it frame-by-frame). Restarting the ROS launch usually solves the problem but this is not an exact solution. I observed that only one camera turns white (never saw 2 cameras became white but it may be possible) and it is left or right one most of the times.

Here are some information about our setup on the robot:
1- Original USB cables are used when connecting to the main computer
2- A sunglasses filter (ND & polarized) is placed in front of the RGB sensor
3- All cameras are connected to USB 3.1 ports of the main computer
4- Auto-exposure and auto-white-balance is enabled on both cameras

I checked old issues and found some solutions but none of them have worked for us. I tried;

1- Setting an auto-exposure ROI didn't change anything
2- Setting different auto-exposure setpoints (100, 200, 400) didn't solve the problem

Any help is appreciated. I would love to share if any additional information is required to understand the problem.

Thanks.

MartyG-RealSense, 2023-09-18:
Hi @doguscank  Does the problem still occur if you reset the camera during launch by adding the command **initial_reset:=true** to your ROS launch instruction?  For example:

ROS1
`roslaunch realsense2_camera rs_camera.launch initial_reset:=true`

ROS2
`ros2 launch realsense2_camera rs_launch.py initial_reset:=true`

If using initial_reset does not prevent the problem then you could try running the RGB stream at a higher FPS than the depth stream, such as 15 FPS depth and 60 FPS RGB.

doguscank, 2023-09-18:
Hi @MartyG-RealSense Yes, I tried to set `itinial_reset` to True but it didn't solve the problem. We are using ROS1 and can't set the camera FPS rates to a higher value due to CPU constraints. However, I can only run 1 camera and try it to be sure. I will update here as soon as possible.

doguscank, 2023-09-20:
Hi @MartyG-RealSense, we set all cameras to 30 FPS and it seems to solve the problem for now. I will update as we continue to test it. If you have any other suggestions, we would love to try them on the robot. Thanks.

doguscank, 2023-09-29:
Setting cameras to 30 FPS resolved the problem. We appreciate your help, thank you!

closed 2023-09-29 by doguscank