realsenseai/realsense-ros #2879: The frequency of topic "/camera/aligned_depth_to_color/image_raw"  can only reach 7Hz

kao-yang, 2023-09-20:
I try to use d435i camera to get color image and aligened depth image。The usb is 3.1，The running environment is the ubuntu20.04 system with arm architecture(rk3588)，and also ros-noetic。The color image setting has a resolution of 1920*1080 and a frequency of 15Hz。To get aligened_depth image， I set that :<arg name="align_depth"               value="true"/>，<arg name="enable_depth"        value="true"/>。But when useing,the frequency of topic "/camera/aligned_depth_to_color/image_raw"  can only reach 7Hz while the frequency of topic "/camera/color/image_raw"  is 15Hz.
The follow is my launch:
<launch>
  <arg name="device_type_camera1"    		default=""/>
  <arg name="serial_no_camera1"    		default=""/>
  <arg name="camera1"              		default="camera"/>
  <arg name="tf_prefix_camera1"         default="$(arg camera1)"/>

  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

  <group ns="$(arg camera1)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="device_type"           value="$(arg device_type_camera1)"/>
      <arg name="serial_no"             value="$(arg serial_no_camera1)"/>
      <arg name="tf_prefix"         		value="$(arg tf_prefix_camera1)"/>
      <arg name="initial_reset"         value="false"/>
      <arg name="reconnect_timeout"     value="6.0"/>
      <arg name="usb_port_id"              value="8-1.1"/>

      <arg name="color_width"              value="1920"/>
      <arg name="color_height"             value="1080"/>
      <arg name="enable_color"             value="true"/>
      <arg name="color_fps"                value="15"/>

      <arg name="depth_width"         value="640"/>
      <arg name="depth_height"        value="480"/>
      <arg name="enable_depth"        value="true"/>
      
      <arg name="enable_sync"               value="true"/>
      <arg name="align_depth"               value="true"/>
      <arg name="publish_tf"                value="false"/>
    </include>
  </group>

</launch>

And the follow is what the Terminal displayed:
... logging to /home/iflytek/.ros/log/6dceb724-5778-11ee-bd43-684e05957a82/roslaunch-superbrain-2395.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://superbrain:34487/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 0
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 1080
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 1920
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
[... 120 lines cut ...]
[ INFO] [1695188531.303484326]: setupPublishers...
[ INFO] [1695188531.317731369]: Expected frequency for depth = 30.00000
[ INFO] [1695188531.337505788]: Expected frequency for color = 15.00000
[ INFO] [1695188531.343125038]: Expected frequency for aligned_depth_to_color = 15.00000
[ INFO] [1695188531.355358123]: setupStreams...
 20/09 13:42:11,622 WARNING [547099038064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1695188531.674236488]: SELECTED BASE:Depth, 0
[ INFO] [1695188531.701066408]: RealSense Node Is Up!
[ WARN] [1695188531.743905662]: 
 20/09 13:42:11,745 WARNING [547099038064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/09 13:42:11,796 WARNING [547099038064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/09 13:42:12,099 WARNING [547099038064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/09 13:42:12,150 WARNING [547099038064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/09 13:42:12,208 WARNING [547099038064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/09 13:42:12,263 WARNING [547099038064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/09 13:44:26,453 WARNING [547099038064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done.
When I use d435i camera in my PC(x86 architecture)，all topics' frequency is right。
How to slove this problem？Thanks for your help.

MartyG-RealSense, 2023-09-20:
Hi @kao-yang  How does the aligned FPS perform if 1280x720 color is used instead of 1920x1080, please?

TemugeB, 2023-09-20:
I have a similar problem. You can see that the max time between two published frames can reach 5 s. The camera is launched with: 
```
ros2 launch realsense2_camera rs_launch.py  "align_depth.enable:=true"
```

[image]

kao-yang, 2023-09-20:
> Hi @kao-yang How does the aligned FPS perform if 1280x720 color is used instead of 1920x1080, please?

When use 1920x1080, the FPS is 7.5. When use 1280x720, the FPS is 8.8. When use 640*480, the FPS is 10.3.I am sure that the usb is 3.1.

kao-yang, 2023-09-20:
> Hi @kao-yang How does the aligned FPS perform if 1280x720 color is used instead of 1920x1080, please?

When use 1920*1080,aligened_depth image topic FPS is 7.5.When use 1280*720,aligened_depth image topic FPS is 8.5.When use 640*480,aligened_depth image topic FPS is 10.6. I am sure that the usb is 3.1

closed 2023-09-20 by kao-yang

MartyG-RealSense, 2023-09-20:
As you are using the 2.3.2 ROS1 wrapper, please try an **rs_rgbd** launch instead of an rs_camera launch.

1.  Install support for an RGBD launch using the command below.

`sudo apt-get install ros-noetic-rgbd-launch`

2.  Use the roslaunch instruction below:

`roslaunch realsense2_camera rs_rgbd.launch`

The rs_rgbd launch file has align_depth enabled by default, so you do not need to enable it yourself.

Does using this launch method improve the FPS?

kao-yang, 2023-09-21:
> As you are using the 2.3.2 ROS1 wrapper, please try an **rs_rgbd** launch instead of an rs_camera launch.
>
> 1. Install support for an RGBD launch using the command below.
>
> `sudo apt-get install [...]

Even if useing rs_rgbd.launch, the FPS is still only 6 Hz,which less than 15Hz.
I install ROS1 wrapper by this,how can I make sure the virsion?
```
sudo apt-get install ros-noetic-realsense2-camera
```

TemugeB, 2023-09-21:
@kao-yang 
Run the install command again and it will say it is already the newest version and will show you the version number.

I switched to:
```ros-noetic-realsense2-camera``` on ```Ubuntu 20.04``` and now it is maintaining 30fps fine. Which version of linux are you running? I think the realsense SDK does not support kernel 6 and above. Could this be a reason?

kao-yang, 2023-09-21:
> @kao-yang Run the install command again and it will say it is already the newest version and will show you the version number.
> 
> I switched to: `ros-noetic-realsense2-camera` on `Ubuntu 20.04` and now it is maintaining 30fps fine. Which version of linux are you running? I think the realsense SDK does not support kernel 6 and above. Could this be a reason?
> 
> My ROS1 is running on kernel 5.15 and ROS2 is running on kernel 6.2 I think.

This picture is the realsense_camera ROS1 wrapper version,2.3.2
[image]
And This picture is the linux version.
[image]

kao-yang, 2023-09-21:
> Hi @kao-yang Regarding your question about which ROS1 wrapper version is installed, it will always be 2.3.2. This is because development of the ROS1 wrapper has ceased and no further versions have been released after 2.3.2 or will be released in the future. Only the ROS2 wrapper is updated now.

So,I already get the newest ROS1 Wrapper version 2.3.2. How can aligened_depth image FPS reach 15Hz?

TemugeB, 2023-09-22:
In ROS2, I tried launching with
```ros2 launch realsense2_camera rs_launch.py "align_depth.enable:=true" "rgb_camera.auto_exposure_priority:=true"```
But I'm still getting about 2~3FPS. 

@MartyG-RealSense 
Should I start another issue or I should keep the discussion in here?

TemugeB, 2023-09-22:
I tried disabling the rgb camera and checked the publish rate of only the depth frames. 
The camera launch is with:
```ros2 launch realsense2_camera rs_launch.py "enable_color:=false"```
The log looks like:
[image]

The publish rate is:

[image]

@kao-yang 
Could you check if the depth frames are being published at 7FPS or only when align depth is true? Our problems may not be related.

MartyG-RealSense, 2023-09-22:
@TemugeB  I think it is useful for you to stay here if you are having a similar problem to @kao-yang as you can bounce ideas off each other.

kao-yang, 2023-09-22:
> I tried disabling the rgb camera and checked the publish rate of only the depth frames. The camera launch is with: `ros2 launch realsense2_camera rs_launch.py "enable_color:=false"` The log looks [...]

The origin depth image FPS is always reach the setting Hz(15 Hz),but the aligened_depth image FPS is only 7Hz.

kao-yang, 2023-09-22:
> @kao-yang Next, please try opening the ROS1 **dynamic_reconfigure** interface using the instructions at the link below.
> 
> https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#set-camera-controls-using-dynamic-reconfigure-params
> 
> Select the **rgb_camera** category from the side-panel and untick the **auto_exposure_priority** option to disable it. Does doing so improve the aligned FPS?
> 
> [image]

When trying use the rqt dynamic_reconfigure , I found the auto_exposure_priority default optiono is untick. Tick or untick the auto_exposure_priority option,aligened_depth image FPS is still around 7Hz.That seems not help.

TemugeB, 2023-09-22:
You could try using the python or C/C++ drivers to measure the FPS. At least it will tell you if the problem is with the ROS bindings or with your system or Camera.

MartyG-RealSense, 2023-09-22:
@kao-yang  As you have different FPS values for depth and color, that could potentially cause issues with synchronization between streams.  Could you try adding an FPS definition for depth to your launch file so that both color and depth are 15 FPS please?  Does doing this make a difference to your aligned FPS?

`<arg name="depth_fps"  value="15"/>`

kao-yang, 2023-09-23:
> @kao-yang As you have different FPS values for depth and color, that could potentially cause issues with synchronization between streams. Could you try adding an FPS definition for depth to your [...]
The setting of origin depth image FPS is already 15Hz before opening this issue.
Today I run the d435i,the FPS of aligened_depth image reach 15Hz! I haven't change anything, that really confuse me.Can reason be arm ubuntu system or hardware?
To replicate this phenomenon, I use another d435i and pc with same setting, however, the aligened_depth image fps still 7Hz.It seems the problem is still there.

MartyG-RealSense, 2023-09-23:
@kao-yang  As FPS increases as resolution is decreased when performing alignment - as described at https://github.com/IntelRealSense/realsense-ros/issues/2879#issuecomment-1727497398 - it is possible that the FPS drop is caused by additional demand placed on the CPU when aligning (as alignment is processed on the CPU and not the camera hardware).

You could confirm this by checking the percentage of CPU usage before and during alignment using an Ubuntu system monitoring tool such as **htop**.  If the percentage usage of the CPU significantly increases when alignment is enabled then the FPS drop is likely due to the CPU struggling to handle the burden of alignment.

kao-yang, 2023-09-24:
> @kao-yang As FPS increases as resolution is decreased when performing alignment - as described at [#2879 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/2879#issuecomment-1727497398) - it is possible that the FPS drop is caused by additional demand placed on the CPU when aligning (as alignment is processed on the CPU and not the camera hardware).
> 
> You could confirm this by checking the percentage of CPU usage before and during alignment using an Ubuntu system monitoring tool such as **htop**. If the percentage usage of the CPU significantly increases when alignment is enabled then the FPS drop is likely due to the CPU struggling to handle the burden of alignment.

If enable aligen,the cpu of realsense is 120%. If disable, the cpu is 70%. 
The bellow is my CPU info.It seems that the cpus still have a lot of free,as the max value is 800%.
Architecture:                    aarch64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
CPU(s):                          8
On-line CPU(s) list:             0-7
Thread(s) per core:              1
Core(s) per socket:              2
Socket(s):                       3
Vendor ID:                       ARM
Model:                           0
Model name:                      Cortex-A55
Stepping:                        r2p0
CPU max MHz:                     2304.0000
CPU min MHz:                     408.0000
BogoMIPS:                        48.00
L1d cache:                       256 KiB
L1i cache:                       256 KiB
L2 cache:                        1 MiB
L3 cache:                        3 MiB
Vulnerability Itlb multihit:     Not affected
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Mmio stale data:   Not affected
Vulnerability Retbleed:          Not affected
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl
Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
Vulnerability Spectre v2:        Vulnerable: Unprivileged eBPF enabled
Vulnerability Srbds:             Not affected
Vulnerability Tsx async abort:   Not affected
Flags:                           fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp

kao-yang, 2023-09-24:
> @kao-yang As you have different FPS values for depth and color, that could potentially cause issues with synchronization between streams. Could you try adding an FPS definition for depth to your [...]

Today, the FPS of aliegned_depth image become 7Hz again without any changes.That really confuse me.

MartyG-RealSense, 2023-09-24:
> If enable aligen,the cpu of realsense is 120%. If disable, the cpu is 70%.

That does seem to provide evidence that the CPU is being over-burdened by alignment.

The librealsense SDK, which is responsible for calculating alignment, only uses a single CPU core by default, so having multiple CPU cores available does not improve performance.  You can though set librealsense to take advantage of multiple cores when generating depth-color alignment if the SDK is built from source code with CMake and the build flag **-DBUILD_WITH_OPENMP=TRUE** is included in the CMake build instruction.

kao-yang, 2023-09-25:
> > If enable aligen,the cpu of realsense is 120%. If disable, the cpu is 70%.
> 
> That does seem to provide evidence that the CPU is being over-burdened by alignment.
> 
> The librealsense SDK, which is responsible for calculating alignment, only uses a single CPU core by default, so having multiple CPU cores available does not improve performance. You can though set librealsense to take advantage of multiple cores when generating depth-color alignment if the SDK is built from source code with CMake and `-DBUILD_WITH_OPENMP=TRUE` is included in the CMake build instruction.

Now I use apt-get to install ROS1 Wrapper. Can I git clone ROS1 Wrapper code and then set something in CMakeLists.txt to use multipe CPU?. As I want to use ros topic ,can you provide some help to achive multipe CPU?

MartyG-RealSense, 2023-09-25:
The only way to enable support for making use of more than one CPU core when performing alignment is to build librealsense from source code.

I have seen cases where RealSense users have set a build flag in CMakeLists.txt instead of in the CMake build instruction, though it is a very rarely used method and I do not have knowledge of how to make the CMakeLists.txt edit, unfortunately.

kao-yang, 2023-09-26:
> The only way to enable support for making use of more than one CPU core when performing alignment is to build librealsense from source code.
>
> I have seen cases where RealSense users have set a build [...]

Thanks,I will try to install SDK for multiple CPU.