realsenseai/realsense-ros #2880: How do I run multiple d435 in ros and record a bag

hushimin-create, 2023-09-25:


MartyG-RealSense, 2023-09-25:
Hi @hushimin-create  The RealSense ROS wrapper provides launch files that can be used to launch more than one camera.

**ROS1**
rs_multiple_devices.launch

**ROS2**
rs_multi_camera_launch.py

********

You can also distinguish between different cameras by assigning a name to them.

**ROS1**
https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#work-with-multiple-cameras

**ROS2**
https://github.com/IntelRealSense/realsense-ros/tree/ros2-development#--camera-name-and-camera-namespace

******

You should be able to record all published topics from the cameras to a bag file if you use the command **rosbag record --all**

hushimin-create, 2023-09-25:
I previously used "rs_multiple_devices.launch" to start two D435 cameras, but always reported the following error：
[ERROR] [1695643398.240246541]: An exception has been thrown: xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy
[ERROR] [1695643398.240284344]: Exception: xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy
Don't know why?

MartyG-RealSense, 2023-09-25:
**xioctl(VIDIOC_S_FMT)** errors typically indicate a conflict between librealsense and the Linux kernel.  Which kernel version are you using on your computer, please?

MartyG-RealSense, 2023-10-03:
Hi @hushimin-create  Do you require further assistance with this case, please?  Thanks!

hushimin-create, 2023-10-10:
Thank you very much for your concern. I am very sorry that I have been busy with other matters for the past two weeks and have not replied in a timely manner. My Linux kernel version is 5.15.0-86 generic. I really need further help on this issue. What should I do next? Thank you!

MartyG-RealSense, 2023-10-10:
It's no problem at all.  Thanks very much for the update!

What method did you use to install the librealsense SDK, please?  Did you build it from source code with CMake?

If you did build from source code with CMake and you did not include the build flag **-DFORCE_RSUSB_BACKEND=TRUE** in your CMake build instruction then I recommend including that build flag in order to eliminate the possibility of a conflict between librealsense and your 5.15 kernel.

hushimin-create, 2023-10-11:
Hello, I have recompiled librealsense according to the method you mentioned and reinstalled realsense ros. I am glad that the previous issue has been resolved. But what is annoying is that a new error has occurred.When I started two cameras using 'rs_multiple_devices. launch', the following error occurred：

 11/10 15:33:00,427 WARNING [139635179398912] (handle-libusb.h:110) failed to claim usb interface, interface 0, is busy - retrying...
 11/10 15:33:00,437 WARNING [139635179398912] (handle-libusb.h:110) failed to claim usb interface, interface 0, is busy - retrying...
 11/10 15:33:00,508 ERROR [139635179398912] (handle-libusb.h:125) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 11/10 15:33:00,508 ERROR [139635204577024] (sensor.cpp:661) acquire_power failed: failed to set power state
[ERROR] [1697009580.509341897]: An exception has been thrown: failed to set power state
[ERROR] [1697009580.509509252]: Exception: failed to set power state
 11/10 15:33:00,652 WARNING [139761540896512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 11/10 15:33:00,662 WARNING [139761540896512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16

I am using a laptop with a USB port of 3.2, and both USB ports are working properly. I can use the Intel realsense viewer to open two realsense cameras normally.What is the reason for this problem? Thanks!

MartyG-RealSense, 2023-10-11:
Have you defined different serial numbers for each camera in the rs_multiple_devices launch instruction, please?

Is the RealSense Viewer inactive (closed) when you are using the launch instruction?

hushimin-create, 2023-10-11:
Thank you very much. I have successfully connected two cameras. I did forget to define the serial number for each camera in the rs_multiple_devices launch instruction.

MartyG-RealSense, 2023-10-11:
You are very welcome.  It's great to hear that you succeeded.  :)

closed 2023-10-17 by MartyG-RealSense