realsenseai/realsense-ros #2883: (d400-factory.cpp) DS5 group_devices is empty

jesusramondovale, 2023-09-27:
Hi! I'm using a DEBIX Model-A (aarch64 ARMv8) board with Ubuntu-22.04 and kernel 5.15.71 installed on it (with ROS2 Humble) 

I've got a problem with my Realsense D435i Camera: I've succesfully compiled from sources the SDK (librealsense) so I can correctly use the `rs-enumerate-devices` command (only with sudo before) 

but, when I execute the ROS2 Wrapper with:
`ros2 run realsense2_camera realsense2_camera_node`

I get this output: 

`[INFO] [launch]: All log files can be found below /home/debix/.ros/log/2023-09-27-06-45-57-373454-DEBIX25-14668 
[INFO] [launch]: Default logging verbosity is set to INFO 
[INFO] [realsense2_camera_node-1]: process started with pid [14682] [realsense2_camera_node-1] 
[INFO] [1695797158.404652736] [camera.camera]: RealSense ROS v4.54.1 [realsense2_camera_node-1] 
[INFO] [1695797158.404968999] [camera.camera]: Built with LibRealSense v2.54.1 [realsense2_camera_node-1] 
[INFO] [1695797158.405047252] [camera.camera]: Running with LibRealSense v2.54.1 [realsense2_camera_node-1]  27/09 06:45:58,435 WARNING [281472476768480] (d400-factory.cpp:1195) DS5 group_devices is empty. [realsense2_camera_node-1]  27/09 06:45:58,830 ERROR [281472476768480] (librealsense-exception.h:52) xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument [realsense2_camera_node-1]  27/09 06:45:58,837 WARNING [281472476768480] (rs.cpp:312) null pointer passed for argument "device" [realsense2_camera_node-1] 
[WARN] [1695797158.837362369] [camera.camera]: Device 1/3 failed with exception: xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument 
[ERROR] [realsense2_camera_node-1]: process has died [pid 14682, exit code -11, cmd '/home/debix/ros2_ws/install/realsense2_camera/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_j9ko_vrh --params-file /tmp/launch_params_w3fbxgpj'].`

*NOTE: I tried installing the ROS2-Wrapper (apt-get install) and also compiling it from source but both options gave the same error*

MartyG-RealSense, 2023-09-27:
**xioctl(VIDIOC_S_EXT_CTRLS)** errors typically indicate that there is a conflict between librealsense and the Linux kernel.  

Kernel version 5.15 should be fine in general, though there is the possibility of problems with specific kernel versions.

Can you confirm whether using **ros2 launch** also does not work for you, please?

`ros2 launch realsense2_camera rs_launch.py`

jesusramondovale, 2023-09-27:
Yes, doing 
`ros2 launch realsense2_camera rs_launch.py` 

gives exactly the same output

MartyG-RealSense, 2023-09-27:
I agree that it could be a permissions issue.

Does it make a difference if you input **sudo su** into the terminal before you input the **ros2 run** launch command?

jesusramondovale, 2023-09-27:
No. It doesn't: 

[image]

MartyG-RealSense, 2023-09-27:
**DS5 group_devices is empty** suggests to me that the camera is not being detected by the launch.

What happens if the camera is reset at launch by adding **initial_reset:=true** to the ros2 launch instruction?

`ros2 launch realsense2_camera rs_launch.py initial_reset:=true`

jesusramondovale, 2023-09-27:
Nothing changed using that flag: (I set the logs to DEBUG mode) 

[image]
[image]

MartyG-RealSense, 2023-09-27:
1.  Is there any difference if you reload the **udev** device handling rules with the instruction below?

`sudo udevadm control --reload-rules && udevadm trigger`

2.  When you installed librealsense from source code, was the build flag **-DFORCE_RSUSB_BACKEND=TRUE** included in the build instruction?  An RSUSB build of the SDK can bypass problems caused by kernel conflicts.

jesusramondovale, 2023-09-27:
I builded it using -DFORCE_RSUSB_BACEND=ON : 

`cmake ../ -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DFORCE_RSUSB_BACKEND=ON -DBUILD_WITH_TM2=false -DIMPORT_DEPTH_CAM_FW=false -DCMAKE_EXE_LINKER_FLAGS="-latomic"`

MartyG-RealSense, 2023-09-27:
If you built with **-DFORCE_RSUSB_BACKEND=ON** then there is not likely to be a conflict with the Linux kernel, as RSUSB bypasses the kernel.

Using sudo superuser mode with the ROS wrapper does not help.  So it may not be a permissions related issue after all, especially as librealsense works normally in sudo mode.

1.  Is the camera detected if, after a failed launch, you unplug the camera, re-insert it and then run the launch instruction again a second time?

2.  Does specifying the camera device type help?

`ros2 launch realsense2_camera rs_launch.py device_type:=d435`

jesusramondovale, 2023-09-27:
But it is -DFORCE_RSUSB_BACKEND=ON or -DFORCE_RSUSB_BACKEND=true? I used ON

1. Yes. I executed the ros2 launch (error output), unplugged and reinserted the camera, launched again and nothing changed. Same error output.
2. No. Same output with
`ros2 launch realsense2_camera rs_launch.py device_type:=d435`

One question: which kernel modules are needed to librealsense to work well?

MartyG-RealSense, 2023-09-27:
Either ON or TRUE should work.

If RSUSB = true then the kernel is bypassed, so it should not matter which kernel the computer is using.  

At the time of writing this, librealsense officially supports kernel versions 4, 4.8, 4.10, 4.13, 4.15, 4.18, 5.0, 5.3, 5.4, 5.13 and 5.15.

jesusramondovale, 2023-09-27:
No, I meant the names of the kernel "packages" like uvcvideo... etc. Which ones it uses?

MartyG-RealSense, 2023-09-27:
The core packages required to build librealsense are:

libssl-dev libusb-1.0-0-dev 
libudev-dev pkg-config 
libgtk-3-dev

Of those packages, the ones referenced as requirements in the 5.15 kernel patch script are:

libssl-dev libusb-1.0-0-dev 
libudev-dev pkg-config 

https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-realsense-ubuntu-lts-hwe.sh

When using RSUSB = true though, a source code build of librealsense does **not** need to have a kernel patch script applied.

jesusramondovale, 2023-09-28:
Hi again! I reinstalled the SDK compiling from source with -DFORCE_RSUSB_BACKEND=true, the `rs-enumerate-devices` works (only with sudo) 

The ROS2 wrapper (installed via apt-get install) doesn't work. Same output.

MartyG-RealSense, 2023-09-28:
You could try deleting the entire **/ros2_ws/src** catkin workspace folder to remove the source code installation of the wrapper and then build the wrapper from source again.

I would recommend removing the realsense2_camera package first though as it is not advisable to have both package and source code installations on the same computer.

If you do not mind reinstalling librealsense before installing the wrapper again then a quick way to remove the package would be to remove all RealSense-related packages from the computer with the instruction below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

jesusramondovale, 2023-09-28:
I don't know if I'm getting crazy or what. I don't believe what has just happened: 
I executed: 
`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

And next I tried to run (to be sure that it is now removed)
`ros2 run realsense2_camera realsense`

and INCREDIBLY now it works!!!! Both ros2 run and ros2 launch works

MartyG-RealSense, 2023-09-28:
That's great to hear.  When everything else has not worked, sometimes you can get lucky with an uninstall without ever knowing what the cause of the problem was.

MartyG-RealSense, 2023-10-05:
Hi @jesusramondovale  Do you require further assistance with this case, please?  Thanks!

jesusramondovale, 2023-10-06:
After a couple of days using the camera, the exact same error appeared again

jesusramondovale, 2023-10-13:
It's very strange because it started working again with no reason, I can't give you good feedback

MartyG-RealSense, 2023-10-13:
Okay, please do let me know if it stops working again.  I will keep this case open in the meantime.

MartyG-RealSense, 2023-10-28:
Hi @jesusramondovale  is ROS still working correctly for you after a 2 week period since your last comment, please?  Thanks!

closed 2023-11-10 by MartyG-RealSense