realsenseai/realsense-ros #2884: json file loading seems not working

hyunseok-yang, 2023-09-27:


MartyG-RealSense, 2023-09-27:
Hi @hyunseok-yang  Does the depth exposure change if you set it in the launch instruction instead of a json?  For example, to set depth exposure to a value of 33000:

`ros2 launch realsense2_camera rs_launch.py depth_module.exposure:=33000`

MartyG-RealSense, 2023-10-03:
Hi @hyunseok-yang  Do you require further assistance with this case, please?  Thanks!

hyunseok-yang, 2023-10-05:
@MartyG-RealSense 

What we trying to do was turning off the `auto exposure` by just set `False` in configuration file like below.
```json
        "controls-autoexposure-auto": "False",
        "controls-color-autoexposure-auto": "False",
```

Tried launching command adding `json_file_path`.
```shell
ros2 launch realsense2_camera rs_launch.py  rgb_camera.profile:=640x480x6  depth_module.profile:=640x480x6 
 json_file_path:=/config.json
```
But, 'auto exposure' feature was NOT turning off.

After several trials, we've found how to launch it as what we desired.

```
ros2 launch realsense2_camera rs_launch.py     align_depth.enable:=true     rgb_camera.profile:=640x480x6     depth_module.profile:=640x480x6  rgb_camera.enable_auto_exposure:=false depth_module.enable_auto_exposure:=false  json_file_path:=/config.json
```

This `rgb_camera.enable_auto_exposure:=false` argument was the key point to turn off the feature. 
Then I've checked that other parameters kind of brigthness are working as we change the value.

I think `controls-autoexposure-auto` parameter in config file should be worked no matter how the ros launch argument is passed or not.

Is there the priority of argument setting? 

Here is the config file as an sample. saved it from the realsense-viewer.
```json
{
    "device": {
        "fw version": "5.15.0.2",
        "name": "Intel RealSense D435",
        "product line": "D400"
    },
    "parameters": {
        "aux-param-autoexposure-setpoint": "1536",
        "aux-param-colorcorrection1": "0.298828",
        "aux-param-colorcorrection10": "-0",
        "aux-param-colorcorrection11": "-0",
        "aux-param-colorcorrection12": "-0",
        "aux-param-colorcorrection2": "0.293945",
        "aux-param-colorcorrection3": "0.293945",
        "aux-param-colorcorrection4": "0.114258",
        "aux-param-colorcorrection5": "-0",
        "aux-param-colorcorrection6": "-0",
        "aux-param-colorcorrection7": "-0",
        "aux-param-colorcorrection8": "-0",
        "aux-param-colorcorrection9": "-0",
        "aux-param-depthclampmax": "65536",
        "aux-param-depthclampmin": "0",
        "aux-param-disparityshift": "0",
        "controls-autoexposure-auto": "False",
        "controls-autoexposure-manual": "8500",
        "controls-color-autoexposure-auto": "False",
        "controls-color-autoexposure-manual": "100",
        "controls-color-backlight-compensation": "0",
        "controls-color-brightness": "0",
        "controls-color-contrast": "50",
        "controls-color-gain": "100",
        "controls-color-gamma": "300",
        "controls-color-hue": "0",
        "controls-color-power-line-frequency": "3",
        "controls-color-saturation": "50",
        "controls-color-sharpness": "50",
        "controls-color-white-balance-auto": "True",
        "controls-color-white-balance-manual": "4960",
        "controls-depth-gain": "16",
        "controls-laserpower": "150",
        "controls-laserstate": "on",
        "ignoreSAD": "0",
        "param-amplitude-factor": "0",
        "param-autoexposure-setpoint": "1536",
        "param-censusenablereg-udiameter": "9",
        "param-censusenablereg-vdiameter": "9",
        "param-censususize": "9",
        "param-censusvsize": "9",
        "param-depthclampmax": "65536",
        "param-depthclampmin": "0",
        "param-depthunits": "1000",
        "param-disableraucolor": "0",
        "param-disablesadcolor": "0",
        "param-disablesadnormalize": "0",
        "param-disablesloleftcolor": "0",
        "param-disableslorightcolor": "0",
        "param-disparitymode": "0",
        "param-disparityshift": "0",
        "param-lambdaad": "800",
        "param-lambdacensus": "26",
        "param-leftrightthreshold": "24",
        "param-maxscorethreshb": "2047",
        "param-medianthreshold": "500",
        "param-minscorethresha": "1",
        "param-neighborthresh": "7",
        "param-raumine": "1",
        "param-rauminn": "1",
        "param-rauminnssum": "3",
        "param-raumins": "1",
        "param-rauminw": "1",
        "param-rauminwesum": "3",
        "param-regioncolorthresholdb": "0.0499022",
        "param-regioncolorthresholdg": "0.0499022",
        "param-regioncolorthresholdr": "0.0499022",
        "param-regionshrinku": "3",
        "param-regionshrinkv": "1",
        "param-robbinsmonrodecrement": "10",
        "param-robbinsmonroincrement": "10",
        "param-rsmdiffthreshold": "4",
        "param-rsmrauslodiffthreshold": "1",
        "param-rsmremovethreshold": "0.375",
        "param-scanlineedgetaub": "72",
        "param-scanlineedgetaug": "72",
        "param-scanlineedgetaur": "72",
        "param-scanlinep1": "60",
        "param-scanlinep1onediscon": "105",
        "param-scanlinep1twodiscon": "70",
        "param-scanlinep2": "342",
        "param-scanlinep2onediscon": "190",
        "param-scanlinep2twodiscon": "130",
        "param-secondpeakdelta": "325",
        "param-texturecountthresh": "0",
        "param-texturedifferencethresh": "0",
        "param-usersm": "1",
        "param-zunits": "1000"
    },
    "schema version": 1,
    "viewer": {
        "stream-depth-format": "Z16",
        "stream-fps": "30",
        "stream-height": "480",
        "stream-width": "848"
    }
}
```

MartyG-RealSense, 2023-10-05:
It is excellent to hear that you achieved a solution!  Thanks so much for sharing information about the solution and your json file with the RealSense ROS community!

As far as I am aware it should not make a difference which order RGB and depth auto-exposure are disabled in during launch, though it does seem to matter in your particular case.

hyunseok-yang, 2023-10-12:
> Hi @hyunseok-yang Do you require further assistance with this case, please? Thanks!

I think we've got what we want now.

If we have a problem, I will ask later. thank you.

closed 2023-10-12 by hyunseok-yang