realsenseai/realsense-ros #2885: Cannot run the camera node.

leelening, 2023-10-01:
Get the following error when run `roslaunch realsense2_camera rs_camera.launch`
[image]

MartyG-RealSense, 2023-10-01:
Hi @leelening   There is a past case of the **Make sure that you are calling the pluginlib_export_class macro** error at https://github.com/IntelRealSense/realsense-ros/issues/2216

RealSense users who commented on that case seemed to be having a problem with more than one version of librealsense installed on their computer.  Performing a removal and then a reinstall of librealsense 2.50.0 fixed the problem (using librealsense 2.51.1 should be fine too).  If you are using Noetic then the ROS wrapper version used with 2.50.0 or 2.51.1 should be **2.3.2**.

leelening, 2023-10-01:
I think my case is that I installed the latestest version (2.54) instead of 2.50.0

closed 2023-10-01 by leelening

MartyG-RealSense, 2023-10-02:
That's excellent news that it is working now, @leelening - thanks very much for the update!

MartyG-RealSense, 2023-10-02:
Whether you have realsense-viewer or not depends on the method that was used to install librealsense.

When installing librealsense from packages, the tools and examples (including realsense-viewer) are installed when the utils package is installed with the instruction `sudo apt-get install librealsense2-utils`

When building librealsense from source code with CMake, realsense-viewer will be included in the SDK build if the CMake build instruction contains the flag `-DBUILD_GRAPHICAL_EXAMPLES=TRUE`

If librealsense 2.50.0 and the ROS1 wrapper are installed from packages at the same time using the instruction `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` then realsense-viewer is not included in the librealsense installation.