realsenseai/realsense-ros #2886: Pointcloud2 color texture Issue with D405 Camera and Ros2 driver

ndaley7, 2023-10-01:
Good Afternoon,

I'm currently trying to utilize a D405 camera with the Ros2 version of the node, and I am experiencing an odd issue.

I am Starting up the camera using the exact commands from the documentation [here](https://github.com/IntelRealSense/realsense-ros/tree/ros2-development#start-camera-node):

```
ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 pointcloud.enable:=true
```
After booting up Rviz2, I add an image module to confirm the color image is working, and a pointcloud2 module to subscribe to the /points message.

As soon as I enable the Poincloud2 module, the Warning below writes out on the terminal:

```
[realsense2_camera_node-1] [WARN] [1696179054.430680446] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
```
Has anyone experienced anything like this before?

I am attempting this on a Jetson Nano Orin 8Gb within a docker containing all of the relevant Ros Modules.

Please see screenshots below with terminal output/version info along with another screenshot showing the firmware on the camera:

[image]

[image]

MartyG-RealSense, 2023-10-02:
Hi @ndaley7  ROS2 wrapper 4.54.1 is designed to be used with librealsense **2.54.1** rather than the 2.53.1 version you are using.  A ROS2 wrapper for 2.53.1 was not created.  It is important to match a particular ROS wrapper version as closely as possible with a specific librealsense version listed in that particular wrapper's release notes.

The camera firmware driver version that should be used with 2.53.1 is **5.14.0.0** rather than 5.15.1.0 (which is designed for use with the current latest 2.54.2 version of librealsense).  The appropriate firmware match for librealsense 2.54.1 is **5.15.0.2**.

The **No stream match for pointcloud chosen texture Process - Color** warning is generated if the wrapper detects missing color image frames.

Does the message still occur if you reduce resolution to 848x480, please?

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=848x480x30 pointcloud.enable:=true`

ndaley7, 2023-10-02:
> Hi @ndaley7 ROS2 wrapper 4.54.1 is designed to be used with librealsense **2.54.1** rather than the 2.53.1 version you are using. A ROS2 wrapper for 2.53.1 was not created. It is important to match [...]

Oh I had no idea about the version mismatch.  As these were the versions set up on the provided docker, I went with it.  I'll give the new resolution a try!

MartyG-RealSense, 2023-10-02:
Given that the session is having difficulty obtaining color frames, is the pointcloud generated if you texture the pointcloud with the **infrared** image instead of the color image by adding **pointcloud.stream_filter:=1** to the end of your launch instruction?

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 pointcloud.enable:=true pointcloud.stream_filter:=1`

ndaley7, 2023-10-02:
> ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 pointcloud.enable:=true pointcloud.stream_filter:=1

[image]
Unfortunately a similar result.

I think I'll also try the previous commands with a d435 that we have onhand to see if there is a difference when the camera model is changed.

ndaley7, 2023-10-03:
A little more info to add around this issue.  I tried to run the same script with the d435 camera and received the same stream error as a result:

[image]

MartyG-RealSense, 2023-10-03:
What happens if Fixed Frame is set to **camera_link**

MartyG-RealSense, 2023-10-04:
There used to be problems with using pointcloud or depth-color alignment on Jetson boards specifically.  When pointcloud or alignment was not enabled, performance was fine.  That problem has become very rare now but it is still possible for the occasional case to be reported.

Instead of enabling the pointcloud, could you try enabling alignment please?  If alignment does not work correctly then it could indicate that you are experiencing this now-rare problem.

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 align_depth.enable:=true`

**********

It appears that we also have not tested whether the pointcloud publishing problem occurs when launching with **ros2 run** instead of ros2 launch.

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p pointcloud.enable:=true`

MartyG-RealSense, 2023-10-11:
Hi @ndaley7  Do you require further assistance with this case, please?  Thanks!

ndaley7, 2023-10-13:
> There used to be problems with using pointcloud or depth-color alignment on Jetson boards specifically. When pointcloud or alignment was not enabled, performance was fine. That problem has become [...]

Sorry for the delay, I'll give these a try

MartyG-RealSense, 2023-10-13:
Now that the pointcloud is publishing, does the **pointcloud.stream_filter** parameter (which can use the infrared stream as the pointcloud texture instead of the RGB stream) work if tried again?

ros2 run realsense2_camera realsense2_camera_node --ros-args -p pointcloud.enable:=true -p pointcloud.stream_filter:=1

ndaley7, 2023-10-19:
> Now that the pointcloud is publishing, does the **pointcloud.stream_filter** parameter (which can use the infrared stream as the pointcloud texture instead of the RGB stream) work if tried again? [...]

[image]
Similar error but this time for the depth stream.

MartyG-RealSense, 2023-10-20:
If you are still using librealsense 2.53.1 then the firmware driver currently installed in your camera (5.15.1.0) is incorrect for that SDK.  The recommended firmware for 2.53.1 is **5.14.0.0**.  Using a firmware other than the recommended one for that particular librealsense version can result in errors.

ndaley7, 2023-10-20:
> If you are still using librealsense 2.53.1 then the firmware driver currently installed in your camera (5.15.1.0) is incorrect for that SDK. The recommended firmware for 2.53.1 is **5.14.0.0**. [...]

Hmm Okay that is good to know.... is there a procedure for downgrading the firmware?

MartyG-RealSense, 2023-10-20:
The easiest method is to use the RealSense Viewer tool.

1.  Visit the link below and download the firmware file for 5.14.0.0 from it.

2.  Open the RealSense Viewer, click on the **More** option near the top of the Viewer's options side-panel and select **Update Firmware** from its drop-down menu.

3.  Doing so causes a file selector to pop up.  Navigate to the location on your computer that you downloaded the firmware file to and select it.  The firmware downgrade should begin automatically.

ndaley7, 2023-10-22:
> The easiest method is to use the RealSense Viewer tool.
>
> 1. Visit the link below and download the firmware file for 5.14.0.0 from it. [...]

I had no idea that the tool could be used to downgrade as well!  I'll give it a try and re-run some of the commands to see if the result is different

ndaley7, 2023-10-22:
Sadly even with the matching firmware the same issue persists:

[image]

**[WARN] [1697988608.741091483] [camera]: No stream match for pointcloud chosen texture Process - Depth**

MartyG-RealSense, 2023-10-23:
What happens if you set Fixed Frames to `camera_link` again and this time also set the **Image** topic to `/camera/color/image_raw` instead of /depth/image_rect_raw so Rviz resembles the picture below with the color image sown at the bottom of the RViz side-panel?

[image]

ndaley7, 2023-10-25:
Appears to be the same issue, I ran the prior command:

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=848x480x30 pointcloud.enable:=true`

[image]

MartyG-RealSense, 2023-10-25:
The problem is likely to be that there are frames being dropped when the pointcloud is enabled (perhaps because of a dropped FPS rate), causing the **No stream match for pointcloud chosen process - Color** message.

I note that we have not yet raised the subject of **CUDA** in this discussion.  The librealsense SDK can be built with CUDA graphics acceleration support on Nvidia Jetson boards.  Because librealsense supports CUDA, so does the ROS wrapper.  This greatly reduces the CPU effort required to process pointclouds or alignment.  This is because the processing work is offloaded from the CPU onto the graphics GPU on the Jetson board.

Which installation method did you use to build librealsense, please?  If it was built using the Jetson packages then CUDA support is automatically included.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

If librealsense was built from source code with CMake though then it is necessary to include the build flag **-DBUILD_WITH_CUDA=TRUE** in the CMake build instruction to add CUDA support.  For example:

`cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=false -DBUILD_WITH_CUDA=true && make -j$(($(nproc)-1)) && sudo make install`

ndaley7, 2023-10-25:
> The problem is likely to be that there are frames being dropped when the pointcloud is enabled (perhaps because of a dropped FPS rate), causing the **No stream match for pointcloud chosen process - Color** message.
> 
> I note that we have not yet raised the subject of **CUDA** in this discussion. The librealsense SDK can be built with CUDA graphics acceleration support on Nvidia Jetson boards. Because librealsense supports CUDA, so does the ROS wrapper. This greatly reduces the CPU effort required to process pointclouds or alignment. This is because the processing work is offloaded from the CPU onto the graphics GPU on the Jetson board.
> 
> Which installation method did you use to build librealsense, please? If it was built using the Jetson packages then CUDA support is automatically included.
> 
> [IntelRealSense/librealsense@`master`/doc/installation_jetson.md#4-install-with-debian-packages](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md?rgh-link-date=2023-10-25T11%3A58%3A23Z#4-install-with-debian-packages)
> 
> If librealsense was built from source code with CMake though then it is necessary to include the build flag **-DBUILD_WITH_CUDA=TRUE** in the CMake build instruction to add CUDA support. For example:
> 
> `cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=false -DBUILD_WITH_CUDA=true && make -j$(($(nproc)-1)) && sudo make install`

The commands are being run on a docker container that came on a premade Jetson image by Nvidia..... I would have assumed that it was CUDA enabled given the hardware, but is there a way I can check that was the case?

MartyG-RealSense, 2023-10-25:
Whilst a Jetson image created by Nvidia may have CUDA installed on it, it would not contain the librealsense CUDA component that adds support for RealSense pointcloud and depth-color alignment acceleration.  That can only be provided by installing librealsense from Jetson packages or building it from source code with the mentioned build flag.

The only way that you would likely know whether librealsense's CUDA support was enabled would be to launch the camera and then look at its CPU percentage usage in an Ubuntu system monitoring tool such as **htop** whilst the ROS session is active.  

With CUDA support enabled the CPU usage when generating a pointcloud might be no higher than 30%, whilst the absence of CUDA support could result in CPU usage of 70% or higher.

ndaley7, 2023-10-25:
> Whilst a Jetson image created by Nvidia may have CUDA installed on it, it would not contain the librealsense CUDA component that adds support for RealSense pointcloud and depth-color alignment [...]

I'll give this a look when home and verify whether cuda is being used or not.

ndaley7, 2023-10-31:
> Whilst a Jetson image created by Nvidia may have CUDA installed on it, it would not contain the librealsense CUDA component that adds support for RealSense pointcloud and depth-color alignment [...]

Good Afternoon,

After digging through the docker build files, it seems that the image does use a librealsense build with CUDA enabled:

[image]

MartyG-RealSense, 2023-11-01:
Thank you for the information and the clarification that you installed librealsense using the JetsonHacks **installLibrealsense** build script.

Which Ubuntu and JetPack versions are you using?  The ideal combinations would be JetPack 4.6 for Ubuntu 18.04 nd JetPack 5.0.2 for Ubuntu 20.04.

MartyG-RealSense, 2023-11-06:
JetPack 5.1.1 is fine for use with Ubuntu 20.04.  However, if librealsense has not been built from source code in RSUSB = true mode then 5.1.1 could be a problem, since librealsense supports the slightly earlier **5.0.2** JetPack version in SDK 2.54.1, and also not until SDK 2.54.1 (you are using SDK 2.53.1).    

As there is no official support in the SDK for any JetPack 5 version until 2.54.1, using an RSUSB = true build of librealsense 2.53.1 from source code (which should bypass JetPack compatibility problems) is definitely worth considering to see whether it resolves your pointcloud enabling issue.  

Using SDK 2.54.1 and ROS2 wrapper 4.54.1 would be even better if possible for you to do so.

MartyG-RealSense, 2023-11-19:
Case closed due to no further comments received.

closed 2023-11-19 by MartyG-RealSense

ndaley7, 2023-12-20:
> Hi @ndaley7 Do you require further assistance with this case, please? Thanks!

> JetPack 5.1.1 is fine for use with Ubuntu 20.04. However, if librealsense has not been built from source code in RSUSB = true mode then 5.1.1 could be a problem, since librealsense supports the slightly earlier **5.0.2** JetPack version in SDK 2.54.1. However, 5.0.2 support was not introduced until SDK 2.54.1 (you are using SDK 2.53.1) so in 2.53.1 the most recent supported JetPack version was **4.6**
> 
> As there is no official support in the SDK for any JetPack 5 version until 2.54.1, using an RSUSB = true build of librealsense 2.53.1 from source code (which should bypass JetPack compatibility problems) is definitely worth considering to see whether it resolves your pointcloud enabling issue.
> 
> Using SDK 2.54.1 and ROS2 wrapper 4.54.1 would be even better if possible for you to do so.

I have not had time to try these fixes, but will report back once I have.

MartyG-RealSense, 2023-12-20:
Thanks very much for the update.  I look forward to your next report.  Good luck!